espressif/esp-idf #4643: BIG Memory leak in esp-mqtt outbox_cleanup()

eespi, 2020-01-17:
## Environment

Module or chip used: ESP32-WROOM-32D
IDF version: v4.0-beta1-279-g0a03a55
Build System: Make
Compiler version: 8.2.0
Operating System: Linux
Power Supply: external 3.3V

## Problem Description

Calling esp_mqtt_destroy_client leaks a lot of memory of memory because outbox is not correctly freed.

The leak is generated inside outbox_cleanup() function, which dequeues only messages that have been tagged with the pending state CONFIRMED.
The application wil leak an amount of bytes tied to the queued message length.

### Expected Behavior
When mqtt client is destroyed, all resources allocated by the outbox should be freed. 
### Actual Behavior

### Steps to reproduce

1. connect to mqtt broker
2. publish on some topics
3. stop mqtt client after message has been published
4. destroy client
5. loop

### Code to reproduce this issue

```cpp
esp_err_t outbox_cleanup(outbox_handle_t outbox, int max_size)
{
    //outbox size is 1458 byte
    while (outbox_get_size(outbox) > max_size) {
        //THIS LINE WILL DEQUEUE ONLY MESSAGES IN CONFIRMED STATE
        outbox_item_handle_t item = outbox_dequeue(outbox, CONFIRMED, NULL);
        //Item is always NULL
        if (item == NULL) { 
            printf("Item NULL, LEAK!!!!!!!!!!!!!\r\n");
            return ESP_FAIL;
        }
        STAILQ_REMOVE(outbox, item, outbox_item, next);
        free(item->buffer);
        free(item);
    }
    return ESP_OK;
}
```
## Debug Logs

```cpp
DEQUEUE OUTBOX SIZE 1458
Item NULL, LEAK!!!!!!!!!!!!!

[..]

36 bytes (@ 0x3ffdbc8c) allocated CPU 0 ccount 0x61d97bb0 caller 0x40150531:0x4014f5d0:0x401502e5:0x400e9ad1
0x40150531: outbox_enqueue at /media/paolo/Data/working/FW/esp32/esp-idf/components/mqtt/esp-mqtt/lib/mqtt_outbox.c:36
0x4014f5d0: mqtt_enqueue at /media/paolo/Data/working/FW/esp32/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:817
0x401502e5: esp_mqtt_client_publish at /media/paolo/Data/working/FW/esp32/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1434
0x400e9ad1: MQTTDRV_Publish at /media/paolo/Data/working/FW/esp32/2018_fw_wifi_esp32_v2/wifi_esp32_prj/components/Mqtt/MqttDrv.c:238

899 bytes (@ 0x3ffdc6c8) allocated CPU 0 ccount 0x61d98a10 caller 0x4015058c:0x4014f5d0:0x401502e5:0x400e9ad1
0x4015058c: outbox_enqueue at /media/paolo/Data/working/FW/esp32/esp-idf/components/mqtt/esp-mqtt/lib/mqtt_outbox.c:45 (discriminator 2)
0x4014f5d0: mqtt_enqueue at /media/paolo/Data/working/FW/esp32/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:817
0x401502e5: esp_mqtt_client_publish at /media/paolo/Data/working/FW/esp32/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1434
0x400e9ad1: MQTTDRV_Publish at /media/paolo/Data/working/FW/esp32/2018_fw_wifi_esp32_v2/wifi_esp32_prj/components/Mqtt/MqttDrv.c:238

36 bytes (@ 0x3ffdbcb4) allocated CPU 0 ccount 0x626b947c caller 0x40150531:0x4014f5d0:0x401502e5:0x400e9ad1
0x40150531: outbox_enqueue at /media/paolo/Data/working/FW/esp32/esp-idf/components/mqtt/esp-mqtt/lib/mqtt_outbox.c:36
0x4014f5d0: mqtt_enqueue at /media/paolo/Data/working/FW/esp32/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:817
0x401502e5: esp_mqtt_client_publish at /media/paolo/Data/working/FW/esp32/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1434
0x400e9ad1: MQTTDRV_Publish at /media/paolo/Data/working/FW/esp32/2018_fw_wifi_esp32_v2/wifi_esp32_prj/components/Mqtt/MqttDrv.c:238

559 bytes (@ 0x3ffdca50) allocated CPU 0 ccount 0x626ba150 caller 0x4015058c:0x4014f5d0:0x401502e5:0x400e9ad1
0x4015058c: outbox_enqueue at /media/paolo/Data/working/FW/esp32/esp-idf/components/mqtt/esp-mqtt/lib/mqtt_outbox.c:45 (discriminator 2)
0x4014f5d0: mqtt_enqueue at /media/paolo/Data/working/FW/esp32/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:817
0x401502e5: esp_mqtt_client_publish at /media/paolo/Data/working/FW/esp32/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1434
0x400e9ad1: MQTTDRV_Publish at /media/paolo/Data/working/FW/esp32/2018_fw_wifi_esp32_v2/wifi_esp32_prj/components/Mqtt/MqttDrv.c:238

[...]

1909 bytes 'leaked' in trace (29 allocations)
total allocations 1165 total frees 18476
```

eespi, 2020-01-28:
nothing?

Alvin1Zhang, 2020-01-29:
Thanks for reporting and sorry for the slow turnaround. We have been working on this and will share once updates are available. Thanks.

eespi, 2020-01-29:
Ok thank you.
I have patched it internally, but I'm still see ~100 bytes leak when client disconnects from broker. Don't know yet if it's esp-mqtt related

david-cermak, 2020-03-03:
@eespi Thanks again for reporting. The resolving mqtt commit https://github.com/espressif/esp-mqtt/commit/f7941e29beae2b426d4a7ec5f95e58fed7a96762 has beed referenced to idf master in https://github.com/espressif/esp-idf/commit/b8e47eb56f7aad00eda76b508816287fa04019b8 so this issue could be closed.

closed 2020-03-03 by Alvin1Zhang