espressif/esp-idf #4644: Docker images does not contain correct version of CMake

volk-christian, 2020-01-17:
## Problem Description
The provided docker image contains the wrong CMake version.
-> Expected version: 3.13.4
(compare: https://github.com/espressif/esp-idf/blob/release/v4.0/tools/tools.json)
-> Actual version: 3.10.2
(compare: docker run -it espressif/idf:release-v4.0 cmake --version)

### Expected Behavior
It is expected, that all provided build environments utilize the exact same tools.

### Actual Behavior
Docker container uses a version of cmake which is deviated from the given tools.json file.

### Steps to reproduce
1. docker run -it espressif/idf:release-v4.0 cmake --version

igrr, 2020-01-17:
Hi @volk-christian, for CMake, the recommendation (reflected in the getting started guides for Linux) is to install it from the OS package manager. The links in tools.json are provided merely for convenience, in case the OS provides only an old version of CMake.

Have you encountered any problem using the version of CMake installed in the Docker image?

volk-christian, 2020-01-17:
Hi @igrr, thanks for your fast reply.
Indeet, we are encoutering problems with installed CMake version in docker image in conjunction with complex library dependencies. We did not have a problem while building our project natively on windows, while we were not able to link the application elf file by the help of the provided docker image.
After investigation and CMake update within the container, it's possible to build our application.

Since I'm always afraid of such kind of problems, I'd highly appreciate to install the very same version of required tools in each possible environment.

igrr, 2020-01-17:
ESP-IDF projects should work with CMake versions from 3.5 and above, so I think you may be running either into a build system bug, or your project does correctly list component requirements (or other library dependencies, if you are using external CMake libraries). Can you provide some details about the failure you are seeing?

volk-christian, 2020-01-17:
Currently we are porting our sw-platform onto the ESP32 chipset.
During this step, we are evaluating ESP-IDF and CMake Build system.
Unfortunately, I cannot share lots of details about our platform, except for the fact that we (currently) have a lot of inter-modular dependencies.
Our linking issues occur definitely in conjunction with our own components, but only in one flavor of the provivded ESP-IDF environments.

I'd expect to get the very same version of the required tools as long as I'm sticking to the very same release branch (or even a tag) at one point of time. As long as you are relying on the OS package manager, an external force can influence generation of the underlying build system. In my eyes this is not consistent.

But you should already have everything to be consistent: By the help of your tools.json and requirements.txt files, you should be able to install the very same environment on each given base system. I'd highly appreciate to utilize this mechanism also during the build of the docker image.

igrr, 2020-01-17:
Sorry, as I understand, if you correctly specify the minimum supported CMake version in your CMakeLists files, the behavior should be the same on different versions of CMake. If this is not the case, I would say that this is a bug which should be investigated. In general, we can't always guarantee that the versions of all tools will be the same on all OSes. For example, we can't install ccache on Linux using the same method as we are using to install the toolchain.

Can you please try building the project under an older version of CMake, to see if the issue is reproduced on Windows as well?

negativekelvin, 2020-01-17:
A container is probably the only place it is practical to provide a frozen environment so it might be worth considering.  But it would not prevent the issue here of comparing a container build and a native build.

volk-christian, 2020-01-20:
Well, defining the CMake min version in our (third party) code is actually what we did.
And the result is, that we cannot use the provided docker image anymore, since it's configuration deviates from the one which is used for installing the windows toolchain and so the build does not start at all because of an outdated CMake version.

I updated cmake in the running docker container by hand to the version referenced in the tools.json file - just to be sure that we really have an issue conjuncted to the version of CMake. And lo and behold, the build issue was gone.

In my eyes, it's worth to consider the way required tools get installed in order to standardize the deployed tool versions among all possible distributions.

igrr, 2020-01-28:
Will keep this open until the fix is also merged into the release branches.

igrr, 2020-05-26:
Fixed on release/v4.0 branch in https://github.com/espressif/esp-idf/commit/1b0473494fbaf53c945d267ac0cfbc2c8b78d98e. Newer version of CMake should now be part of the docker image with `release-v4.0` and `v4.0.1` tags.

closed 2020-05-26 by igrr