espressif/esp-idf #4646: [a2dp_sink] AVRCP Absolute volume control commands not Working.

Tarun-Narain, 2020-01-18:
I was adding **AVRCP absolute volume commands**, while trying to increase volume using absolute volume commandas passthrough volume commands are not supporter by android. i was able to PLAY and PAUSE music using passthrough commands, cant change volume with the same.
```
 esp_avrc_ct_send_set_absolute_volume_cmd(1, 0);
```
This command should mute the sound on Phone(android), but the following error arises : 
```
W (46988) BT_BTC: handle_rc_metamsg_rsp: code 10 error 0x0
```

Thanks
Tarun Narain

AbnerFederer, 2020-01-19:
Hi @Tarun-Narain,
Maybe you can change
`#define AVRC_INCLUDED              FALSE`
to
`#define AVRC_INCLUDED               TRUE`

Have a try. If the issue still occurs, please provide more info for us to debug.

Thanks
Abner

Tarun-Narain, 2020-01-20:
Hi @AbnerFederer,
It is already True there.

Thanks
Tarun Narain

AbnerFederer, 2020-01-20:
Hi, @Tarun-Narain 

Have you call 
```c
esp_avrc_ct_send_set_absolute_volume_cmd(APP_RC_CT_TL_RN_VOLUME_CHANGE, event_parameter->volume + 5);
        bt_av_volume_changed();
```
like above?

AbnerFederer, 2020-02-19:
Hi, @Tarun-Narain

`Absolute volume cmd ` and `Passthrough cmd` are two different methods to change the volume. You should use them separately.

Thanks
Abner

Tarun-Narain, 2020-03-03:
Hi guys,
 i just figured out that these commands are not working on Xiaomi Smartphones but working on Samsung Smartphones, the volume simulation in A2dp_sink example too is not working on Xiaomi.

AbnerFederer, 2020-03-04:
Hi, @Tarun-Narain 

Which commands? Passthrough or Absolute Volume?
Which Xiaomi phone and Sumsang phone?
Could you provide your code for me to reproduce?

Thanks
Abner

Alvin1Zhang, 2020-03-27:
@Tarun-Narain Thanks for reporting, would you please help share more details? Thanks.

Alvin1Zhang, 2020-04-07:
@Tarun-Narain Thanks for reporting, any more details could be shared? Thanks.

Tarun-Narain, 2020-04-07:
my device is Xiomi Y1 Lite, when i tested a2dp_sinnk example on this device volume simulation was not working but working on every Samsung smartphones and i cant provide any further details as i dont know if its an issue with Xiomi OS (Smartphone).

benjavita, 2020-05-15:
Hello,

Anyone has been able to resolve the problem, I'm having the same issue...
`W (27106) BT_BTC: handle_rc_metamsg_rsp: code 10 error 0x0`
when calling esp_avrc_ct_send_set_absolute_volume_cmd
while esp_avrc_ct_send_passthrough_cmd doesn't seem to work at all with volume.

Thanks

Tarun-Narain, 2020-05-18:
sorry @benjavita, i found nothing on this issue;

Thanks

xiongweichao, 2020-09-29:
If you want to control the volume of your phone, you can use `esp_ avrc_ ct_ send_ passthrough_ cmd `or `notification event`. You can refer to a2dp_sink_demo. Of course, some phones don't support this feature.
Wiki:https://en.wikipedia.org/wiki/List_of_Bluetooth_profiles#Audio/Video_Remote_Control_Profile_(AVRCP)

Alvin1Zhang, 2020-10-12:
@Tarun-Narain Thanks for reporting and sharing updates, would you please help check if your issue has been resolved? Thanks.

closed 2020-10-25 by Tarun-Narain

expresspotato, 2024-12-31:
To anyone reading this, esp_avrc_ct_send_set_absolute_volume_cmd is when the device is a source and not a sink. 

The correct way to change the source's absolute volume from the sink is:

```
esp_avrc_tg_init();
esp_avrc_tg_register_callback(app_bt_sys_callback_avrc_target);
```

Then make a request to change the absolute volume

```
esp_avrc_rn_param_t rn_param;
rn_param.volume = 32; // 0..127
esp_avrc_tg_send_rn_rsp(ESP_AVRC_RN_VOLUME_CHANGE, ESP_AVRC_RN_RSP_INTERIM, &rn_param);
```

And wait for the notification event in the AVRC callback and send the request again.

```
void app_bt_sys_callback_avrc_target(esp_avrc_tg_cb_event_t event, esp_avrc_tg_cb_param_t *param) {
  case ESP_AVRC_TG_REGISTER_NOTIFICATION_EVT: {
          ESP_LOGI(TAG, "AVRC register event notification: %d, param: 0x%"PRIx32, rc->reg_ntf.event_id, rc->reg_ntf.event_parameter);
          if (rc->reg_ntf.event_id == ESP_AVRC_RN_VOLUME_CHANGE) {
            esp_avrc_rn_param_t rn_param;
            rn_param.volume = 32; // 0..127
            esp_avrc_tg_send_rn_rsp(ESP_AVRC_RN_VOLUME_CHANGE, ESP_AVRC_RN_RSP_INTERIM, &rn_param);
          }
          break;
  }
}
```