espressif/esp-idf #4648: mistake in the ICMP Echo documentation

nopnop2002, 2020-01-20:
Error:

	void initialize_ping()
	{
	    /* convert URL to IP address */
	    getaddrinfo("www.espressif.com", NULL, &hint, &res) == 0);
	    struct in_addr addr4 = ((struct sockaddr_in *) (res->ai_addr))->sin_addr;
	    inet_addr_to_ip4addr(ip_2_ip4(&target_addr), &addr4);
	    freeaddrinfo(res);

	    esp_ping_config_t ping_config = ESP_PING_DEFAULT_CONFIG();
	    ping_config.target_addr = target_addr;          // target IP address
	    ping_config.count = ESP_PING_COUNT_INFINITE;    // ping in infinite mode, esp_ping_stop can stop it

	    /* set callback functions */
	    esp_ping_callbacks_t cbs;
	    cbs.on_ping_success = test_on_ping_success;
	    cbs.on_ping_timeout = test_on_ping_timeout;
	    cbs.on_ping_end = test_on_ping_end;
	    cbs.cb_args = "foo";  // arguments that will feed to all callback functions, can be NULL
	    cbs.cb_args = eth_event_group;

	    esp_ping_handle_t ping;
	    esp_ping_new_session(&ping_config, &cbs, &ping);
	}

Correct:

	void initialize_ping()
	{
	    /* convert URL to IP address */
	    ip_addr_t target_addr;
	    struct addrinfo hint;
	    struct addrinfo *res = NULL;
	    getaddrinfo("www.espressif.com", NULL, &hint, &res);
	    struct in_addr addr4 = ((struct sockaddr_in *) (res->ai_addr))->sin_addr;
	    inet_addr_to_ip4addr(ip_2_ip4(&target_addr), &addr4);
	    freeaddrinfo(res);

	    esp_ping_config_t ping_config = ESP_PING_DEFAULT_CONFIG();
	    ping_config.target_addr = target_addr;          // target IP address
	    ping_config.count = ESP_PING_COUNT_INFINITE;    // ping in infinite mode, esp_ping_stop can stop it

	    /* set callback functions */
	    esp_ping_callbacks_t cbs = {
	        .on_ping_success = cmd_ping_on_ping_success,
	        .on_ping_timeout = cmd_ping_on_ping_timeout,
	        .on_ping_end = cmd_ping_on_ping_end,
	        .cb_args = NULL
	    };
	    esp_ping_handle_t ping;
	    esp_ping_new_session(&ping_config, &cbs, &ping);
	    esp_ping_start(ping);
	}

suda-morris, 2020-01-20:
@nopnop2002 Well spotted! Thank you.

nopnop2002, 2020-01-20:
@suda-morris 
Is this related to this issue? ?
https://github.com/espressif/esp-idf/issues/4479#issuecomment-575904023

suda-morris, 2020-01-20:
@nopnop2002  Nope.

nopnop2002, 2020-01-20:
OK
Please check anker of this document.

	WiFi Station: examples/wifi/getting_started/station/main/station_example_main.c
	WiFi Access Point: examples/wifi/getting_started/softAP/main/softap_example_main.c
	Ethernet examples/ethernet/basic/main/ethernet_example_main.c

nopnop2002, 2020-01-20:
I tested initialize_ping().

My environment:

	freeRTOS version:V8.2.0
	NEWLIB version:3.0.0
	lwIP version:2-1-3-0
	ESP-IDF version:v4.1-dev-1935-g647cb628a

**Case 1**
hint variables are not initialized.

	void initialize_ping()
	{
	    /* convert URL to IP address */
	    ip_addr_t target_addr;
	#if 1
	    struct addrinfo hint;
	#else
	    const struct addrinfo hint = {
	        .ai_family = AF_INET,
	        .ai_socktype = SOCK_STREAM,
	    };
	#endif
	    struct addrinfo *res = NULL;
	    int err = getaddrinfo("www.espressif.com", NULL, &hint, &res);
	    if(err != 0 || res == NULL) {
	        ESP_LOGE(TAG, "DNS lookup failed err=%d res=%p", err, res);
	        return;
	    } else {
	        ESP_LOGI(TAG, "DNS lookup success");
	    }

	    //struct in_addr addr4 = &((struct sockaddr_in *)res->ai_addr)->sin_addr;

	    //struct in_addr addr4 = &((struct sockaddr_in *)res->ai_addr)->sin_addr;
	    struct in_addr addr4 = ((struct sockaddr_in *) (res->ai_addr))->sin_addr;

	    inet_addr_to_ip4addr(ip_2_ip4(&target_addr), &addr4);
	    freeaddrinfo(res);
	    ESP_LOGI(TAG, "target_addr.type=%d", target_addr.type);
	    ESP_LOGI(TAG, "target_addr=%s", ip4addr_ntoa(&(target_addr.u_addr.ip4)));

	    esp_ping_config_t ping_config = ESP_PING_DEFAULT_CONFIG();
	    ping_config.target_addr = target_addr;          // target IP address
	    ping_config.count = ESP_PING_COUNT_INFINITE;    // ping in infinite mode, esp_ping_stop can stop it

	    /* set callback functions */
	    esp_ping_callbacks_t cbs = {
	        .on_ping_success = cmd_ping_on_ping_success,
	        .on_ping_timeout = cmd_ping_on_ping_timeout,
	        .on_ping_end = cmd_ping_on_ping_end,
	        .cb_args = NULL
	    };
	    esp_ping_handle_t ping;
	    esp_ping_new_session(&ping_config, &cbs, &ping);
	    esp_ping_start(ping);
	}

I got this error.

	E (2612) wifi station: DNS lookup failed err=204 res=0x0

**Case 2**
hint variables are initialized.

	void initialize_ping()
	{
	    /* convert URL to IP address */
	    ip_addr_t target_addr;
	#if 0
	    struct addrinfo hint;
	#else
	    const struct addrinfo hint = {
	        .ai_family = AF_INET,
	        .ai_socktype = SOCK_STREAM,
	    };
	#endif
	    struct addrinfo *res = NULL;
	    int err = getaddrinfo("www.espressif.com", NULL, &hint, &res);
	    if(err != 0 || res == NULL) {
	        ESP_LOGE(TAG, "DNS lookup failed err=%d res=%p", err, res);
	        return;
	    } else {
	        ESP_LOGI(TAG, "DNS lookup success");
	    }

	    //struct in_addr addr4 = &((struct sockaddr_in *)res->ai_addr)->sin_addr;

	    //struct in_addr addr4 = &((struct sockaddr_in *)res->ai_addr)->sin_addr;
	    struct in_addr addr4 = ((struct sockaddr_in *) (res->ai_addr))->sin_addr;

	    inet_addr_to_ip4addr(ip_2_ip4(&target_addr), &addr4);
	    freeaddrinfo(res);
	    ESP_LOGI(TAG, "target_addr.type=%d", target_addr.type);
	    ESP_LOGI(TAG, "target_addr=%s", ip4addr_ntoa(&(target_addr.u_addr.ip4)));

	    esp_ping_config_t ping_config = ESP_PING_DEFAULT_CONFIG();
	    ping_config.target_addr = target_addr;          // target IP address
	    ping_config.count = ESP_PING_COUNT_INFINITE;    // ping in infinite mode, esp_ping_stop can stop it

	    /* set callback functions */
	    esp_ping_callbacks_t cbs = {
	        .on_ping_success = cmd_ping_on_ping_success,
	        .on_ping_timeout = cmd_ping_on_ping_timeout,
	        .on_ping_end = cmd_ping_on_ping_end,
	        .cb_args = NULL
	    };
	    esp_ping_handle_t ping;
	    esp_ping_new_session(&ping_config, &cbs, &ping);
	    esp_ping_start(ping);
	}

I got this error:

	I (4181) wifi station: DNS lookup success
	I (4181) wifi station: target_addr.type=80
	I (4181) wifi station: target_addr=119.9.92.99
	lwip_sendto: invalid address
	E (4191) ping_sock: send error=0
	From 119.9.92.99 icmp_seq=1 timeout
	lwip_sendto: invalid address
	E (5191) ping_sock: send error=0
	From 119.9.92.99 icmp_seq=2 timeout
	lwip_sendto: invalid address
	E (6191) ping_sock: send error=0
	From 119.9.92.99 icmp_seq=3 timeout
	lwip_sendto: invalid address
	E (7191) ping_sock: send error=0

suda-morris, 2020-01-20:
Thanks @nopnop2002 for your detailed experiment. We will fix the broken link ASAP.

closed 2020-01-23 by espressif-bot