espressif/esp-idf #4649: Impact of CONFIG_MBEDTLS_SSL_IN_CONTENT_LEN

AlpinProgrammer, 2020-01-20:
## Environment

- Development Kit:      ESP32-DevKitC
- Module or chip used:  ESP32-WROOM-32
- IDF version: v3.3 stable
- Build System:         Make
- Compiler version: crosstool-ng-1.22.0-80-g6c4433a - 5.2.0
- Operating System:    Linux
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

Hello,
I encounter a problem while performing an OTA.

Allocating 16384 for CONFIG_MBEDTLS_SSL_IN_CONTENT_LEN in the sdkconfig works. However, any value lower than 16384 seems not working. I tried values from 8192 to 16384.

The SSL certificate we use has 1260 characters and we are trying to perform an OTA on a 0.8MB firmware and I cannot figure out how CONFIG_MBEDTLS_SSL_IN_CONTENT_LEN impacts the procedure.

I attached our sdkconfig. We chose to enable CONFIG_MBEDTLS_ASYMMETRIC_CONTENT_LEN
[sdkconfig_esp_issue.txt](https://github.com/espressif/esp-idf/files/4085700/sdkconfig_esp_issue.txt)

### Expected Behavior
The OTA is performed successfully

### Actual Behavior
The function "esp_http_client_fetch_headers" below returns MBEDTLS_ERR_SSL_INVALID_RECORD.

### Code to reproduce this issue

```
...
    esp_http_client_config_t config = {
            .url = url,
            .cert_pem = (char *)server_cert_str,
            .event_handler = _http_event_handler,
    };

    esp_http_client_handle_t client = esp_http_client_init(config);
    esp_err_t err = esp_http_client_open(client, 0);
    esp_http_client_fetch_headers(client);
...
```

Thanks in advance for your help.

mahavirj, 2020-01-20:
@AlpinProgrammer It is not possible to customize content length for `in` buffer unless you have some sort of control over server configuration (or sure that it won't send message longer than size of this buffer). Hence default values are kept as 16K and 4K (for `in` and `out` buffers) when asymmetric content length feature is enabled, since for `out` buffer we can control this from device/client side.

Here is relevant information, https://github.com/espressif/esp-idf/blob/647cb628a11a9ebd6952276fb7e6f354d4e3171d/components/mbedtls/Kconfig#L48-L57

AlpinProgrammer, 2020-01-20:
Thank you very much for your fast answer. I can close this issue.
Regards.

closed 2020-01-20 by AlpinProgrammer