espressif/esp-idf #4650: DHCPS lease enable== false on default wifi ap

no1seman, 2020-01-20:
## Environment
- Development Kit:  ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v4.1
- Module or chip used: ESP32-D0WD
- IDF version: v4.1-dev-1935-g647cb628a
- Build System: CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Linux
- Power Supply:  USB|external 5V&Battery
## Problem Description

Running SoftAP example got dhcps_lease.enable == false (see the code below). Why?

### Expected Behavior
dhcps_lease.enable == true

### Actual Behavior
dhcps_lease.enable == false
### Steps to reproduce

1. Take an SoftAP example
2. Modify by adding code below into main_app()

### Code to reproduce this issue

```cpp
void app_main(void)
{
...
    vTaskDelay(pdMS_TO_TICKS(20 * 1000));
    dhcps_lease_t dhcps_lease;
    ESP_ERROR_CHECK(esp_netif_dhcps_option(esp_netif_wifi_ap, ESP_NETIF_OP_GET, ESP_NETIF_REQUESTED_IP_ADDRESS, &dhcps_lease, sizeof(dhcps_lease)));
    ESP_LOGI(TAG, "dhcps_lease.enable: %s", dhcps_lease.enable?"True":"False");
}
```

## Debug Logs

```
entry 0x400806f8
I (28) boot: ESP-IDF v4.1-dev-1935-g647cb628a 2nd stage bootloader
I (29) boot: compile time 14:48:05
I (29) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (49) boot.esp32: SPI Speed      : 40MHz
I (49) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x17930 ( 96560) map
I (147) esp_image: segment 1: paddr=0x00027958 vaddr=0x3ffb0000 size=0x0389c ( 14492) load
I (154) esp_image: segment 2: paddr=0x0002b1fc vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /home/user/esp/esp-mdf/esp-idf/components/freertos/xtensa_vectors.S:1778

I (155) esp_image: segment 3: paddr=0x0002b608 vaddr=0x40080404 size=0x04a10 ( 18960) load
I (171) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x74dc0 (478656) map
0x400d0020: _stext at ??:?

I (345) esp_image: segment 5: paddr=0x000a4de8 vaddr=0x40084e14 size=0x10ae0 ( 68320) load
0x40084e14: wDev_ProcessFiq at ??:?

I (387) boot: Loaded app from partition at offset 0x10000
I (387) boot: Disabling RNG early entropy source...
I (388) cpu_start: Pro cpu up.
I (391) cpu_start: Application information:
I (396) cpu_start: Project name:     wifi_softAP
I (401) cpu_start: App version:      1
I (406) cpu_start: Compile time:     Jan 20 2020 14:47:57
I (412) cpu_start: ELF file SHA256:  cb253be069c0f263...
I (418) cpu_start: ESP-IDF:          v4.1-dev-1935-g647cb628a
I (424) cpu_start: Starting app cpu, entry point is 0x40081280
0x40081280: call_start_cpu1 at /home/user/esp/esp-mdf/esp-idf/components/esp32/cpu_start.c:275

I (0) cpu_start: App cpu up.
I (435) heap_init: Initializing. RAM available for dynamic allocation:
I (442) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (448) heap_init: At 3FFB9178 len 00026E88 (155 KiB): DRAM
I (454) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (460) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (467) heap_init: At 400958F4 len 0000A70C (41 KiB): IRAM
I (473) cpu_start: Pro cpu start user code
I (491) spi_flash: detected chip: generic
I (492) spi_flash: flash io: dio
W (501) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (504) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (578) wifi softAP: ESP_WIFI_MODE_AP
I (598) wifi: wifi driver task: 3ffc0eec, prio:23, stack:3584, core=0
I (598) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (598) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (628) wifi: wifi firmware version: 0849c77
I (628) wifi: config NVS flash: enabled
I (628) wifi: config nano formating: disabled
I (628) wifi: Init dynamic tx buffer num: 32
I (628) wifi: Init data frame dynamic rx buffer num: 32
I (638) wifi: Init management frame dynamic rx buffer num: 32
I (638) wifi: Init management short buffer num: 32
I (648) wifi: Init static rx buffer size: 1600
I (648) wifi: Init static rx buffer num: 10
I (658) wifi: Init dynamic rx buffer num: 32
I (758) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (758) wifi: mode : softAP (b4:e6:2d:ee:83:f6)
I (758) wifi: Total power save buffer number: 16
I (758) wifi: Init max length of beacon: 752/752
I (768) wifi: Init max length of beacon: 752/752
I (768) wifi softAP: wifi_init_softap finished. SSID:myssid password:mypassword
I (20778) wifi softAP: dhcps_lease.enable: **False** <-????

```

freakyxue, 2020-04-16:
hi @no1seman 
I found root cause :
In dhcps_poll_set function：
This function does not set this dhcps_lease.enable to enable when setting the dhcps address。

You can do this：
[image]

But this variable has no effect on dhcps。

freakyxue, 2020-04-21:
hi @no1seman
There are still problems with this. I will close this problem. If you have any questions in the future, please reopen it

closed 2020-08-19 by Alvin1Zhang