espressif/esp-idf #4652: WiFi Channel switches makes WiFi unavailable

Noel93, 2020-01-21:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-WROVER]
- IDF version (run ``git describe --tags`` to find it):
    // v4.1-dev-281-g96b96ae24
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

While in AP mode, when switching WiFi channels with esp_wifi_set_channel to a primary channel other than 1, the WiFi becomes unavailable (it becomes a hidden network). Any connected station will disconnect and fail to reconnect. Disconnecting all stations via esp_wifi_deauth_sta(0) before switching channels isn't changing the outcome.

### Expected Behavior

Ideally, the stations would stay connected and just get a Channel Switch Announcement. At least, the station should be able to reconnect on the new channel.

Noel93, 2020-02-03:
Morning,
could you make any comment on this? I'm not too sure if it's a bug or I'm simply misunderstanding the functionality and/or limits of the channel switch API.

Alvin1Zhang, 2020-02-04:
Thanks for reporting and sorry for the slow turnaround. Sorry that we are now during the holiday of Chinese Lunar New Year. Will update to you when we are back, thanks. @Noel93

sagb2015, 2020-02-04:
@Noel93 _esp_wifi_set_channel_ API is supposed to be used in Sniffer mode only. For setting AP channel please use _esp_wifi_set_config_ along with esp_wifi_set_bandwidth API. For changing the primary channel, you will have to call  _esp_wifi_stop_ and _esp_wifi_start_ again.  Also, we do not support sniffer + AP mode.  So, "add lines to softAP to set promiscuous mode = true ", may not work as you are intending to.

Noel93, 2020-02-07:
@sagb2015 thanks, I got it working.
This is another topic, but do you coincidentally know of a way to use a fixed transmission power? There's set_max_tx_power, but this is just setting the maximum to my understanding. Is it possible to set the tx_power to a fixed value instead?

sagb2015, 2020-02-11:
@Noel93 - No, we do not have such API. However, wifi will try to use power-level just below the max. So you can achieve desired results using _esp_wifi_set_max_tx_power_ and setting fixed rate _esp_wifi_internal_set_fix_rate_.

sagb2015, 2020-02-11:
Closing the original issue. We can continue discussing here for any related queries.

danielgroening, 2021-01-19:
> @Noel93 _esp_wifi_set_channel_ API is supposed to be used in Sniffer mode only. For setting AP channel please use _esp_wifi_set_config_ along with _esp_wifi_set_bandwidth_ API. For changing the [...]

Somebody can confirm that this is still the case (in version 4.1) and there are no plans in changing this (maybe it isn't even possible due to (software) restrictions in non-promiscous mode?
The API docs  (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv420esp_wifi_set_channel7uint8_t18wifi_second_chan_t) don't mention any restrictions at all
+
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#home-channel says "Switching channel in SoftAP is working, even with Channel Switch Announcement" (which is not true?"

sagb2015, 2021-02-16:
@danielgroening  You are right that we support `esp_wifi_set_channel` in softAP mode as stated in the document, but only when no stations are connected. We think it is a valid requirement to have an API to switch the channel using CSA mechanism provided by the protocol. We will take that as a feature request.

closed 2022-01-05 by jack0c