espressif/esp-idf #4654: Unable to read microphone input from ADC2_6 channel using I2S.

ankur2854, 2020-01-22:
I am using ADC2_6 channel to read microphone input using I2S. But from the error log we get to know I2S not support ADC2 unit.
Can anyone help me on it. Is there any way to do read microphone input from ADC2 channels? 

Using SDK v4.0
Module - esp32 wroom32D
Microphone connect to physical pin 17.

Thanks in advance.

koobest, 2020-04-15:
Hi, 
Because ADC2 is used by WIFI, so, we do not provide the support.  The attachment may provide some reference for you.

[i2s_adc_scale.tar.gz](https://github.com/espressif/esp-idf/files/4478960/i2s_adc_scale.tar.gz)

thanks !!

Alvin1Zhang, 2020-05-25:
@ankur2854 Thanks for reporting, free free to reopen if any updates. Thanks.

closed 2020-05-25 by Alvin1Zhang