espressif/esp-idf #4655: ESP32 idf/eclipse debug nightmare

Mat-Alm, 2020-01-22:
idf version used v4.1-dev-1931-g68be5f6ca and additional eclipse last plugins provided by vendor.
debug interface interface jlink openocd version: v0.10.0-esp32-20190313 (the most stable version for me)

1º Problem 

Open any idf example project on eclipse. Disable booloader verbose log, set  compiler optimization to -O0 (no optimization) and build the project. Load the executable with with serial bootloader. 
Result -> the device keep restarting!
Do the same with -Og optimization and the main application runs fine!

Now comes the odd things. If i start an openocd section (just succesfull connection) with debugger conected to the board it runs the application just ok. But wait you can claim that debug interface is interfering with boot strap right! But not... If I just flash (by serial) other board without any other connection than usb cable it also keep restarting with -O0 optmization!

But here is a good thing you can debug fine the aplication with -O0 settings if you first upload the bootloader and partiton data by serial. (Yes I'am flashing only application section at every new debug section). The previous bug still present by far...

So you will question.. Why you need -O0 optimization? Because with  -Og (or any other debug level rather than -O0) you get a really bad debug experience... Single step is simple useless.

2º Problem: Main app offset and debug. 

So you want to have booloader verbose turned on to see what is going on right? So you need to increase bootloader section. To do this you need to start the main_app at some address different of 0x10000. So let increase the boot section to 0x10000 and start the main_app at 0x20000.

And here we start another kind of troubles. 
If you try to debug your project with this settings you can't set any breakpoints at functions that reside in flash region. If the function is in a ram section it's work with the addtion with previous bugs mentioned. The debugger flash upload region main app is adjusted as needed.

Here is the log with default (0x10000) address app_main :

```
Open On-Chip Debugger  v0.10.0-esp32-20190313 (2019-03-13-09:57)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 10000 kHz
Info : Configured 2 cores
esp32 interrupt mask on
Started by GNU MCU Eclipse
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : J-Link ARM / Flasher ARM V4 compiled Oct  6 2017 16:42:52
Info : Hardware version: 4.00
Info : VTarget = 3.282 V
Info : clock speed 10000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : Target halted. PRO_CPU: PC=0x400D58FA (active)    APP_CPU: PC=0x400EDE58 
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Error: No symbols for FreeRTOS
Info : Target halted. PRO_CPU: PC=0x4009171A (active)    APP_CPU: PC=0x400EDE58 
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 30 KB
Info : Flash mapping 1: 0x20020 -> 0x400d0020, 121 KB
Info : Target halted. PRO_CPU: PC=0x4009171A (active)    APP_CPU: PC=0x400EDE58 
Info : Auto-detected flash size 4096 KB
Info : Using flash size 4096 KB
Info : Target halted. PRO_CPU: PC=0x4009171A (active)    APP_CPU: PC=0x400EDE58 
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 30 KB
Info : Flash mapping 1: 0x20020 -> 0x400d0020, 121 KB
Info : Using flash size 124 KB
Info : Target halted. PRO_CPU: PC=0x4009171A (active)    APP_CPU: PC=0x400EDE58 
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 30 KB
Info : Flash mapping 1: 0x20020 -> 0x400d0020, 121 KB
Info : Using flash size 32 KB
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32: Debug controller 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : Target halted. PRO_CPU: PC=0x5000004B (active)    APP_CPU: PC=0x00000000 
Info : esp32: Core 0 was reset (pwrstat=0x1F, after clear 0x0F).
Info : esp32: Debug controller 1 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core 1 was reset (pwrstat=0x5F, after clear 0x0F).
Info : Target halted. PRO_CPU: PC=0x40000400 (active)    APP_CPU: PC=0x40000400 
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
[... 78 lines cut ...]
Info : Target halted. PRO_CPU: PC=0x40000400 (active)    APP_CPU: PC=0x40000400 
** Programming Started **
auto erase enabled
Info : Target halted. PRO_CPU: PC=0x4009171A (active)    APP_CPU: PC=0x40000400 
Info : Target halted. PRO_CPU: PC=0x4009171A (active)    APP_CPU: PC=0x40000400 
wrote 229376 bytes from file C:/Users/ryzen/eclipse-workspace1/test/build/app-template.bin in 5.247182s (42.690 KiB/s)
** Programming Finished **
** Verify Started **
Info : Target halted. PRO_CPU: PC=0x4009171A (active)    APP_CPU: PC=0x40000400 
read 228960 bytes from file C:/Users/ryzen/eclipse-workspace1/test/build/app-template.bin and flash bank 0 at offset 0x00020000 in 2.831826s (78.957 KiB/s)
contents match
** Verified OK **

```
If try put put any break point on functions that are in flash section and try to run/step you will get:

Info : Target halted. PRO_CPU: PC=0x400D58FA (active)    APP_CPU: PC=0x400EDE58 
Error: esp32: Failed to read insn (-4)!
Error: esp32: Failed to add SW BP!
Error: can't add breakpoint: resource not available
Error: esp32: Failed to read insn (-4)!
Error: esp32: Failed to add SW BP!
Error: can't add breakpoint: resource not available

Should I giveup to ESP32? I waste more than 5 days to see why this is not working.. Should I spend more time on this Soc?

igrr, 2020-01-23:
@Mat-Alm thanks for the report and sorry for the trouble you had to go through. For the first issue, it looks like there is some regression in -O0 builds. Will investigate and report back.

For the 2nd issue, have you tried telling OpenOCD about the application binary image offset? It can be done when running OpenOCD:

```
openocd -f board/esp32-wrover-kit-3.3v.cfg -c "init; halt; esp32 appimage_offset 0x210000"
```

(replace the cfg file name with the one you use)

You are right that modifying application binary offset and then debugging afterwards is not ergonomic. We will try to improve this workflow in the IDE plugins.

Lisa999, 2020-07-04:
> @Mat-Alm thanks for the report and sorry for the trouble you had to go through. For the first issue, it looks like there is some regression in -O0 builds. Will investigate and report back.
>
> For the [...]

openocd -f board/esp32-wrover-kit-3.3v.cfg -c "init; halt; esp32 appimage_offset 0x20000"
Skip the 1 in the 0x210000

No i get the next error: Error: FreeRTOS: uxTopUsedPriority is not defined, consult the OpenOCD manual for a work-around

Pffff, esp-idf debugging is indeed a nightmare... ;-(

ashigupta, 2021-05-24:
Any solution for this issue. I am also facing a similar issue, see logs below 
[image]

Debugging worked once and then its just not going in debug mode.