espressif/esp-idf #4656: Can't get core dump from flash - error "Invalid core dump CRC d59a4f1e, should be ffffffff"

YuriK75, 2020-01-22:
#4100  Environment

- Development Kit: none
- Module or chip used:  ESP32-WROOM-32
- IDF version 3.3.1
- Build System:    idf.py
- Compiler version:   1.22.0-80-g6c4433a5
- Operating System: Windows
- (Windows only) environment type: ESP Command Prompt.
- Using an IDE?: No
- Power Supply:  external 3.3V

## Problem Description

I have ESP32D0WDQ6 on my custom board.
I modified hello_world example for making abort in task watchdog
and trying to get core dump from flash.

In menuconfig i set options for writing core dump to flash and for abort when
task watchdog timer triggers. I use custom partition table with coredump at 0x110000
(it is default address in espcoredump.py).

In main() i started task with watchdog and then delay for 10000 ms.
Task watchdog triggers, abort() called, dump writed to flash.

### Expected Behavior
Useful espcoredump.py output.

### Actual Behavior

espcoredump.py returns error "Invalid core dump CRC d59a4f1e, should be ffffffff"

### Steps to reproduce

1. Copy helloworld example
2. Replace sdkconfig and hello_world_main.c, create custom mypartitions.csv as attached.

### Code to reproduce this issue

```cpp
1. hello_world_main.c:

    #include <stdio.h>
    #include "freertos/FreeRTOS.h"
    #include "freertos/task.h"
    #include "esp_system.h"
    #include "esp_task_wdt.h"
     
    void myRoutine(void *arg)
    {
        printf("myRoutine start\n");
        esp_task_wdt_add(NULL);
        vTaskDelay(10000);
    }
     
    void app_main()
    {
        printf("Hello world!\n");
       
        esp_task_wdt_add(NULL);
     
        xTaskCreatePinnedToCore(myRoutine,   // функция потока
                  "myRoutine",     // текстовая метка
                  10000,           // размер стека в СЛОВАХ (2 байта)
                  NULL,          // входной аргумент
                  1,             // приоритет
                  NULL, 0);   // handle
     
        while(true)
        {
[... 353 lines cut ...]
    esptool.py v2.8
    Serial port COM3
    Connecting........_
    Chip is ESP32D0WDQ6 (revision 1)
    Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
    Crystal is 40MHz
    MAC: 30:ae:a4:f7:a5:34
    Uploading stub...
    Running stub...
    Stub running...
    5668 (100 %)
    5668 (100 %)
    Read 5668 bytes at 0x110000 in 0.5 seconds (85.3 kbit/s)...
    Hard resetting via RTS pin...
     
     
    A fatal error occurred: Invalid core dump CRC d59a4f1e, should be ffffffff

```

## Other items if possible

BIN, ELF, sdkconfig, hello_world_main.c ... etc attached in archive.

[esp-idf-issue-Invalid-core-dump-CRC.zip](https://github.com/espressif/esp-idf/files/4097705/esp-idf-issue-Invalid-core-dump-CRC.zip)

pumelo, 2020-05-26:
I'm having exactly the same issue. Did you find a way to resolve this?

I guess the key to this is
`E (5249) esp_core_dump_common: Skipped 1 tasks with bad TCB!`
This seems to make the core dump unusable ..
This does not happen if the reset was not triggered by a watchdog reset ...

YuriK75, 2020-05-26:
Hi!

No, I didn't... It is now not so important for me, my program is now
stable. But i am still interesting about it!

gerekon, 2021-03-09:
There is a coredump bug in IDF v3.3. Here is the patch to fix it. We will backport that fix to 3.3. 
```patch
diff --git a/components/espcoredump/src/core_dump_port.c b/components/espcoredump/src/core_dump_port.c
index e045b19a4e..fbbdc22a48 100644
--- a/components/espcoredump/src/core_dump_port.c
+++ b/components/espcoredump/src/core_dump_port.c
@@ -47,8 +47,11 @@ bool esp_core_dump_process_tcb(void *frame, core_dump_task_header_t *task_snapho
         return false;
     }
     if (task_snaphort->tcb_addr == xTaskGetCurrentTaskHandleForCPU(xPortGetCoreID())) {
-        // Set correct stack top for current task
-        task_snaphort->stack_start = (uint32_t)exc_frame;
+        // Set correct stack top for current task; only modify if we came from the task,
+        // and not an ISR that crashed.
+        if (!xPortInterruptedFromISRContext()) {
+            task_snaphort->stack_start = (uint32_t)exc_frame;
+        }
         // This field is not initialized for crashed task, but stack frame has the structure of interrupt one,
         // so make workaround to allow espcoredump to parse it properly.
         if (exc_frame->exit == 0)
```

igrr, 2022-02-24:
This issue has been fixed in 4.x releases. I'm closing this issue since IDF 3.3 has reached EoL.

closed 2022-02-24 by igrr