espressif/esp-idf #4657: ota over ssl failed to verify ssl certificate  esp-tls: mbedtls_ssl_handshake returned -0x2700

umer-ilyas, 2020-01-23:
## Environment

- Development Kit:      [custom]
- Kit version 
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it):
    // v4.0-rc
- Build System:         [idf.py]
- Compiler version  xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     [Windows]
- (Windows only) environment type: Plain Command Prompt
- Using an IDE?: [MScode]
- Power Supply:         Battery

## Problem Description

ota over ssl failed to verify ssl certificate 9 out of 10 times and through the following error 
 `esp-tls: mbedtls_ssl_handshake returned -0x2700`

> #define MBEDTLS_ERR_X509_CERT_VERIFY_FAILED               -0x2700  /**< Certificate verification failed, e.g. CRL, CA or signature check failed. */

this code is based on the simple ota example and it was working in v3.3 and when updated it v4.0-rc somehow it got some error. (certificate is same  ) 
although certificate is signed by a CA but i am not setting any  global CA and  `.use_global_ca_store = false,`  but this error is also gets printed 
```
I (19896) esp-tls: Failed to verify peer certificate!
I (19896) esp-tls: verification info:   ! The certificate is not correctly signed by the trusted CA
```
certificate is attached 

[blob_windows_net.pem.txt](https://github.com/espressif/esp-idf/files/4101711/blob_windows_net.pem.txt)

umer-ilyas, 2020-01-23:
update:
now if init global ca store before calling the ota update function 
```
        err = esp_tls_init_global_ca_store();
        err = esp_tls_set_global_ca_store(ca_pem_start, ca__pem_end -ca_ _pem_start);
```
`and use        .use_global_ca_store = true,` it is working fine every time

shubhamkulkarni97, 2020-02-06:
Hi @umer-ilyas 
I tried using the certificate provided by you, as I did not have the URL you are connecting to, I encountered same error. This usually occurs when the URL you are connecting to does not match with the Common Name Embedded in the certificate.

I see you're trying to connect to Azure using Intermediate Certificate. I have attached a leaf certificate for Azure. Can you try using this certificate?
```
-----BEGIN CERTIFICATE-----
MIIIszCCBpugAwIBAgITLQAHLHrGsBsrwo6qLAAAAAcsejANBgkqhkiG9w0BAQsF
ADCBizELMAkGA1UEBhMCVVMxEzARBgNVBAgTCldhc2hpbmd0b24xEDAOBgNVBAcT
B1JlZG1vbmQxHjAcBgNVBAoTFU1pY3Jvc29mdCBDb3Jwb3JhdGlvbjEVMBMGA1UE
CxMMTWljcm9zb2Z0IElUMR4wHAYDVQQDExVNaWNyb3NvZnQgSVQgVExTIENBIDUw
HhcNMTkwNzI0MjEzMzUyWhcNMjAwNzI0MjEzMzUyWjAeMRwwGgYDVQQDExNhenVy
ZS5taWNyb3NvZnQuY29tMIIBIjANBgkqhkiG9w0BAQEFAAOCAQ8AMIIBCgKCAQEA
w7Q/Vu9b4DuTY5vb+csSouH9bh1XnzxBtWw7zljF/2Rkcq98Eie2hsKIC6B3AaQY
bTxjKQ7qtOD8pLym1gaSH/57zTo25sg8+Quff3JmJtemwctNeAW3vW9JnlQuL1N0
FvSqQltFjJNIAg3AHP5lIuVuTSmKZGmsPY+oNnMuDEaHG2vnt45yzcrosDNWA6ra
kcITiIpwSqpUDrcRi7CxMDtWi/AH4qPmHc4XWfIVHgvQIKFM96moupFcrisXZkDW
M2idhwD1i4IGibzrSpSmXiStMu04LSUCwvBaU84/HTjVznPHhdWw6ji0YPjMgSjB
DsClWisZ2l7VsLu0GtipTQIDAQABo4IEejCCBHYwggH3BgorBgEEAdZ5AgQCBIIB
5wSCAeMB4QB3ALvZ37wfinG1k5Qjl6qSe0c4V5UKq1LoGpCWZDaOHtGFAAABbCXw
2ckAAAQDAEgwRgIhAIG4MRBryTc4UbArJzYPct0b1Ta0KRkKbMDYOr3FWo16AiEA
/axhWaIQeYf9D+5yfZQIxotMdKDjOgoq6CY0Qrhw6iYAdgDwlaRZ8gDRgkAQLS+T
iI6tS/4dR+OZ4dA0prCoqo6ycwAAAWwl8NkOAAAEAwBHMEUCIQCmtiox2r+wIMZW
TQk773VI6B6CYlvGaItieccUSyYY1AIgdonicFpbmRT42llJlTU+ZulYIWNjmT4C
wHdzDDZd6dsAdgCyHgXMi6LNiiBOh2b5K7mKJSBna9r6cOeySVMt74uQXgAAAWwl
8NnLAAAEAwBHMEUCIQDPKThpUwMboqpeaqMIq2ndGRG4QG2zOW08PoK7IoSqDgIg
XSoG6NurhyrKwGXeZW7mUswjHSoA4OX7PiOmcOeDfg0AdgBVgdTCFpA2AUrqC5tX
PFPwwOQ4eHAlCBcvo6odBxPTDAAAAWwl8Ni5AAAEAwBHMEUCIFXsbZy88x0P7XMB
4zsloZ6IOAe0eXzEROTw1/rwkeXXAiEAngS7r8LwotAP4GKuCDRJ/IGvqVvaCna4
965CDD5c/cYwJwYJKwYBBAGCNxUKBBowGDAKBggrBgEFBQcDAjAKBggrBgEFBQcD
ATA+BgkrBgEEAYI3FQcEMTAvBicrBgEEAYI3FQiH2oZ1g+7ZAYLJhRuBtZ5hhfTr
YIFdhNLfQoLnk3oCAWQCAR0wgYUGCCsGAQUFBwEBBHkwdzBRBggrBgEFBQcwAoZF
aHR0cDovL3d3dy5taWNyb3NvZnQuY29tL3BraS9tc2NvcnAvTWljcm9zb2Z0JTIw
SVQlMjBUTFMlMjBDQSUyMDUuY3J0MCIGCCsGAQUFBzABhhZodHRwOi8vb2NzcC5t
c29jc3AuY29tMB0GA1UdDgQWBBQVPnrwEq/O7zunkd0ThS8tJcYb4DALBgNVHQ8E
BAMCBLAwHgYDVR0RBBcwFYITYXp1cmUubWljcm9zb2Z0LmNvbTCBrAYDVR0fBIGk
MIGhMIGeoIGboIGYhktodHRwOi8vbXNjcmwubWljcm9zb2Z0LmNvbS9wa2kvbXNj
b3JwL2NybC9NaWNyb3NvZnQlMjBJVCUyMFRMUyUyMENBJTIwNS5jcmyGSWh0dHA6
Ly9jcmwubWljcm9zb2Z0LmNvbS9wa2kvbXNjb3JwL2NybC9NaWNyb3NvZnQlMjBJ
VCUyMFRMUyUyMENBJTIwNS5jcmwwTQYDVR0gBEYwRDBCBgkrBgEEAYI3KgEwNTAz
BggrBgEFBQcCARYnaHR0cDovL3d3dy5taWNyb3NvZnQuY29tL3BraS9tc2NvcnAv
Y3BzMB8GA1UdIwQYMBaAFAj+JZ906ocEwry7jqg4XzPG0WxlMB0GA1UdJQQWMBQG
CCsGAQUFBwMCBggrBgEFBQcDATANBgkqhkiG9w0BAQsFAAOCAgEAkwWXqx+BonQ+
QnwT9zueY/50aab1xu6/+RvAGWiMoSaMvzp6Eej3KkaqLmSE9xT3YtvudpFl4W3I
1+Vio2rMptq6g5xOOilLvzNn6dhCmwbp+YAC9ZNZrIg9Ku9UivP7Bo9KRfWixpeS
rwOH7tHzDsITrCQPPk6RFDFvUTKt09foMcy04pyPV35eTVGY9lSxGVWlOy0iO8cV
haXJX3K8ICXDV0s9nIn+tsc63rT2TmGh9GP/audw6w6eEZN3nceSMCoIrtagsuME
8jORpGxES/7MeJCixT+WY3PnPKX6K0JhMuwS5IaDFPcy8Igdx9oUy8ENDVGyTvq5
YL+WokdCSPflwvkxcGPVeGlVyyxt+IIROswpxqfVjm1VPA55tEnt+EEPyrH4Sj6O
f4dPX5T9Z/V/7fhQ7jF1Ka0OPBDpPrSDtrTYU3IZiiRnTQHm/C4L5xD41jo9NMh7
psUdAD4eFMgvZHgHy5b++G3SVZVUWfAzS5xKpu9i2bMZ0xfZvLt39FPOAkIcmeaM
gKFjv03zHtutLKVzODXwR3j6XxqJWK7cITbwKE9FnR9PZBzBMpj7J2PY5ktyXNwr
mNJEA/Fi57G4q+SJLSu/a2caGISTAjKq3gKbD3t0U4/RasuhYi9Yle7rFer/ovhd
Hv0II9FnmCeuM7BCVun7rdiXDfY/pxM=
-----END CERTIFICATE-----
```

umer-ilyas, 2020-02-07:
thanks i'll check

plus FYI I used this flag   `.skip_cert_common_name_check = true,` for above certificate

shubhamkulkarni97, 2020-02-10:
@umer-ilyas 
Do you have any updates on the issue?

umer-ilyas, 2020-02-18:
used this certificate and issue resolved 

```
-----BEGIN CERTIFICATE-----
MIIFtDCCBJygAwIBAgIQC2qzsD6xqfbEYJJqqM3+szANBgkqhkiG9w0BAQsFADBa
MQswCQYDVQQGEwJJRTESMBAGA1UEChMJQmFsdGltb3JlMRMwEQYDVQQLEwpDeWJl
clRydXN0MSIwIAYDVQQDExlCYWx0aW1vcmUgQ3liZXJUcnVzdCBSb290MB4XDTE2
MDUyMDEyNTIzOFoXDTI0MDUyMDEyNTIzOFowgYsxCzAJBgNVBAYTAlVTMRMwEQYD
VQQIEwpXYXNoaW5ndG9uMRAwDgYDVQQHEwdSZWRtb25kMR4wHAYDVQQKExVNaWNy
b3NvZnQgQ29ycG9yYXRpb24xFTATBgNVBAsTDE1pY3Jvc29mdCBJVDEeMBwGA1UE
AxMVTWljcm9zb2Z0IElUIFRMUyBDQSA0MIICIjANBgkqhkiG9w0BAQEFAAOCAg8A
MIICCgKCAgEAq+XrXaNrOZ71NIgSux1SJl19CQvGeY6rtw7fGbLd7g/27vRW5Ebi
kg/iZwvjHHGk1EFztMuZFo6/d32wrx5s7XEuwwh3Sl6Sruxa0EiB0MXpoPV6jx6N
XtOtksDaxpE1MSC5OQTNECo8lx0AnpkYGAnPS5fkyfwA8AxanTboskDBSqyEKKo9
Rhgrp4qs9K9LqH5JQsdiIMDmpztd65Afu4rYnJDjOrFswpTOPjJry3GzQS65xeFd
2FkngvvhSA1+6ATx+QEnQfqUWn3FMLu2utcRm4j6AcxuS5K5+Hg8y5xomhZmiNCT
sCqDLpcRHX6BIGHksLmbnG5TlZUixtm9dRC62XWMPD8d0Jb4M0V7ex9UM+VIl6cF
JKLb0dyVriAqfZaJSHuSetAksd5IEfdnPLTf+Fhg9U97NGjm/awmCLbzLEPbT8QW
0JsMcYexB2uG3Y+gsftm2tjL6fLwZeWO2BzqL7otZPFe0BtQsgyFSs87yC4qanWM
wK5c2enAfH182pzjvUqwYAeCK31dyBCvLmKM3Jr94dm5WUiXQhrDUIELH4Mia+Sb
vCkigv2AUVx1Xw41wt1/L3pnnz2OW4y7r530zAz7qB+dIcHz51IaXc4UV21QuEnu
sQsn0uJpJxJuxsAmPuekKxuLUzgG+hqHOuBLf5kWTlk9WWnxcadlZRsCAwEAAaOC
AUIwggE+MB0GA1UdDgQWBBR6e4zBz+egyhzUa/r74TPDDxqinTAfBgNVHSMEGDAW
gBTlnVkwgkdYzKz6CFQ2hns6tQRN8DASBgNVHRMBAf8ECDAGAQH/AgEAMA4GA1Ud
DwEB/wQEAwIBhjAnBgNVHSUEIDAeBggrBgEFBQcDAQYIKwYBBQUHAwIGCCsGAQUF
BwMJMDQGCCsGAQUFBwEBBCgwJjAkBggrBgEFBQcwAYYYaHR0cDovL29jc3AuZGln
aWNlcnQuY29tMDoGA1UdHwQzMDEwL6AtoCuGKWh0dHA6Ly9jcmwzLmRpZ2ljZXJ0
LmNvbS9PbW5pcm9vdDIwMjUuY3JsMD0GA1UdIAQ2MDQwMgYEVR0gADAqMCgGCCsG
AQUFBwIBFhxodHRwczovL3d3dy5kaWdpY2VydC5jb20vQ1BTMA0GCSqGSIb3DQEB
CwUAA4IBAQAR/nIGOiEKN27I9SkiAmKeRQ7t+gaf77+eJDUX/jmIsrsB4Xjf0YuX
/bd38YpyT0k66LMp13SH5LnzF2CHiJJVgr3ZfRNIfwaQOolm552W95XNYA/X4cr2
du76mzVIoZh90pMqT4EWx6iWu9El86ZvUNoAmyqo9DUA4/0sO+3lFZt/Fg/Hjsk2
IJTwHQG5ElBQmYHgKEIsjnj/7cae1eTK6aCqs0hPpF/kixj/EwItkBE2GGYoOiKa
3pXxWe6fbSoXdZNQwwUS1d5ktLa829d2Wf6l1uVW4f5GXDuK+OwO++8SkJHOIBKB
ujxS43/jQPQMQSBmhxjaMmng9tyPKPK9
-----END CERTIFICATE-----

```

closed 2020-02-18 by umer-ilyas