espressif/esp-idf #4658: kconfig path- slashes wrong

Emiel84, 2020-01-23:
I'm trying to build the AT-command set including classic bt support.
But when running a make the paths are wrong for some\all esd-idf references used.

IOError: C:/esp/esp-idf/Kconfig:59: Could not open 'C:espesp-idfcomponentsapp_updateKconfig.projbuild' (ENOENT: No such file or directory). Perhaps the $srctree environment variable (which was unset) is set incorrectly. Note that the current value of $srctree is saved when the Kconfig instance is created (for consistency and to cleanly separate instances). Also note that e.g. $FOO in a 'source' statement does not refer to the environment variable FOO, but rather to the Kconfig Symbol FOO (which would commonly have 'option env="FOO"' in its definition).
make: *** [/c/esp/esp-idf/components/esp32/Makefile.projbuild:51: /c/esp/esp-at/build/esp32/esp32.project.ld] Error 1

I'm using the 'released' branch of the AT-command set for ESP32 (Branch_Branch_v2.0.0.0_esp32).

During a investigation I did see the following in the auto.conf.cmd file.
(This file is generated\updated when I perform the steps to 'make' the AT library, therefore manually fixing it in this file didn't solve it)

deps_config := \
	/c/esp/esp-at/main/interface/sdio/Kconfig \
	/c/esp/esp-at/main/interface/socket/Kconfig \
	/c/esp/esp-at/main/interface/hspi/Kconfig \
	/c/esp/esp-at/main/interface/uart/Kconfig \
	C:\esp\esp-idf\components\app_trace\Kconfig \
	C:\esp\esp-idf\components\aws_iot\Kconfig \
	C:\esp\esp-idf\components\bt\Kconfig \
	C:\esp\esp-idf\components\driver\Kconfig \

It seems that somewhere the slashed are 'windows' formatted, but only for the esp-idf references.
Can someone point me into the direction to solve this issue ?

dobairoland, 2020-01-23:
Hi @Emiel84. I hope I got you right and you use this project: https://github.com/espressif/esp-at.
There seem to be a lot of similar issues reported there: https://github.com/espressif/esp-at/issues/232, https://github.com/espressif/esp-at/issues/230, https://github.com/espressif/esp-at/issues/203, https://github.com/espressif/esp-at/issues/157. Please take a look at those issues and see if they help you. Currently I know just a little about that project but reading through them I think maybe you have a wrong IDF version which is not supported by the AT project.

I assume you cannot reproduce the issue with the examples of ESP-IDF. Am I right?

Emiel84, 2020-01-23:
You are correct about the project. Will look at the links you are referring to.
I was using branch V3.3 as mentioned in the steps to allow compiling the ESP set.

You are correct about that I'm not able to reproduce the issue with the Hello world example.

Emiel84, 2020-01-23:
@dobairoland  

https://github.com/espressif/esp-at/issues/230  solved the issue.
Replaced the ldgen.mk file with the one in the ticked.

I expect it would help to mention\refer to this file in the 'step by step' for building the AT command set.

Emiel84, 2020-01-23:
One remaining question:
Once I create the library with BT classic enabled using a Make flash call.
The software is loaded into the ESP-WROOM-32 without any issues it seems.
But still the BTINIT? return an error, so it seems that it still isn;t enabled.

The Wroom was having ESP32-WROOM-32_AT_Bin_V2.0 loaded into it before this.
Currently rebuilding it again, following the steps from start for building the AT set, hope is works now.

I perform the make flash directly after I did a make menuconfig to set:

1. Enable BT classic
Component config -> AT -> [*] AT bt command support.

2. Set the UART port to align with ESP-WROOM-32
Component config -> EPS32-Specific-> [*] UART for console output -> Custom.
Component config -> EPS32-Specific-> [*] UART TX  17 
Component config -> EPS32-Specific-> [*] UART RX 16

3. Set the correct output (connected with the ESP32 programming port).
Serial Flashed config - /Dev/COM7

dobairoland, 2020-01-23:
Thanks @Emiel84 for the feedback. We will track down the issue and see if some fix should be added into IDF or not.

I suggest to open a new issue for your second (unrelated) problem for better tracking. Thanks!

xcguang, 2020-02-04:
Hi,
Please try it using the last master branch. This issue should be fixed on master branch.

closed 2020-08-19 by Alvin1Zhang