espressif/esp-idf #4659: Using the scan_rsp with BLE mesh

AlpinProgrammer, 2020-01-23:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Module or chip used:  ESP32-WROVER-B
- IDF version: v4.0-rc
- Build System:         Make
- Compiler version: 8.2.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

Hello,
In order to add proprietary information in the advertising, I need to use the scan response to add the MANUFACTURER_SPECIFIC_DATA in the advertising.

I already dug into esp-idf v4.0-rc. I could spot where the adv was set but I did not find any documentation related to the scan response.

Besides, when I try to merge gatt_server and ble_mesh_client_model examples, and I set the scan_rsp in the gatt_server part, it gets overwitten by the ble_mesh_init function (esp_ble_mesh_node_prov_enable(ESP_BLE_MESH_PROV_ADV | ESP_BLE_MESH_PROV_GATT); more accurately).

Thanks in advance for your help :)

### Expected Behavior
Using the scan response to add the manfuacturer specific data field (0xFF).

### Actual Behavior
The scan_rsp settings based on gatt_server example are cancelled by ble_mesh_init function.

### Code to reproduce this issue

The ble_mesh_client_model example code

WCCWCC, 2020-02-17:
Hi @AlpinProgrammer ，

* Do you want to use scan rsponse before provisioning or after provisioning?

* Currently  **ble mesh stack**  has no api to set scan response.

* Currently devices in **ble mesh stack** use scan response to display device names when they are not connected to the network.

AlpinProgrammer, 2020-02-17:
Hi @WCCWCC , thanks for your answer !

I'd like to use the scan_rsp most importantly before provisioning. Ideally, I'd like to use it at any time.

According to what I saw, using nRF Mesh for the provisioning and nRF Connect to check the data advertised, the scan response is used to display the device name at any time.

I was able to locate the piece of code (set_adv_data function) that sends the scan_rsp and to append the manufacturer specific data, checking that we didn't exceed the 31 byte limit.

Campou, 2020-08-10:
Hi @AlpinProgrammer 

For using `gatt_server` combined with BLE Mesh, recommended to enable gatt server firstly. After the gatt server is connected with the gatt client (e.g. a phone), then BLE Mesh can be initialized and enabled. Because currently if enabling gatt server and BLE Mesh at the same time, the advertising data or scan response data set by the gatt server will be overwritten by BLE Mesh, since BLE Mesh needs to advertising for most the time when it is not provisioned.

Another option is to enable the option `Support sending normal BLE advertising packets` in BLE Mesh menuconfig. With this, the device can support to interleave the BLE Mesh advertising packets and  normal BLE advertising packets.

Thanks.

chegewara, 2020-08-10:
I would like to share my experience here. Looks like i have more luck than experience, because i could enable ble advertising together with ble mesh which allows me to connect with app and perform OTA update over ble.
I didnt have chance to test this yet:
`Support sending normal BLE advertising packets`
but update takes ages (~18 min), because of update is paused every few packets for some time.

AlpinProgrammer, 2020-08-10:
Hi @Campou ,
Thank you for the additional information.
As you mentioned, BLE mesh initialization overwrites the modifications on scan_rsp, which I need to use at all times.
Therefore, I managed to find the files in esp-idf BLE mesh, where it's possible to edit the scan_rsp. This is how I solved my problem.
With all this information you provided I can close the issue. I'll make sure to check _Support sending normal BLE advertising packets_ of i encounter any problem in the future.

@chegewara , thank you for your having a look. I was also able to use BLE server with BLE mesh node simultaneously using both initializing procedures. Connection with a BLE scan app on a smartphone works well, as well as BLE mesh operations (generic onoff client). I didn't have to code OTA over BLE yet. So far so good :)

Alvin1Zhang, 2020-08-11:
Thanks for reporting and sharing the updates, glad to hear that the problem is resolved. Feel free to reopen if the issue still happens. Thanks.

closed 2020-08-11 by Alvin1Zhang

LeBellier, 2022-11-14:
Hi, 
Is there some update to change advertising data and rsp scan data with BLE mesh ?