espressif/esp-idf #4660: Bluetooth HFP

KollarRichard, 2020-01-23:
## Environment

- Development Kit :       ESP32-Wrover-Kit
- Kit version :                  v4.1
- Module or chip used : ESP32-WROVER-B]
- IDF version :                 v4.1-dev-1935-g647cb628a
- Build System :              Make
- Compiler version :       xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System :     macOS
- Using an IDE? :            No
- Power Supply :            USB

## Problem Description

After connecting to bluetooth headset with HFP_AG [example](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/classic_bt/hfp_ag) i am getting Guru Meditation Error. And ESP remains in loop with connecting and failing and rebooting.

### Expected Behavior

Succesfull conection and trying to run commands provided in example.

### Actual Behavior

After connecting example causes core panic.

### Steps to reproduce

1. Clone the hgp_ag example.
2. Change MAC adress of bluetooth headset in _app_hf_msg_set.c_ on line 16:
```cpp
// if you want to connect a specific device, add it's bda here
// esp_bd_addr_t hf_peer_addr = {0x70,0x26,0x05,0xca,0xeb,0x21};
```
3. Set the example in menuconfig to HCI and using mSBC.
4. Run the example and connect to bluetooth headset.

## Debug Logs

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40103533  PS      : 0x00060d30  A0      : 0x80103578  A1      : 0x3ffccd10  
0x40103533: bdcmp at /Users/richard.kollar/esp/esp-idf/components/bt/host/bluedroid/stack/include/stack/bt_types.h:756
 (inlined by) btc_hf_idx_by_bdaddr at /Users/richard.kollar/esp/esp-idf/components/bt/host/bluedroid/btc/profile/std/hf_ag/btc_hf_ag.c:123

A2      : 0x00000000  A3      : 0x3f80194c  A4      : 0x00000006  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000001  A9      : 0x3ffcccc0  
A10     : 0x00000016  A11     : 0x3ffcd100  A12     : 0x3ffccd10  A13     : 0x00000000  
A14     : 0x3ffb65fc  A15     : 0x00000001  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4008ff41  LEND    : 0x4008ff51  LCOUNT  : 0xffffffff  
0x4008ff41: strlen at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S:84

0x4008ff51: strlen at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S:96

ELF file SHA256: 94f28f538aede58ef2089b486ecfe4525230900e9e75ae9edc04e87494ca8f2a

Backtrace: 0x40103530:0x3ffccd10 0x40103575:0x3ffccd40 0x40103b09:0x3ffccd60 0x40104a81:0x3ffccfa0 0x400fd09d:0x3ffccfd0 0x4012605d:0x3ffccff0 0x4009586d:0x3ffcd010
0x40103530: bdcmp at /Users/richard.kollar/esp/esp-idf/components/bt/host/bluedroid/stack/include/stack/bt_types.h:756
 (inlined by) btc_hf_idx_by_bdaddr at /Users/richard.kollar/esp/esp-idf/components/bt/host/bluedroid/btc/profile/std/hf_ag/btc_hf_ag.c:123

0x40103575: is_connected at /Users/richard.kollar/esp/esp-idf/components/bt/host/bluedroid/btc/profile/std/hf_ag/btc_hf_ag.c:133

0x40103b09: btc_hf_unat_response at /Users/richard.kollar/esp/esp-idf/components/bt/host/bluedroid/btc/profile/std/hf_ag/btc_hf_ag.c:466

0x40104a81: btc_hf_call_handler at /Users/richard.kollar/esp/esp-idf/components/bt/host/bluedroid/btc/profile/std/hf_ag/btc_hf_ag.c:1070

0x400fd09d: btc_thread_handler at /Users/richard.kollar/esp/esp-idf/components/bt/common/btc/core/btc_task.c:155

0x4012605d: osi_thread_run at /Users/richard.kollar/esp/esp-idf/components/bt/common/osi/thread.c:68

0x4009586d: vPortTaskWrapper at /Users/richard.kollar/esp/esp-idf/components/freertos/port.c:143

```

## Probable cause 

After some debugging I found this [line](https://github.com/espressif/esp-idf/blob/c1ab87b58057f0ddb8bad7e15a48a6697e8cda27/components/bt/host/bluedroid/btc/profile/std/hf_ag/btc_hf_ag.c#L464):

```cpp
// Pre-formatted AT response, typically in response to unknown AT cmd
static bt_status_t btc_hf_unat_response(bt_bdaddr_t *bd_addr, const char *unat)
{
    int idx = btc_hf_idx_by_bdaddr(bd_addr);
    CHECK_HF_INIT();
    if ((idx < 0) || (idx >= BTC_HF_NUM_CB)) {
        BTC_TRACE_ERROR("%s: Invalid index %d", __FUNCTION__, idx);
        return BT_STATUS_FAIL;
    }

    if (is_connected(NULL) && (idx != BTC_HF_INVALID_IDX))
    {
        tBTA_AG_RES_DATA    ag_res;
        /* Format the response and send */
        memset(&ag_res, 0, sizeof(ag_res));
        strncpy(ag_res.str, unat, BTA_AG_AT_MAX_LEN);
        BTA_AgResult(hf_local_param[idx].btc_hf_cb.handle, BTA_AG_UNAT_RES, &ag_res);
        return BT_STATUS_SUCCESS;
    }
    return BT_STATUS_FAIL;
}
```
This line is sending NULL to other function which causes the issue, when it is trying to read memory at given address. 
```cpp
is_connected(NULL)
```

Alvin1Zhang, 2020-01-29:
Thanks for reporting and sorry for the slow turnaround. Sorry that we are now during the holiday of Chinese Lunar New Year. Will update to you when we are back, thanks.

AbnerFederer, 2020-02-08:
Hi @KollarRichard ，

This issue has been fixed but it needs time to merge.
If you wish to use right now, please apply this patch.

[fix_ag_conn.patch.zip](https://github.com/espressif/esp-idf/files/4173836/fix_ag_conn.patch.zip)

Thanks
Abner

KollarRichard, 2020-03-11:
Hi, i was working on different things and yesterday i tried bluetooth in latest version4.1 it works.

closed 2020-03-11 by KollarRichard