espressif/esp-idf #4663: undefined reference to a function on all libraries compiled

ilijamr, 2020-01-23:
Hi,
I am using GNU MAKE build system on ESP-IDF v4.0-beta2 and my Makefile looks like this:

PROJECT_NAME := ili9488_lvgl_gui1
EXTRA_COMPONENT_DIRS := $(PROJECT_PATH)components/lcd \
						$(PROJECT_PATH)components/lvgl-gui \
						$(PROJECT_PATH)components/param \
						$(PROJECT_PATH)components/calibration \
						$(PROJECT_PATH)components/touch 
include $(IDF_PATH)/make/project.mk

All the paths are correct when I type 'make list-components':

COMPONENT_DIRS (components searched for here)
/home/xsd/Dokumenti/esp32_projekti/ili9488/ili9488_lvgl_gui1/components
/home/xsd/Dokumenti/esp32_projekti/ili9488/ili9488_lvgl_gui1/components/lcd
/home/xsd/Dokumenti/esp32_projekti/ili9488/ili9488_lvgl_gui1/components/lvgl-gui
/home/xsd/Dokumenti/esp32_projekti/ili9488/ili9488_lvgl_gui1/components/param
/home/xsd/Dokumenti/esp32_projekti/ili9488/ili9488_lvgl_gui1/components/calibration
/home/xsd/Dokumenti/esp32_projekti/ili9488/ili9488_lvgl_gui1/components/touch
/home/xsd/esp/esp-idf/components
/home/xsd/Dokumenti/esp32_projekti/ili9488/ili9488_lvgl_gui1/main

All esp and custom libraries are compiled correctly. 

When collect2.exe starts to link them it prints out error saying that 'calibration' does not recognize functions (undefined reference) from 'param'.
I have tried to put 'COMPONENT_DEPENDS += param' in 'component.mk' of 'calibration' since it depends on it. It does change the order of compilation, 'param' first then 'calibration', but it does not fix the error.

Generating esp32.project.ld
LD build/ili9488_lvgl_gui1.elf
/home/xsd/esp/crosstool-NG/builds/xtensa-esp32-elf/lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/xsd/Dokumenti/esp32_projekti/ili9488/ili9488_lvgl_gui1/build/calibration/libcalibration.a(lv_calibration.o):(.literal.touch_load_calibration+0x10): undefined reference to `iot_param_load'
/home/xsd/esp/crosstool-NG/builds/xtensa-esp32-elf/lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/xsd/Dokumenti/esp32_projekti/ili9488/ili9488_lvgl_gui1/build/calibration/libcalibration.a(lv_calibration.o):(.literal.touch_save_calibration+0x4): undefined reference to `iot_param_save'
/home/xsd/esp/crosstool-NG/builds/xtensa-esp32-elf/lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/xsd/Dokumenti/esp32_projekti/ili9488/ili9488_lvgl_gui1/build/calibration/libcalibration.a(lv_calibration.o): in function `touch_load_calibration':
/home/xsd/Dokumenti/esp32_projekti/ili9488/ili9488_lvgl_gui1/components/calibration/src/lv_calibration.c:114: undefined reference to `iot_param_load'
/home/xsd/esp/crosstool-NG/builds/xtensa-esp32-elf/lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/xsd/Dokumenti/esp32_projekti/ili9488/ili9488_lvgl_gui1/build/calibration/libcalibration.a(lv_calibration.o): in function `touch_save_calibration':
/home/xsd/Dokumenti/esp32_projekti/ili9488/ili9488_lvgl_gui1/components/calibration/src/lv_calibration.c:106: undefined reference to `iot_param_save'
collect2: error: ld returned 1 exit status
/home/xsd/esp/esp-idf/make/project.mk:522: recipe for target '/home/xsd/Dokumenti/esp32_projekti/ili9488/ili9488_lvgl_gui1/build/ili9488_lvgl_gui1.elf' failed
make: *** [/home/xsd/Dokumenti/esp32_projekti/ili9488/ili9488_lvgl_gui1/build/ili9488_lvgl_gui1.elf] Error 1

Any help is appreciated.

bitmakk, 2020-01-24:
PROJECT_NAME := ili9488_lvgl_gui1

EXTRA_COMPONENT_DIRS := $(abspath components/lvgl-gui) $(abspath components/param) $(abspath components/calibration) $(abspath components/touch)

include $(IDF_PATH)/make/project.mk

I use this Makefile in my project (I replaced the names to yours), and it works well at me. Try use this format!

ilijamr, 2020-01-24:
Hi, thanx for reply, I have tried the abs paths. The difference is that it makes only one 'libcomponents.a' however (after changing few includes) it did compile but with more linker errors.

closed 2020-01-26 by ilijamr