espressif/esp-idf #4665: BLE AND BLUETOOTH MODE SWTICHING issue in runtime

sowmimalu, 2020-01-24:
Hai, 
    In esp32 , i am using bluetooth and ble (dual mode ) ..now i need to switch the both modes .
at first it should be in ble mode then if interrupt occurs it should switch for dual mode (ble+bluettooth) for a particular time ...then switch back to ble mode ...

i tried to initialize the ble mode at first then disabled the ble mode after that changing the mode to BTDM (DUAL MODE) but it is not switch over ..shows the error  as follwes ...

E (293) BLE_CONFIG: configBleEnable initialize controller failed: ESP_ERR_INVALID_STATE

int ble_enable()
{
esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
 if ((ret = esp_bt_controller_init(&bt_cfg)) != ESP_OK) {
		ESP_LOGE(SPP_TAG, "%s initialize controller failed: %s\n", __func__,
      esp_err_to_name(ret));
		return 1;
	 }
if(value ==1)
	{
	if ((ret = esp_bt_controller_enable(ESP_BT_MODE_BLE)) != ESP_OK) {
		ESP_LOGE(SPP_TAG, "%s enable controller failed: %s\n", __func__,
				esp_err_to_name(ret));
		return 1;
	}
	}
		else if(value == 10)
		{
			if ((ret = esp_bt_controller_enable(ESP_BT_MODE_BTDM)) != ESP_OK) {
					ESP_LOGE(SPP_TAG, "%s enable controller failed: %s\n", __func__,
							esp_err_to_name(ret));
					return 1;
}
		}
}
 ////***********disable mode **************/
int configBleDisable() {

	ESP_LOGI(SPP_TAG, "Disabling Bluetooth");

	if ((ret = esp_bluedroid_disable()) != ESP_OK) {
		ESP_LOGE(SPP_TAG, "%s disable bluedroid failed: %s\n", __func__,
				esp_err_to_name(ret));
		return 1;
	}

	if ((ret = esp_bluedroid_deinit()) != ESP_OK) {
		ESP_LOGE(SPP_TAG, "%s deinitialize bluedroid failed: %s\n", __func__,
				esp_err_to_name(ret));
		return 1;
	}

	if ((ret = esp_bt_controller_disable()) != ESP_OK) {
		ESP_LOGE(SPP_TAG, "%s disable controller failed: %s\n", __func__,
				esp_err_to_name(ret));
		return 1;
	}

	if ((ret = esp_bt_controller_deinit()) != ESP_OK) {
		ESP_LOGE(SPP_TAG, "%s deinitialize controller failed: %s\n", __func__,
				esp_err_to_name(ret));
		return 1;
	}

	esp_bt_mem_release(ESP_BT_MODE_BLE);

	return 0;

}
 can you please tell me how to switch modes???

Alvin1Zhang, 2020-01-29:
Thanks for reporting and sorry for the slow turnaround. Sorry that we are now during the holiday of Chinese Lunar New Year. Will update to you when we are back, thanks.

dhrishi, 2020-01-29:
@sowmimalu If you want to use BLE and BTDM modes alternatively, then do not use the API `esp_bt_mem_release` in any of your functions. This API is used when you want to release the memory of a particular mode that is not required in the current boot cycle. See the documentation here: https://github.com/espressif/esp-idf/blob/master/components/bt/include/esp_bt.h#L402

Another thing, your ble_enable() API should init and enable the controller as well as the host (Bluedroid). In the code above, you are just initializing and enabling the controller but not the host.

sowmimalu, 2020-01-30:
hai ,
    i have sended u  my half of my ble_enable() code part  ..i have initialised the blueroid also ..
i enabled dual mode in  bluetooth controller option of menuconfig option .
now i want to work on ble mode at first .for that i used the below code

/***************************************************************************/
int ble_enable()
{
	esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
      if ((ret = esp_bt_controller_init(&bt_cfg)) != ESP_OK) {
		ESP_LOGE(SPP_TAG, "%s initialize controller failed: %s\n", __func__,
				esp_err_to_name(ret));
		return 1;
	}
if ((ret = esp_bt_controller_enable(ESP_BT_MODE_BLE)) != ESP_OK) {
		ESP_LOGE(SPP_TAG, "%s enable controller failed: %s\n", __func__,
				esp_err_to_name(ret));
		return 1;
	}
if ((ret = esp_bluedroid_init()) != ESP_OK) {
		ESP_LOGE(SPP_TAG, "%s initialize bluedroid failed: %s\n", __func__,
				esp_err_to_name(ret));
		return 1;
	}

	if ((ret = esp_bluedroid_enable()) != ESP_OK) {
		ESP_LOGE(SPP_TAG, "%s enable bluedroid failed: %s\n", __func__,
				esp_err_to_name(ret));
		return 1;
	}
if ((ret = esp_bt_gap_register_callback(esp_bt_gap_cb)) != ESP_OK) {
		ESP_LOGE(SPP_TAG, "%s gap register failed: %s\n", __func__,
				esp_err_to_name(ret));
		return 1;
	}
if ((ret = esp_spp_register_callback(esp_spp_cb)) != ESP_OK) {
		ESP_LOGE(SPP_TAG, "%s spp register failed: %s\n", __func__,
				esp_err_to_name(ret));
		return 1;	}
if ((ret = esp_spp_init(esp_spp_mode)) != ESP_OK) {
		ESP_LOGE(SPP_TAG, "%s spp init failed: %s\n", __func__,
				esp_err_to_name(ret));
		return 1;

#if (CONFIG_BT_SSP_ENABLED == true)

	esp_bt_sp_param_t param_type = ESP_BT_SP_IOCAP_MODE;
	esp_bt_io_cap_t iocap = ESP_BT_IO_CAP_IO;
	esp_bt_gap_set_security_param(param_type, &iocap, sizeof(uint8_t));
#endif
	esp_bt_pin_type_t pin_type = ESP_BT_PIN_TYPE_VARIABLE;
	esp_bt_pin_code_t pin_code;
	esp_bt_gap_set_pin(pin_type, 0, pin_code);

	}
//   register the  callback function to the gap module
	ret = esp_ble_gap_register_callback(esp_gap_cb);
	if (ret != ESP_OK) {
		ESP_LOGE(GATTC_TAG, "%s gap register failed, error code = %x\n",
				__func__, ret);
		return 1;
	}
	//register the callback function to the gattc module
	ret = esp_ble_gattc_register_callback(esp_gattc_cb);
	if (ret) {
		ESP_LOGE(GATTC_TAG, "%s gattc register failed, error code = %x\n",
				__func__, ret);
		return 1;
	}
	ret = esp_ble_gattc_app_register(PROFILE_A_APP_ID);
	if (ret) {
		ESP_LOGE(GATTC_TAG, "%s gattc app register failed, error code = %x\n",
				__func__, ret);
	}
	ret = esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_SCAN , ESP_PWR_LVL_P9 );
	if(ret){
		ESP_LOGE(GATTC_TAG, "set ESP_BLE_PWR_TYPE_SCAN failed, error code = %x\n", ret);
	}
	ret = esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_ADV , ESP_PWR_LVL_P9 );
	if(ret){
		ESP_LOGE(GATTC_TAG, "set ESP_BLE_PWR_TYPE_ADV failed, error code = %x\n", ret);
	}
	ret = esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_DEFAULT , ESP_PWR_LVL_P9 );
	if(ret){
		ESP_LOGE(GATTC_TAG, "set ESP_BLE_PWR_TYPE_DEFAULT failed, error code = %x\n", ret);
	}

	esp_err_t local_mtu_ret = esp_ble_gatt_set_local_mtu(500);
	if (local_mtu_ret) {
		ESP_LOGE(GATTC_TAG, "set local  MTU failed, error code = %x",
				local_mtu_ret);
	}
	return 0;
}
/***************************************************************************/
but it does not work in ble mode ..it showing error as invalid argument ...may b becoz of i initilized dual mode in menuconfig.. to over come that i hardcoded the mode 
esp_bt_controller_config_t bt_cfg = {
.mode = ESP_BT_MODE_BLE,
};  // instaed of using default config which i written above.
its also not working..
my question is 
is it possible to switch modes at runtime ??
for me it should be in ble mode at first then if i given  interrupt it should be change to dual mode that i need bluetooth and ble should work for a particular  time ..
then after timer ends it should work asusual in ble mode only ...

to make this scenario , i follwed like this ..at first i kept the option as in  ble mode and wriiten the enable code as shown above  ..controller config is in default mode ..now its working for me in ble mode ..
but when i changed my option as dual mode in menuconfig its not worked in ble mode ..
after that i disabled the ble . ..which disable code is shown in my above question ..
then again i enable my dual mode using ble_enable() function ..little bit changed the code and kept as seperate function ...
esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
controller enable api is in dual mode   
if ((ret = esp_bt_controller_enable(ESP_BT_MODE_BTDM)) != ESP_OK) {
		ESP_LOGE(SPP_TAG, "%s enable controller failed: %s\n", __func__,
				esp_err_to_name(ret));
		return 1;
	}
and i kept controller config as defualt to init the controller ..its not worked and also i changed the controller config as like hardcoded its also not working  ...
can  you please tell me  the procedure for how to switch the modes ...and what config it should be????
Thanks ,
Sowmiya

dhrishi, 2020-01-31:
@sowmimalu This seems incorrect as the other members of bt_cfg would be incorrectly initialized
```
esp_bt_controller_config_t bt_cfg = {
.mode = ESP_BT_MODE_BLE,
};
```
Instead you can try -

```
    esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    bt_cfg.mode = <the new mode required>
```
And through menuconfig, you should set the controller to BTDM. Also, as mentioned above, there shouldn't be `esp_bt_mem_release` in your code as you require BLE as well as BTDM in the same boot.

sowmimalu, 2020-02-03:
hai,
 tried as u said , but at first time both modes are working .i did as below ...what i am doing wrong here??
ESP_LOGI(SPP_TAG, "Enabling  bluetooth");
int ble_enable(
	esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
	    bt_cfg.mode = ESP_BT_MODE_BLE;

if ((ret = esp_bt_controller_init(&bt_cfg)) != ESP_OK) {
		ESP_LOGE(SPP_TAG, "%s initialize controller failed: %s\n", __func__,
				esp_err_to_name(ret));
		return 1;
	}
	if ((ret = esp_bt_controller_enable(ESP_BT_MODE_BLE)) != ESP_OK) {
		ESP_LOGE(SPP_TAG, "%s enable controller failed: %s\n", __func__,
				esp_err_to_name(ret));
		return 1;
	}
if ((ret = esp_bluedroid_init()) != ESP_OK) {
		ESP_LOGE(SPP_TAG, "%s initialize bluedroid failed: %s\n", __func__,
				esp_err_to_name(ret));
		return 1;
	}

	if ((ret = esp_bluedroid_enable()) != ESP_OK) {
		ESP_LOGE(SPP_TAG, "%s enable bluedroid failed: %s\n", __func__,
				esp_err_to_name(ret));
		return 1;
	}
if ((ret = esp_bt_gap_register_callback(esp_bt_gap_cb)) != ESP_OK) {
		ESP_LOGE(SPP_TAG, "%s gap register failed: %s\n", __func__,
				esp_err_to_name(ret));
		return 1;
	}

	if ((ret = esp_spp_register_callback(esp_spp_cb)) != ESP_OK) {
		ESP_LOGE(SPP_TAG, "%s spp register failed: %s\n", __func__,
				esp_err_to_name(ret));
		return 1;
	}
if ((ret = esp_spp_init(esp_spp_mode)) != ESP_OK) {
		ESP_LOGE(SPP_TAG, "%s spp init failed: %s\n", __func__,
				esp_err_to_name(ret));
		return 1;

#if (CONFIG_BT_SSP_ENABLED == true)

	esp_bt_sp_param_t param_type = ESP_BT_SP_IOCAP_MODE;
	esp_bt_io_cap_t iocap = ESP_BT_IO_CAP_IO;
	esp_bt_gap_set_security_param(param_type, &iocap, sizeof(uint8_t));
#endif
	esp_bt_pin_type_t pin_type = ESP_BT_PIN_TYPE_VARIABLE;
	esp_bt_pin_code_t pin_code;
	esp_bt_gap_set_pin(pin_type, 0, pin_code);

	}
//   register the  callback function to the gap module
	ret = esp_ble_gap_register_callback(esp_gap_cb);
	if (ret != ESP_OK) {
		ESP_LOGE(GATTC_TAG, "%s gap register failed, error code = %x\n",
				__func__, ret);
		return 1;
	}

	//register the callback function to the gattc module
	ret = esp_ble_gattc_register_callback(esp_gattc_cb);
	if (ret) {
		ESP_LOGE(GATTC_TAG, "%s gattc register failed, error code = %x\n",
				__func__, ret);
		return 1;
	}
	ret = esp_ble_gattc_app_register(PROFILE_A_APP_ID);
	if (ret) {
		ESP_LOGE(GATTC_TAG, "%s gattc app register failed, error code = %x\n",
				__func__, ret);
	}
	ret = esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_SCAN , ESP_PWR_LVL_P9 );
	if(ret){
		ESP_LOGE(GATTC_TAG, "set ESP_BLE_PWR_TYPE_SCAN failed, error code = %x\n", ret);
	}
	ret = esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_ADV , ESP_PWR_LVL_P9 );
	if(ret){
		ESP_LOGE(GATTC_TAG, "set ESP_BLE_PWR_TYPE_ADV failed, error code = %x\n", ret);
	}
	ret = esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_DEFAULT , ESP_PWR_LVL_P9 );
	if(ret){
		ESP_LOGE(GATTC_TAG, "set ESP_BLE_PWR_TYPE_DEFAULT failed, error code = %x\n", ret);
	}

	esp_err_t local_mtu_ret = esp_ble_gatt_set_local_mtu(500);
	if (local_mtu_ret) {
		ESP_LOGE(GATTC_TAG, "set local  MTU failed, error code = %x",
				local_mtu_ret);
	}
	return 0;
}

dhrishi, 2020-02-03:
@sowmimalu Please format the code appropriately so that it is readable.  

The same mode (BLE or BTDM - the one selected) should be passed to `esp_bt_controller_enable()` as well.

sowmimalu, 2020-02-03:
in menuconfig set option as dual mode ... and in my code hardcoded as u said ..

/****************************************/
esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
bt_cfg.mode = ESP_BT_MODE_BLE;
   if ((ret = esp_bt_controller_init(&bt_cfg)) != ESP_OK) {
ESP_LOGE(SPP_TAG, "%s initialize controller failed: %s\n", func,
esp_err_to_name(ret));
return 1;
}
  if ((ret = esp_bt_controller_enable(ESP_BT_MODE_BLE)) != ESP_OK) {
ESP_LOGE(SPP_TAG, "%s enable controller failed: %s\n", func,
esp_err_to_name(ret));
return 1;
}
  if ((ret = esp_bluedroid_init()) != ESP_OK) {
ESP_LOGE(SPP_TAG, "%s initialize bluedroid failed: %s\n", func,
esp_err_to_name(ret));
return 1;
}
 if ((ret = esp_bluedroid_enable()) != ESP_OK) {
	ESP_LOGE(SPP_TAG, "%s enable bluedroid failed: %s\n", __func__,
			esp_err_to_name(ret));
	return 1;
}
 /**********************************************/

sowmimalu, 2020-03-12:
Hai ,
    Tried as you said , at first time BLE MODE working when i tried to connect with BLUETOOTH MODE device gets rebooting ...
then i changed the mode to ESP_BT_MODE_BTDM  before enabling the mode , i disabled the ble using the following function i didnt use any mem_release function in that ...

/*****************************************************/
int configBleDisable() {

	ESP_LOGI(SPP_TAG, "Disabling Bluetooth");

	if ((ret = esp_bluedroid_disable()) != ESP_OK) {
		ESP_LOGE(SPP_TAG, "%s disable bluedroid failed: %s\n", __func__,
				esp_err_to_name(ret));
		return 1;
	}

	if ((ret = esp_bluedroid_deinit()) != ESP_OK) {
		ESP_LOGE(SPP_TAG, "%s deinitialize bluedroid failed: %s\n", __func__,
				esp_err_to_name(ret));
		return 1;
	}

	if ((ret = esp_bt_controller_disable()) != ESP_OK) {
		ESP_LOGE(SPP_TAG, "%s disable controller failed: %s\n", __func__,
				esp_err_to_name(ret));
		return 1;
	}

	if ((ret = esp_bt_controller_deinit()) != ESP_OK) {
		ESP_LOGE(SPP_TAG, "%s deinitialize controller failed: %s\n", __func__,
				esp_err_to_name(ret));
		return 1;
	}
return 0;
}
/*******************************************************/

after disabling my dual mode function is below ...
in that BLE MODE is working fine...when i tried to connect with bluetooth mode 
again its gets rebooting ....
/*********************************************************/

ESP_LOGI(SPP_TAG, "Enabling  bluetooth");

	//esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
	esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
	//    bt_cfg.mode = ESP_BT_MODE_BTDM;
	
if ((ret = esp_bt_controller_init(&bt_cfg)) != ESP_OK) {
		ESP_LOGE(SPP_TAG, "%s initialize controller failed: %s\n", __func__,
				esp_err_to_name(ret));
		return 1;
	}
	if ((ret = esp_bt_controller_enable(ESP_BT_MODE_BTDM)) != ESP_OK) {
		ESP_LOGE(SPP_TAG, "%s enable controller failed: %s\n", __func__,
				esp_err_to_name(ret));
		return 1;
	}
if ((ret = esp_bluedroid_init()) != ESP_OK) {
		ESP_LOGE(SPP_TAG, "%s initialize bluedroid failed: %s\n", __func__,
				esp_err_to_name(ret));
		return 1;
	}

	if ((ret = esp_bluedroid_enable()) != ESP_OK) {
		ESP_LOGE(SPP_TAG, "%s enable bluedroid failed: %s\n", __func__,
				esp_err_to_name(ret));
		return 1;
	}
	if ((ret = esp_bt_gap_register_callback(esp_bt_gap_cb)) != ESP_OK) {
		ESP_LOGE(SPP_TAG, "%s gap register failed: %s\n", __func__,
				esp_err_to_name(ret));
		return 1;
	}
	if ((ret = esp_spp_register_callback(esp_spp_cb)) != ESP_OK) {
		ESP_LOGE(SPP_TAG, "%s spp register failed: %s\n", __func__,
				esp_err_to_name(ret));
		return 1;
	}
	if ((ret = esp_spp_init(esp_spp_mode)) != ESP_OK) {
		ESP_LOGE(SPP_TAG, "%s spp init failed: %s\n", __func__,
				esp_err_to_name(ret));
		return 1;
#if (CONFIG_BT_SSP_ENABLED == true)

	esp_bt_sp_param_t param_type = ESP_BT_SP_IOCAP_MODE;
	esp_bt_io_cap_t iocap = ESP_BT_IO_CAP_IO;
	esp_bt_gap_set_security_param(param_type, &iocap, sizeof(uint8_t));
#endif
	esp_bt_pin_type_t pin_type = ESP_BT_PIN_TYPE_VARIABLE;
	esp_bt_pin_code_t pin_code;
	esp_bt_gap_set_pin(pin_type, 0, pin_code);

	}
//   register the  callback function to the gap module
	ret = esp_ble_gap_register_callback(esp_gap_cb);
	if (ret != ESP_OK) {
		ESP_LOGE(GATTC_TAG, "%s gap register failed, error code = %x\n",
				__func__, ret);
		return 1;
	}

	//register the callback function to the gattc module
	ret = esp_ble_gattc_register_callback(esp_gattc_cb);
	if (ret) {
		ESP_LOGE(GATTC_TAG, "%s gattc register failed, error code = %x\n",
				__func__, ret);
		return 1;
	}
	ret = esp_ble_gattc_app_register(PROFILE_A_APP_ID);
	if (ret) {
		ESP_LOGE(GATTC_TAG, "%s gattc app register failed, error code = %x\n",
				__func__, ret);
	}
	ret = esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_SCAN , ESP_PWR_LVL_P9 );
	if(ret){
		ESP_LOGE(GATTC_TAG, "set ESP_BLE_PWR_TYPE_SCAN failed, error code = %x\n", ret);
	}
	ret = esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_ADV , ESP_PWR_LVL_P9 );
	if(ret){
		ESP_LOGE(GATTC_TAG, "set ESP_BLE_PWR_TYPE_ADV failed, error code = %x\n", ret);
	}
	ret = esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_DEFAULT , ESP_PWR_LVL_P9 );
	if(ret){
		ESP_LOGE(GATTC_TAG, "set ESP_BLE_PWR_TYPE_DEFAULT failed, error code = %x\n", ret);
	}

	esp_err_t local_mtu_ret = esp_ble_gatt_set_local_mtu(500);
	if (local_mtu_ret) {
		ESP_LOGE(GATTC_TAG, "set local  MTU failed, error code = %x",
				local_mtu_ret);
	}
	return 0;
}
could you please  you find out whether to change anything here??