espressif/esp-idf #4666: BLE Mesh + iBeacon simultaneous support

linkineo, 2020-01-24:
I could successfully implement a BLE Mesh application with esp-idf and also try out the iBeacon example you provide.
I'd like however to implement support for both simultaneously.

I could confirm it's possible with some hack.
In ```mesh_bearer_adapt.c```I added a function
```
void do_iBeacon_start(u16_t min, u16_t max){
   tBLE_BD_ADDR p_dir_bda = {0};
   uint8_t data[30] = {...};
   uint32_t len =30 ;

   BLE_MESH_BTM_CHECK_STATUS(BTM_BleWriteAdvDataRaw(data, len));

   BLE_MESH_BTM_CHECK_STATUS(
         BTM_BleSetAdvParamsAll(min, max, BLE_MESH_ADV_IND,
                               BLE_MESH_ADDR_PUBLIC, &p_dir_bda,
                               BLE_MESH_ADV_CHNL_37 | BLE_MESH_ADV_CHNL_38 |            BLE_MESH_ADV_CHNL_39, 
                               BLE_MESH_AP_SCAN_CONN_ALL, NULL));
   BLE_MESH_BTM_CHECK_STATUS(BTM_BleStartAdv());
}
```

and I do call this function right after ```adv_send(struct net_buf *buf)``` located in ```adv.c```like this 
```
   do_iBeacon_start(param.interval_min,param.interval_max);
    k_sleep(duration);
    bt_le_adv_stop();
```

This hack has the consequence of sending an iBeacon advertisement right after sending each regular mesh message. The idea was not to pollute any queues and act sequentially within a sending loop.
It's fully clear that this is a hack and not a good solution. It served however to prove it works.

So my questions:
- Will you implement support for simultaneous support of iBeacon & BLE Mesh ?
- Is the whole idea of trying to do both bad, should I reconsider it ?
- Would you accept a pull request of something more professional (not the current hack), is it worth trying ?
- I've seen BLE Mesh as a standard also provides support for so called Mesh Beacons. Even though my use case really requires iBeacon, do you think BLE Mesh beacons could cover a similar use case ? And if yes, could you provide or point to a coding example ?

Campou, 2020-02-12:
Hi @linkineo 

Thanks for the feedback.

We will implement a solution for sending normal BLE advertising packets and mesh packets simultaneously. This is a good idea.

Mesh Beacons are just used for mesh provisioning and security update, which are sent in a certain period. And the format of Mesh Beacons are fixed.

And I have a question here:
What are the parameters used for sending iBeacon packets? For example, the duration of each iBeacon sending operation, the interval between two sending operations, and the number of sending operations?
If convenient, please introduce the behavior of sending iBeacon packets in your application, which will be a great help for us.

Thanks.

Alvin1Zhang, 2020-04-10:
@linkineo Thanks for reporting, would you please help share if any updates for the question raised by @Campou ? Thanks.

Campou, 2020-07-27:
Hi @linkineo 

We have added the support of sending normal BLE advertising packets when BLE Mesh is enabled. Please enable the `Support sending normal BLE advertising packets` in the menuconfig and have a try.

Thanks.

Alvin1Zhang, 2020-08-04:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates. Thanks.

closed 2020-08-04 by Alvin1Zhang

mrzant, 2020-12-23:
Hi everyone, 

I have a similar issue as @linkineo where I am trying to send BLE advertisement packets when BLE Mesh is enabled. As mentioned by @Campou I enabled the necessary option in menuconfig for sending normal BLE advertising packets. Then, I am using `bt_mesh_ble_adv_start()` defined in `mesh_bearer_adapt.c` with NimBLE host. The current implementation with defined advertising parameters and data does not raise any error, however, it seems once when BLE Mesh advertisement procedure is generated,  it stops overall gap advertisement procedure. This is clearly defined in `adv.c` in `adv_send()` with `bt_le_adv_stop()`.

Therefore, I would like to know how is it possible to handle separately BLE Mesh and BLE advertisements. The only thing that comes to my mind, is that gap callback function should handle this. 

Thanks :)

InfiniteYuan, 2020-12-25:
Hi @mrzant , Please use those APIs declared in components/bt/esp_ble_mesh/api/core/include/esp_ble_mesh_ble_api.h.

Reference example:：
[onoff_server_gatt.zip](https://github.com/espressif/esp-idf/files/5741054/onoff_server_gatt.zip)

mrzant, 2020-12-30:
Hi @InfiniteYuan, 

Thank you for your help. 

I tried using APIs declared in the header file which you mentioned and I initially observed that the duration of BLE advertisement affects scanning and BLE Mesh advertisement procedure. I will come back to you when I explore this in more detail including your example. As I am interested in infinite BLE advertisement duration this will, of course, collide with BLE Mesh as expected. But, it should be handled in a way that minimizes packet loss and average delay, which means non-blocking and synchronized operation between BLE and BLE Mesh advertisement.