espressif/esp-idf #4668: nmea parser example with wrong speed calcutation

neorevx, 2020-01-25:
## Environment

- Development Kit: Custom
- Module or chip used:  ESP32-WROOM-32U

## Problem Description

The speed calculation of nmea parser is outputing in unit km/h, not in m/s as expected.

> The knot is a unit of speed equal to one nautical mile per hour, exactly 1.852 km/h

There two pieces of wrong code:

https://github.com/espressif/esp-idf/blob/c1ab87b58057f0ddb8bad7e15a48a6697e8cda27/examples/peripherals/uart/nmea0183_parser/main/nmea_parser.c#L274-L275

https://github.com/espressif/esp-idf/blob/c1ab87b58057f0ddb8bad7e15a48a6697e8cda27/examples/peripherals/uart/nmea0183_parser/main/nmea_parser.c#L348-L349

It must be divided by 3.6 to archieve m/s.

This piece of code is right:
https://github.com/espressif/esp-idf/blob/c1ab87b58057f0ddb8bad7e15a48a6697e8cda27/examples/peripherals/uart/nmea0183_parser/main/nmea_parser.c#L351-L352

Aparentely this statment is common, then this error will happen eventualy.

Alvin1Zhang, 2020-01-29:
Thanks for reporting and sorry for the slow turnaround. Sorry that we are now during the holiday of Chinese Lunar New Year. Will update to you when we are back, thanks.

meltdown03, 2020-02-02:
Does this example work correctly for you, @neorevx? I can't get it to parse any GPS data. If I uncheck GSA and GSV in menuconfig, I see the raw sentences, so I know it's receiving data. Leaving it checked does not decode. The sentences are $GNGSV, $GPGSV, $GAGSV, $GNGSA, $GNRMC, etc.

neorevx, 2020-02-05:
@meltdown03 
I use an old modified version. I read all sentences without problem.
First, I removed configurations from kconfig and put constants by my self.
I think there's event loop, I removed also.

meltdown03, 2020-02-07:
> I use an old modified version. I read all sentences without problem.
> First, I removed configurations from kconfig and put constants by my self.
> I think there's event loop, I removed also.

Could you share your source code with me? I still cannot get it to work. Thank you...

neorevx, 2020-02-10:
What model of chip are you using? I think you chip don't send all statements, then this `if` will not be true.

neorevx, 2020-02-10:
Here is my code:
https://gist.github.com/neorevx/5249deb702965f056d77c63c65a47aec
https://gist.github.com/neorevx/d1bb6b9a0defd14b4ef159f239985556

but you should observe the modifications:
- I do not use events;
- The values are stored as int. Multipliers/Dividers are in gps.h
- Maybe has some more things...

PS: I use UBLOX NEO 6M module/chip.

meltdown03, 2020-02-22:
I got the example to work without modifications. Originally, I wasn't using a GPS module, I was using captured sentences from my phone and sending them through a USB-to-UART module. It must have been sending extra info.

Alvin1Zhang, 2020-03-30:
@meltdown03 @neorevx Glad you resolved the issue, feel free to reopen if the issue still happens. Thanks.

closed 2020-03-30 by Alvin1Zhang