espressif/esp-idf #4671: dual mode in esp32 (ble + bluetooth )switching issue

vivekalways, 2020-01-27:
Hai,
In esp32 , i am using bluetooth and ble (dual mode ) ..now i need to switch the both modes .
at first it should be in ble mode then if interrupt occurs it should switch for dual mode (ble+bluettooth) for a particular time ...then switch back to ble mode ...

i tried to initialize the ble mode at first then disabled the ble mode after that changing the mode to BTDM (DUAL MODE) but it is not switch over ..shows the error as follwes ...

E (293) BLE_CONFIG: configBleEnable initialize controller failed: ESP_ERR_INVALID_STATE

int ble_enable()
{
esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
if ((ret = esp_bt_controller_init(&bt_cfg)) != ESP_OK) {
ESP_LOGE(SPP_TAG, "%s initialize controller failed: %s\n", func,
esp_err_to_name(ret));
return 1;
}
if(value ==1)
{
if ((ret = esp_bt_controller_enable(ESP_BT_MODE_BLE)) != ESP_OK) {
ESP_LOGE(SPP_TAG, "%s enable controller failed: %s\n", func,
esp_err_to_name(ret));
return 1;
}
}
else if(value == 10)
{
if ((ret = esp_bt_controller_enable(ESP_BT_MODE_BTDM)) != ESP_OK) {
ESP_LOGE(SPP_TAG, "%s enable controller failed: %s\n", func,
esp_err_to_name(ret));
return 1;
}
}
}
////***********disable mode **************/
int configBleDisable() {

ESP_LOGI(SPP_TAG, "Disabling Bluetooth");

if ((ret = esp_bluedroid_disable()) != ESP_OK) {
	ESP_LOGE(SPP_TAG, "%s disable bluedroid failed: %s\n", func,
			esp_err_to_name(ret));
	return 1;
}

if ((ret = esp_bluedroid_deinit()) != ESP_OK) {
	ESP_LOGE(SPP_TAG, "%s deinitialize bluedroid failed: %s\n", func,
			esp_err_to_name(ret));
	return 1;
}

if ((ret = esp_bt_controller_disable()) != ESP_OK) {
	ESP_LOGE(SPP_TAG, "%s disable controller failed: %s\n", func,
			esp_err_to_name(ret));
	return 1;
}

if ((ret = esp_bt_controller_deinit()) != ESP_OK) {
	ESP_LOGE(SPP_TAG, "%s deinitialize controller failed: %s\n", func,
			esp_err_to_name(ret));
	return 1;
}

esp_bt_mem_release(ESP_BT_MODE_BLE);

return 0;
}
can you please tell me how to switch modes???

Alvin1Zhang, 2020-01-29:
Thanks for reporting and sorry for the slow turnaround. Sorry that we are now during the holiday of Chinese Lunar New Year. Will update to you when we are back, thanks.

dhrishi, 2020-01-29:
@Alvin1Zhang Please close this, looks like a duplicate of https://github.com/espressif/esp-idf/issues/4665

Alvin1Zhang, 2020-01-29:
Thanks for updates. Will keep #4665 open for tracking.

closed 2020-01-29 by Alvin1Zhang