espressif/esp-idf #4672: Web-Server

hiflord, 2020-01-27:
**Is your feature request related to a problem? Please describe.**

I want to setup a webserver. Theese are some of AT command that I used for setup web-server.
AT+CWMODE=3
AT+CWLAP
AT+CWJAP="ALDOGAN_WIFI","ALD02468"
AT+CIFSR

AT+CIPMUX=1				
AT+CIPSERVER=1,80				
AT+CIPSEND=0,30	
123456789012345678901234567890		
AT+CIPCLOSE=0
After AT+CIPSERVER=1,80 command I am trying to send request from my phone with 192.168.1.168 this ip via opera. When I finished this command sets I have to refresh my page to see  123456789012345678901234567890 datas. It is not usefull for user. I am trying two times to enter the page. I want to just  one time send request to server to see data. How can I do that with at commands?
Thank you for helping me.

hiflord, 2020-01-27:
Sory I closed wrongly.

xcguang, 2020-02-04:
Can you capture wifi packet? The data should be sent to the client, after `AT+CIPSEND=0,30\r\n, 
123456789012345678901234567890`

hiflord, 2020-02-07:
Yes. Thank you.

closed 2020-02-07 by hiflord