espressif/esp-idf #4675: Missing WIFI_EVENT / IP_EVENT (v3.3.1)

boarchuz, 2020-01-27:
**IDE: VS Code with the official extension.
git describe --tags : v3.3.1
Problem: WIFI_EVENT and IP_EVENT are not declared or defined anywhere**

I'm trying to implement a basic WiFi event handler using the new Event Loop Library. However, the documentation, components, and examples seem to conflict with each other as to whether or not this is supported in v3.3.1.

The docs (https://docs.espressif.com/projects/esp-idf/en/release-v3.3/api-reference/system/esp_event.html) specifically mention WIFI_EVENT and IP_EVENT, which are bases I am obviously interested in to subscribe to connection, disconnection, and got_ip events.
Yet I can't find these declared anywhere in the IDF.

Further confusing me is that the MQTT examples in this IDF version reference these non-existent event bases, eg.:
https://github.com/espressif/esp-mqtt/blob/e3b013e2db58124ea68cf7c8f44a8cba6e1572b7/examples/emitter-client/main/app_main.c#L63
Though I'm guessing they won't compile.

And yet all of the esp_event_loop API seems to be implemented. What am I missing?

closed 2020-02-06 by boarchuz

pauloacmartinez, 2020-04-06:
Hi @boarchuz, I ran into this same issue, how did you solve it?

boarchuz, 2020-04-06:
Hi @pauloacmartinez,
I came to the conclusion that this is not supported in 3.3.1. While this release supports new components using the new events implementation, system events (including WiFi) still use the legacy event system.