espressif/esp-idf #4677: Terminate smart config smoothly when user enter wrong password for hotspot in ESPTOUCH

esonhon, 2020-01-27:
There is a little problem in smart config example of the ESP-IDF master branch.
If user entered wrong password of the WIFI, the esp32 will retry to connect again and agian, and ESPTOUCH wait for long period to report failed status.
We can terminate the ESP32 wait easily, but how can send a message to ESPTOUCH that the password is wrong? How can end smart config in ESPTOUCH smoothly and rapidly?

zhangyanjiaoesp, 2020-06-01:
hi @esonhon ，
If user entered wrong password, the disconnect reason will be 15, user can set the number of the reconnection when receiving disconnected event.
ESPTOUCH APP will wait 1min to report failed status, currently we can't send message to ESPTOUCH  to show that password is wrong.

Alvin1Zhang, 2020-06-02:
@esonhon Thanks for reporting, feel free to reopen if any updates or file a new ticket for new issue/new feature request.

closed 2020-06-02 by Alvin1Zhang