espressif/esp-idf #4679: CMake macro with version

DiegoActionItem, 2020-01-27:
**Is your feature request related to a problem? Please describe.**

A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**

A clear and concise description of what you want to happen.

**Describe alternatives you've considered**

A clear and concise description of any alternative solutions or features you've considered.

Please give as many details as you can. Include suggestions for useful APIs or interfaces if relevant.

**Additional context**

Add any other context or screenshots about the feature request here.

DiegoActionItem, 2020-01-27:
I am missing a CMake macro to lock my build to a specific version (like Qt's `find_package(Qt5Core 5.10 REQUIRED)`. Also, a set of macros to know the exact version on runtime. So we can fail both build, and configuration if the wrong version has been checkedout).

igrr, 2020-01-28:
@DiegoActionItem these macros are available starting from IDF v4.0 release (the final release is not yet available, latest is v4.0-rc). More information: [compile time macros](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/system.html#sdk-version) and [build system variables](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system.html#build-project-variables).

DiegoActionItem, 2020-01-29:
@igrr this solves the compilation part, great.
However - there is no solution to abort configuration. (A workaround is to fail compilation using the macros above, not optimal).

igrr, 2020-01-29:
Do you mean something like this? (in project CMakeLists.txt):

```cmake
cmake_minimum_required(VERSION 3.5)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(my-project)

if (NOT "${IDF_VERSION_MAJOR}.${IDF_VERSION_MINOR}" STREQUAL "4.1")
message(FATAL_ERROR "This version of IDF is not supported by the project")
endif()
```

DiegoActionItem, 2020-01-29:
yeap, something like this. Lets iterate (untested, as I am using v3.3.1 yet...)
```
# should be in $ENV{IDF_PATH}/tools/cmake/project.cmake)
macro(idf_minimum_required vmaj vmin vpath)
    if (NOT ${IDF_VERSION_MAJOR} EQUAL ${vmaj})
        message(FATAL_ERROR "This version of IDF is not supported by the project")
    endif()

    if (NOT ${IDF_VERSION_MIN} LESS ${vmin})
        message(FATAL_ERROR "This version of IDF is not supported by the project")
    endif()

    if (NOT ${IDF_VERSION_PATCH} LESS ${vpatch})
        message(FATAL_ERROR "This version of IDF is not supported by the project")
    endif()
endmacro()

# example usage: 
idf_minimum_required(3,1,1)

```

igrr, 2020-01-29:
Unfortunately these variables have been added in v4.0, so for v3.3 this won't work...

igrr, 2020-08-10:
Backported to 3.3 release in https://github.com/espressif/esp-idf/commit/ca6b009c268a4802157a316cf82ffb9a4315bbf5#diff-b9547c815b64448fe56edb94865e0ddb.

closed 2020-08-10 by igrr