espressif/esp-idf #4682: IRAM usage overflow when activate PSRAM in config

H0bbes, 2020-01-28:
## Environment

- Development Kit:      ESP32-Wrover
- Module or chip used: ESP32-WROVER-B
- IDF version: v4.0-rc
- Build System:         Make
- Compiler version: (crosstool-NG esp-2019r2) 8.2.0
- Operating System:    Linux
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

When I compile my project without external PSRAM support then i can compile without problem.
```
 LD build/ESP32_LINKIO_MESH.elf
Total sizes:
 DRAM .data size:   16180 bytes
 DRAM .bss  size:   40424 bytes
Used static DRAM:   56604 bytes (  67976 available, 45.4% used)
Used static IRAM:  114351 bytes (  16721 available, 87.2% used)
      Flash code: 1014013 bytes
    Flash rodata:  272036 bytes
Total image size:~1416580 bytes (.bin may be padded larger)
```
But if i activate the PSRAM (in make menuconfig) then it fails compilation:

```
LD build/ESP32_LINKIO_MESH.elf
/home/user/Work/software/hub/espressif/project/esp32-linkio-mesh/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/user/Work/software/hub/espressif/project/esp32-linkio-mesh/main-app/build/ESP32_LINKIO_MESH.elf section `.iram0.text' will not fit in region `iram0_0_seg'
/home/user/Work/software/hub/espressif/project/esp32-linkio-mesh/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: IRAM0 segment data does not fit.
/home/user/Work/software/hub/espressif/project/esp32-linkio-mesh/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: region `iram0_0_seg' overflowed by 8699 bytes
collect2: error: ld returned 1 exit status
/home/user/Work/software/hub/espressif/project/esp32-linkio-mesh/esp-idf-v4.0-rc/make/project.mk:528: recipe for target '/home/user/Work/software/hub/espressif/project/esp32-linkio-mesh/main-app/build/ESP32_LINKIO_MESH.elf' failed
make: *** [/home/user/Work/software/hub/espressif/project/esp32-linkio-mesh/main-app/build/ESP32_LINKIO_MESH.elf] Error 1
```
I have the same result if i choose to use the external RAM as map memory or making it RAM allocatable.

### Expected Behavior
Activating external RAM doesn't use ~20k IRAM

### Actual Behavior
Activating external RAM seems to use ~20k of IRAM, making the compilation to faile

igrr, 2020-01-29:
@H0bbes do I understand correctly that your project could be compiled successfully with IDF 3.3 + GCC 5, but with IDF 4.0 + GCC 8 same code does not get linked anymore? (IDF 4 + GCC 5 is not a supported combination and linking results might have been incorrect).
The +20kB RAM usage with PSRAM enabled is possible, depending on how much of the C library you use, when PSRAM is enabled. It is possible to move some parts of the C library into Flash, but I don't know which parts of it are used by your application. The list of object files being pinned to IRAM is here: https://github.com/espressif/esp-idf/blob/master/components/newlib/esp32-spiram-rom-functions-c.lf

If you have an option of switching to ESP32 revision 3 for this project, it would also solve the issue, since PSRAM bug workaround will not be enabled, and the C library will stay in ROM and Flash.

H0bbes, 2020-01-30:
Thank you for your answer.
Exactly: IDF 4.0 + Gcc5.2 compiled and IDF 4.0 + GCC 8.2 linker failed.
The solution of choosing Revision 3 fixed my problem. With this the IRAM usage almost does not not change when i enable/disable the PSRAM. How can i normaly identify that I have a revision 3 W-ROVER ?
To be sure if i plan to only use revision 3 for my development/ production i can keep this parameter to 3 and thus fixing my PSRAM problem?

igrr, 2020-01-30:
>  IDF 4.0 + Gcc5.2 compiled and IDF 4.0 + GCC 8.2 linker failed.
"IDF 4.0 + Gcc5.2 compiled" is not a supported combination, it might have resulted in incorrect placement of code into Flash/IRAM.
Did you code work with "IDF 3.3 + Gcc5.2"?

> How can i normaly identify that I have a revision 3 W-ROVER ?

When you run `esptool.py --port PORT chip_id`, or `idf.py flash`, you should see "revision 3" in the output:
```
Detecting chip type... ESP32
Chip is ESP32D0WDQ5 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
```

> To be sure if i plan to only use revision 3 for my development/ production i can keep this parameter to 3 and thus fixing my PSRAM problem?

Yes, correct.

H0bbes, 2020-01-30:
The devkit i have is 

> Chip is ESP32D0WDQ6 (revision 1)

So how can i know which module to buy that is revision 3?

igrr, 2020-01-30:
Please refer to the product ordering information: https://www.espressif.com/sites/default/files/documentation/espressif_products_ordering_information_en.pdf. The module name is ESP32-WROVER-E (should have "ESP32 ECO v3" in the product description column)

[image]

For more information about revision 3, see https://www.espressif.com/sites/default/files/documentation/ESP32_ECO_V3_User_Guide__EN.pdf.

H0bbes, 2020-01-30:
Thank you for all your answer.
I will close this issue.

closed 2020-01-30 by H0bbes

ESP-Marius, 2023-04-11:
On latest master we've added a kconfig option for using parts of SRAM1 as IRAM in https://github.com/espressif/esp-idf/commit/5cbd311ecf0b381a0711ef8a96980a67b30a135f which will help with this issue.

We've also focused on reducing IRAM usage/making placement optional for several components. For an full overview of all such option see [Optimizing-iram-usage](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/ram-usage.html#optimizing-iram-usage). All new options for saving IRAM we add will be listed in this chapter.