espressif/esp-idf #4685: lwip/dhcp: Hostname not set correctly for discover packets

lsgunth, 2020-01-28:
## Environment

- Development Kit:      Custom Board
- Module or chip used:  [ESP32-WROOM-32D
- IDF version:          v4.1-dev-1957-gaedf89237f84
- Build System:         make
- Compiler version: 8.2.0
- Operating System:   Linux
- Using an IDE?:  No

## Problem Description

Fails to send hostname in DHCP DISCOVER packets. Which causes the device to fail to obtain
an IP address on any DHCP server that has rules based on the hostname. The server will
ISSUE an incorrect IP address which will then be rejected when the ESP device sends a REQUEST with the correct hostname.

### Expected Behavior

The ESP device sends a dhcp DISCOVER packet with the hostname.

### Actual Behavior

The hostname is omitted from the dhcp DISCOVER packet.

### Steps to reproduce

Obtain an IP address from a DHCP server that has a rule based on the hostname

### References

Two months ago I submitted a PR to esp-lwip that has not been acted on:  espressif/esp-lwip#9 

This same issue was fixed for release/v3.2 and release/v3.3 in commit espressif/esp-lwip@0865edf80ad0a96e0eaf2bbedbad350cab248115 but the fix did not make it into master or any of the v4 release branches.

freakyxue, 2020-02-17:
@lsgunth
DHCP hostname add in 
dhcp_select(struct netif *netif)

lsgunth, 2020-02-18:
What? I don't understand.

To be clear: the issue has been solved, I just need someone to merge the fix in the references section for the v4 branches. It was already fixed in v3.2 and v3.3.

beta-tester, 2020-03-02:
i see the same issue on ESP-IDF v4.2
it looks like there is not always "**client provides name:** ..." available from the ESP.
it is only available at stage DHCPREQUEST.

this behavior makes giving an ESP32 device giving a specific IP by its hostname impossible.

here the log on the server:
```
dnsmasq-dhcp[] available DHCP range: 192.168.251.100 -- 192.168.251.250
dnsmasq-dhcp[] DHCPDISCOVER(wlan0) xx:xx:xx:xx:xx:xx 
dnsmasq-dhcp[] tags: wlan0
dnsmasq-dhcp[] DHCPOFFER(wlan0) 192.168.251.101 xx:xx:xx:xx:xx:xx 
dnsmasq-dhcp[] requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
dnsmasq-dhcp[] next server: 192.168.251.1
dnsmasq-dhcp[] sent size:  1 option: 53 message-type  2
dnsmasq-dhcp[] sent size:  4 option: 54 server-identifier  192.168.251.1
dnsmasq-dhcp[] sent size:  4 option: 51 lease-time  infinite
dnsmasq-dhcp[] sent size:  4 option:  1 netmask  255.255.255.0
dnsmasq-dhcp[] sent size:  4 option: 28 broadcast  192.168.251.255
dnsmasq-dhcp[] sent size:  4 option:  3 router  192.168.251.1
dnsmasq-dhcp[] sent size:  4 option:  6 dns-server  192.168.251.1

dnsmasq-dhcp[] available DHCP range: 192.168.251.100 -- 192.168.251.250
dnsmasq-dhcp[] client provides name: sensor_1
dnsmasq-dhcp[] DHCPREQUEST(wlan0) 192.168.251.101 xx:xx:xx:xx:xx:xx 
dnsmasq-dhcp[] tags: known_wlan0_m1, known, wlan0
dnsmasq-dhcp[] DHCPACK(wlan0) 192.168.251.101 xx:xx:xx:xx:xx:xx sensor_1
dnsmasq-dhcp[] requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
dnsmasq-dhcp[] next server: 192.168.251.1
dnsmasq-dhcp[] sent size:  1 option: 53 message-type  5
dnsmasq-dhcp[] sent size:  4 option: 54 server-identifier  192.168.251.1
dnsmasq-dhcp[] sent size:  4 option: 51 lease-time  infinite
dnsmasq-dhcp[] sent size:  4 option:  1 netmask  255.255.255.0
dnsmasq-dhcp[] sent size:  4 option: 28 broadcast  192.168.251.255
dnsmasq-dhcp[] sent size:  4 option:  3 router  192.168.251.1
dnsmasq-dhcp[] sent size:  4 option:  6 dns-server  192.168.251.1
```

in contrast to a raspberrypi, there is at each stage a "**client provides name:** ..." available:
```
dnsmasq-dhcp[] available DHCP range: 192.168.251.100 -- 192.168.251.250
dnsmasq-dhcp[] client provides name: raspberrypi

dnsmasq-dhcp[] available DHCP range: 192.168.251.100 -- 192.168.251.250
dnsmasq-dhcp[] client provides name: raspberrypi
dnsmasq-dhcp[] DHCPDISCOVER(wlan0) zz:zz:zz:zz:zz:zz 
dnsmasq-dhcp[] tags: client_is_a_pi, wlan0
dnsmasq-dhcp[] reply delay: 2
dnsmasq-dhcp[] DHCPOFFER(wlan0) 192.168.251.225 zz:zz:zz:zz:zz:zz 
dnsmasq-dhcp[] requested options: 1:netmask, 3:router, 26:mtu, 252, 42:ntp-server, 
dnsmasq-dhcp[] requested options: 15:domain-name, 6:dns-server, 12:hostname
dnsmasq-dhcp[] server name: 0.0.0.0
dnsmasq-dhcp[] next server: 192.168.251.1
dnsmasq-dhcp[] sent size:  1 option: 53 message-type  2
dnsmasq-dhcp[] sent size:  4 option: 54 server-identifier  192.168.251.1
dnsmasq-dhcp[] sent size:  4 option: 51 lease-time  1h
dnsmasq-dhcp[] sent size:  4 option:  1 netmask  255.255.255.0
dnsmasq-dhcp[] sent size:  4 option: 28 broadcast  192.168.251.255
dnsmasq-dhcp[] sent size:  4 option:  3 router  192.168.251.1
dnsmasq-dhcp[] sent size:  4 option:  6 dns-server  192.168.251.1
dnsmasq-dhcp[] sent size:  4 option: 42 ntp-server  192.168.251.1

dnsmasq-dhcp[] available DHCP range: 192.168.251.100 -- 192.168.251.250
dnsmasq-dhcp[] client provides name: raspberrypi
dnsmasq-dhcp[] DHCPREQUEST(wlan0) 192.168.251.225 zz:zz:zz:zz:zz:zz 
dnsmasq-dhcp[] tags: client_is_a_pi, wlan0
dnsmasq-dhcp[] DHCPACK(wlan0) 192.168.251.225 zz:zz:zz:zz:zz:zz raspberrypi
dnsmasq-dhcp[] requested options: 1:netmask, 3:router, 26:mtu, 252, 42:ntp-server, 
dnsmasq-dhcp[] requested options: 15:domain-name, 6:dns-server, 12:hostname
dnsmasq-dhcp[] server name: 0.0.0.0
dnsmasq-dhcp[] next server: 192.168.251.1
dnsmasq-dhcp[] sent size:  1 option: 53 message-type  5
dnsmasq-dhcp[] sent size:  4 option: 54 server-identifier  192.168.251.1
dnsmasq-dhcp[] sent size:  4 option: 51 lease-time  1h
dnsmasq-dhcp[] sent size:  4 option:  1 netmask  255.255.255.0
dnsmasq-dhcp[] sent size:  4 option: 28 broadcast  192.168.251.255
dnsmasq-dhcp[] sent size:  4 option:  3 router  192.168.251.1
dnsmasq-dhcp[] sent size:  4 option:  6 dns-server  192.168.251.1
dnsmasq-dhcp[] sent size:  9 option: 12 hostname  raspberrypi
dnsmasq-dhcp[] sent size:  4 option: 42 ntp-server  192.168.251.1
```

freakyxue, 2020-03-03:
hi @beta-tester 
This bug has been fixed, and it will be synchronized to GitHub later

beta-tester, 2020-03-20:
for me, it is fixed now... thank you
https://github.com/espressif/esp-lwip/pull/9
https://github.com/espressif/esp-lwip/commit/13665c14e19dfda6b2ad30c1e5d88e0415cf8b7a

lsgunth, 2020-03-31:
Any chance this will make it into the v4.1 release branch? I fixed this 4 months ago and I still have to carry the patch in my own fork (which is a huge pain and quite frustrating).

freakyxue, 2020-04-01:
hi @lsgunth 
This bug has been fixed,This fix will be included in the next idf4.1 release。
I'm sorry to keep you waiting

Alvin1Zhang, 2020-07-01:
@lsgunth Thanks for reporting and sorry for slow turnaround, the fix is available at https://github.com/espressif/esp-idf/commit/b8c2fedec95d6915c7ddf140b4a9af094862c403, feel free to reopen if the issue still happens. Thanks.

closed 2020-07-01 by Alvin1Zhang