espressif/esp-idf #4686: install.bat creates massive amount of python.exe instances

axbusch, 2020-01-28:
```cD:\esp\idf>install.bat
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, esp32ulp-elf, cmake, openocd-esp32, mconf, ninja, idf-exe, ccache
Installing xtensa-esp32-elf@esp-2019r2-8.2.0
Downloading xtensa-esp32-elf-gcc8_2_0-esp-2019r2-win32.zip to D:\esp\toolchain\.espressif\dist\xtensa-esp32-elf-gcc8_2_0-esp-2019r2-win32.zip.tmp
Done
Extracting D:\esp\toolchain\.espressif\dist\xtensa-esp32-elf-gcc8_2_0-esp-2019r2-win32.zip to D:\esp\toolchain\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0
Installing esp32ulp-elf@2.28.51.20170517
Downloading binutils-esp32ulp-win32-2.28.51-esp32ulp-20180809.zip to D:\esp\toolchain\.espressif\dist\binutils-esp32ulp-win32-2.28.51-esp32ulp-20180809.zip.tmp
Done
Extracting D:\esp\toolchain\.espressif\dist\binutils-esp32ulp-win32-2.28.51-esp32ulp-20180809.zip to D:\esp\toolchain\.espressif\tools\esp32ulp-elf\2.28.51.20170517
Installing cmake@3.13.4
Downloading cmake-3.13.4-win64-x64.zip to D:\esp\toolchain\.espressif\dist\cmake-3.13.4-win64-x64.zip.tmp
Done
Extracting D:\esp\toolchain\.espressif\dist\cmake-3.13.4-win64-x64.zip to D:\esp\toolchain\.espressif\tools\cmake\3.13.4
Installing openocd-esp32@v0.10.0-esp32-20190313
Downloading openocd-esp32-win32-0.10.0-esp32-20190313.zip to D:\esp\toolchain\.espressif\dist\openocd-esp32-win32-0.10.0-esp32-20190313.zip.tmp
Done
Extracting D:\esp\toolchain\.espressif\dist\openocd-esp32-win32-0.10.0-esp32-20190313.zip to D:\esp\toolchain\.espressif\tools\openocd-esp32\v0.10.0-esp32-20190313
Installing mconf@v4.6.0.0-idf-20190628
Downloading mconf-v4.6.0.0-idf-20190628-win32.zip to D:\esp\toolchain\.espressif\dist\mconf-v4.6.0.0-idf-20190628-win32.zip.tmp
Done
Extracting D:\esp\toolchain\.espressif\dist\mconf-v4.6.0.0-idf-20190628-win32.zip to D:\esp\toolchain\.espressif\tools\mconf\v4.6.0.0-idf-20190628
Installing ninja@1.9.0
Downloading ninja-1.9.0-win64.zip to D:\esp\toolchain\.espressif\dist\ninja-1.9.0-win64.zip.tmp
Done
Extracting D:\esp\toolchain\.espressif\dist\ninja-1.9.0-win64.zip to D:\esp\toolchain\.espressif\tools\ninja\1.9.0
Installing idf-exe@1.0.1
Downloading idf-exe-v1.0.1.zip to D:\esp\toolchain\.espressif\dist\idf-exe-v1.0.1.zip.tmp
Done
Extracting D:\esp\toolchain\.espressif\dist\idf-exe-v1.0.1.zip to D:\esp\toolchain\.espressif\tools\idf-exe\1.0.1
Installing ccache@3.7
Downloading ccache-3.7-w64.zip to D:\esp\toolchain\.espressif\dist\ccache-3.7-w64.zip.tmp
Done
Extracting D:\esp\toolchain\.espressif\dist\ccache-3.7-w64.zip to D:\esp\toolchain\.espressif\tools\ccache\3.7
Setting up Python environment
Creating a new Python environment in D:\esp\toolchain\.espressif\python_env\idf4.0_py3.7_env
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
[... 36 lines cut ...]
    return self._wait(timeout=timeout)
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\lib\subprocess.py", line 1259, in _wait
    timeout_millis)
KeyboardInterrupt
Traceback (most recent call last):
  File "D:\esp\idf\tools\idf_tools.py", line 1372, in <module>
    main(sys.argv[1:])
  File "D:\esp\idf\tools\idf_tools.py", line 1368, in main
    action_func(args)
  File "D:\esp\idf\tools\idf_tools.py", line 1184, in action_install_python_env
    stdout=sys.stdout, stderr=sys.stderr)
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\lib\subprocess.py", line 358, in check_call
    retcode = call(*popenargs, **kwargs)
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\lib\subprocess.py", line 341, in call
    return p.wait(timeout=timeout)
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\lib\subprocess.py", line 1019, in wait
    return self._wait(timeout=timeout)
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\lib\subprocess.py", line 1259, in _wait
    timeout_millis)
...
...
```

Win7 (yeah, deprecated)
IDF 4.0

igrr, 2020-01-29:
Hi @axbusch, if you open CMD.exe, and try creating a virtual environment without IDF, does this work? I.e.:

```
D:
cd D:\esp
where python
python -m virtualenv myenv
```

axbusch, 2020-01-29:
```
C:\Windows\system32>d:

D:\>esp
Der Befehl "esp" ist entweder falsch geschrieben oder
konnte nicht gefunden werden.

D:\>cd esp

D:\esp>python
Python 3.7.5 (tags/v3.7.5:5c02a39a0b, Oct 15 2019, 00:11:34) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> ^Z

D:\esp>python -m virtualenv myenv
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Running virtualenv with interpreter C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\python.exe
Traceback (most recent call last):
  File "C:\Users\funky\AppData\Roaming\Python\Python37\site-packages\virtualenv.py", line 2611, in <module>
    main()
  File "C:\Users\funky\AppData\Roaming\Python\Python37\site-packages\virtualenv.py", line 815, in main
    raise SystemExit(sub_process_call.wait())
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\lib\subprocess.py", line 1019, in wait
    return self._wait(timeout=timeout)
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\lib\subprocess.py", line 1259, in _wait
    timeout_millis)
KeyboardInterrupt
Traceback (most recent call last):
  File "C:\Users\funky\AppData\Roaming\Python\Python37\site-packages\virtualenv.py", line 2611, in <module>
    main()
  File "C:\Users\funky\AppData\Roaming\Python\Python37\site-packages\virtualenv.py", line 815, in main
    raise SystemExit(sub_process_call.wait())
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\lib\subprocess.py", line 1019, in wait
    return self._wait(timeout=timeout)
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\lib\subprocess.py", line 1259, in _wait
    timeout_millis)
KeyboardInterrupt
Traceback (most recent call last):
  File "C:\Users\funky\AppData\Roaming\Python\Python37\site-packages\virtualenv.py", line 2611, in <module>
    main()
  File "C:\Users\funky\AppData\Roaming\Python\Python37\site-packages\virtualenv.py", line 815, in main
    raise SystemExit(sub_process_call.wait())
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\lib\subprocess.py", line 1019, in wait
    return self._wait(timeout=timeout)
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\lib\subprocess.py", line 1259, in _wait
    timeout_millis)
KeyboardInterrupt
Traceback (most recent call last):
  File "C:\Users\funky\AppData\Roaming\Python\Python37\site-packages\virtualenv.py", line 2611, in <module>
    main()
  File "C:\Users\funky\AppData\Roaming\Python\Python37\site-packages\virtualenv.py", line 815, in main
    raise SystemExit(sub_process_call.wait())
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\lib\subprocess.py", line 1019, in wait
    return self._wait(timeout=timeout)
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\lib\subprocess.py", line 1259, in _wait
    timeout_millis)
KeyboardInterrupt
Traceback (most recent call last):
  File "C:\Users\funky\AppData\Roaming\Python\Python37\site-packages\virtualenv.py", line 2611, in <module>
    main()
  File "C:\Users\funky\AppData\Roaming\Python\Python37\site-packages\virtualenv.py", line 815, in main
    raise SystemExit(sub_process_call.wait())
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\lib\subprocess.py", line 1019, in wait
[... 29 lines cut ...]
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\lib\subprocess.py", line 1019, in wait
    return self._wait(timeout=timeout)
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\lib\subprocess.py", line 1259, in _wait
    timeout_millis)
KeyboardInterrupt
Traceback (most recent call last):
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\funky\AppData\Roaming\Python\Python37\site-packages\virtualenv.py", line 2611, in <module>
    main()
  File "C:\Users\funky\AppData\Roaming\Python\Python37\site-packages\virtualenv.py", line 815, in main
    raise SystemExit(sub_process_call.wait())
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\lib\subprocess.py", line 1019, in wait
    return self._wait(timeout=timeout)
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\lib\subprocess.py", line 1259, in _wait
    timeout_millis)
KeyboardInterrupt

D:\esp>

D:\esp>

```

axbusch, 2020-01-29:
i have also python 2.7

`C:\Python27>
C:\Python27>
C:\Python27>
C:\Python27>
C:\Python27>python
Python 2.7.17 (v2.7.17:c2f86d86e6, Oct 19 2019, 21:01:17) [MSC v.1500 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> ^Z

C:\Python27>python -m virtualenv myenv
New python executable in C:\Python27\myenv\Scripts\python.exe
Installing setuptools, pip, wheel...
done.

C:\Python27>`

igrr, 2020-01-29:
Okay, it seems that there is some problem with the Python version which came with Visual Studio.

You can either a) change your user PATH environment variable so that Python 2.7 becomes the default one, or b) install another copy of Python 3.7 from python.org, and make it the default one (probably better, since python 2 is EOL), or c) use IDF Tools Installer for Windows and choose your Python 2 installation when prompted (https://docs.espressif.com/projects/esp-idf/en/latest/get-started/windows-setup.html#esp-idf-tools-installer).

Option 3 has lowest possible impact on other software in your system, since no global environment paths will be changed. However option 2 is probably the better long term solution, provided that VS doesn't depend on its python copy being the default one.

Alvin1Zhang, 2020-05-26:
@axbusch Thanks for reporting and sorry for the slow turnaround, would you please help check if any updates for the issue? Thanks.

georgik, 2021-03-26:
@axbusch Problem with Python already solved in Windows Installer 2.5+ which contains Embedded Python 3.7.
Please download version from https://dl.espressif.com/dl/esp-idf/

closed 2021-03-26 by tom-borcin