espressif/esp-idf #4687: ninja-build fails as a tool for eclipse with the eclipse add-on in centos7.

pushtoopen, 2020-01-29:
## Environment

- IDF version v4.1-dev-2047-g600d542f5

- Build System:         [CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     [Linux] - centos-release-7-7.1908.0.el7.centos.x86_64
- Using an IDE?: [Yes] eclipse

## Problem Description
ninja-build fails to work properly in v1.9.0 within a centos 7 environment
link below describes issue.
https://github.com/ninja-build/ninja/issues/1533 
ninja-build 1.10.0 fixes this issue. installing the IDF tools within the eclipse environment forces the download of v1.9.0 in the tools.json file. 

Please fix this to the latest ninja-build version so that this is no longer an issue.

pushtoopen, 2020-01-29:
my apologies, this should exist in https://github.com/espressif/idf-eclipse-plugin.

igrr, 2020-01-29:
It's okay, Eclipse plugin installs whatever is specified in `tools/tools.json` for the specific IDF version. We will update it and backport to older releases.

closed 2020-03-19 by espressif-bot