espressif/esp-idf #4688: Make use of CMUX (GSM 07.10) protocol for PPPoS and GSM modules

GeorgeGkinis, 2020-01-29:
### **Is your feature request related to a problem? Please describe.**
While the **PPPoS** example works, the need to exit **PPPoS** and enter **command mode** to send
**AT commands** or read **NMEA** information from the GSM modems proves difficult and requires alot of task handling.

There is also a delay of 1 second required to successfully exit PPPoS mode.

This proves problematic in practice when frequent measuring/sending of data is required.

### _**Describe the solution you'd like**_
I would like to see the [GSM 07.10 protocol](https://www.etsi.org/deliver/etsi_ts/101300_101399/101369/07.01.00_60/ts_101369v070100p.pdf) implemented.
Possibly in the [modem](https://github.com/espressif/esp-idf/tree/master/examples/protocols/pppos_client/components/modem) component.

I suppose the solution could be to implement something like _**virtual UART**_ ports.
Each Channel would be a different **UART** virtual device.

This would allow compatibility with libraries to parse NMEA, send/receive AT commands, lwIP/PPPos and other GSM module related libraries.

The following channels are desired :
- **PPPoS** : to be used by the network stack.
- **AT Command** : to be able to control the module without disconnecting.
- **NMEA**:  to read incoming GPS information without affecting communications.

### **_A clear and concise description of what you want to happen._**

I would like to :
- Setup network connectivity via PPPoS and utilise the ESP32 Network stack.
- Be able to send/read SMS while staying connected
- Be able to send various AT commands to the module while staying connected
- Be able to receive and parse incoming NMEA sequences

### **_A clear and concise description of any alternative solutions or features you've considered._**
For the solution I am working at I communicate via MQTTS to the internet.
I can use the modems MQTTS capabilities via AT commands and only connect through PPPoS for OTA firmware updates.

This keeps me in AT Command mode and I can read/send SMS, read NMEA sequences and send telemetry.

If another communication protocol needs to be used like HTTPS, the complexity would increase significantly and also power usage and data overhead,
as disconnecting from PPPoS would break all my connections and handshakes should be performed at each reconnect.

### **_Please give as many details as you can. Include suggestions for useful APIs or interfaces if relevant._**

The **gsm07.10 protocol** **IS used** within the [bluedroid component](https://github.com/espressif/esp-idf/tree/master/components/bt/host/bluedroid/stack/rfcomm) in ESP-IDF.  
More details [here](https://www.amd.e-technik.uni-rostock.de/ma/gol/lectures/wirlec/bluetooth_info/rfcomm.html#Multiple%20Emulated%20Serial%20Ports)  
It seems that repurposing that library would be the quickest route to add this feature.

A basic driver is created for linux [here](https://github.com/ya-jeks/gsmmux).
Beware, the licence is GPL.

The [OVMS project](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3) has a wonderful [implementation of GSM 07.10](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/tree/master/vehicle/OVMS.V3/components/simcom/src)
and is the only ESP32 compatible solution that I have found so far.
Check this [post](http://lists.openvehicles.com/pipermail/ovmsdev/2017-October/003359.html)

Unfortunately it cannot be used directly as a component in ESP-IDF without including a host of other OVMS modules.

Implementing a generic UART based GSM 07.10 protocol for ESP-IDF could really boost it's popularity, as there seems to be no (open source) implementation anywhere for embedded systems...

### **Additional context**

Nothing more to add right now.

gunarkroeger, 2020-01-29:
I'll add that the pppos example does not work that well. There are some unresolved issues that make using GSM in a comercial product unreliable. But I agree with @GeorgeGkinis that even if pppos worked reliably, not being able to even get the rssi while connected to internet is really bad.

suda-morris, 2020-02-26:
@GeorgeGkinis Cool, thank you for such a  detailed request and I agree it's a nice idea!
But we still need some time to get this feature implemented.

GeorgeGkinis, 2020-03-11:
I understand that it will be a complicated task to seamlessly integrate with the network stack.
Thanks for showing interest!

d3vil-st, 2020-05-19:
I've made PoC with current component to support CMUX as an option and it eventually works.
At least I'm able to connect to MQTT broker, but seems there are a lot of work to make it stable.
Once it was able to make OTA update with this CMUX implementation.
https://github.com/d3vil-st/esp-idf/tree/release/v4.1

suda-morris, 2020-05-20:
Congratulations @d3vil-st It's really nice to have this feature!

olliiiver, 2021-01-10:
Thanks @d3vil-st for your contribution. I was able to use your component. On top I did some work for better frame handling. 

Currently I can talk to MQTT servers (TLS enabled as well) without any issues.

The Repo can be found here:

https://github.com/olliiiver/esp_lte_modem

d3vil-st, 2021-01-10:
> 
> 
> Thanks @d3vil-st for your contribution. I was able to use your component. On top I did some work for better frame handling.
> 
> Currently I can talk to MQTT servers (TLS enabled as well) without any issues.
> 
> The Repo can be found here:
> 
> https://github.com/olliiiver/esp_lte_modem

Cool and nice to hear!

Did you tried to perform OTA update through this? Cause in my implementation it was able to perform it only in 1 from 10 attempts.

AxelLin, 2021-03-02:
Hi @d3vil-st  @olliiiver 

What is the recommend error handling if esp_dte->buffer[0] != SOF_MARKER or Missing end SOF?

olliiiver, 2021-03-02:
@d3vil-st Haven't had the chance to test something other than MQTT
@AxelLin Like mentioned in my repo a small buffer size leads to problems. There is currently no implementation to recover lost frames.

closed 2021-09-10 by espressif-bot