espressif/esp-idf #4689: BLE connectivity problem ESP-IDF-3.3.1

andrew-krv, 2020-01-29:
## Environment

- Development Kit: Custom board with ESP32-Wrover-B
- Module or chip used:  ESP32-WROVER-B
- IDF version: v3.3.1
- Build System:  Make
- Compiler version: 5.2.0
- Operating System: macOS v10.15.2
- Using an IDE?: No
- Power Supply: external 5V

## Problem Description

Hi, I have faced BLE disconnection problem while sending data to device from Samsung Galaxy S10. This smartphone definitely does something different, compared to other smartphones, but I guess it could be handled. Bluedroid Bluetooth stack is being used. Problem is easy to reproduce using example from ESP-IDF v3.3.1 (GATT_SERVICE_TABLE_DEMO) and there is no such such problem with the same example provided in ESP-IDF v4.0-rc or master. Also there is no such problem using NimBLE (I have tried to create the same behaviour app, but with new stack).

### Expected Behavior

- Connecting to device
- Requesting MTU
- Sending large data block (more than 40-60 bytes)

### Actual Behavior

- Connecting to device
- Requesting MTU (any size causes problem)
- Trying to send data
- Disconnecting with reason 0x8 (timeout)

### Steps to reproduce

- Connect to device flashed with GATT_SERVICE_TABLE_DEMO
- Requesting MTU (for example 400)
- Send large data block (more than 40-60 bytes)

### Code to reproduce this issue

ESP-IDF v3.3.1 - GATT_SERVICE_TABLE_DEMO example

## Debug Logs (nrfConnect)

```
[Log](https://gist.github.com/andrew-krv/5ceea4806e22a84ff427b2eb18dfd848l) - failed sending
[Log](https://gist.github.com/andrew-krv/ffa74b860cc4c2ce7f5ef520fa4b7788) - expected behaviour while sending
```

Ben139348, 2020-02-01:
We have exactly the same issue. It also happens with the Samsung Galaxy Note10+

GYC-Espressif, 2020-02-02:
Hello, we have solved this problem before and the backport to release/v3.3 is in process. Thank you for your report. Please wait for our update.

pootytang3, 2020-02-18:
@andrew-krv @Ben139348 are you using Android 10 OS on your devices?

andrew-krv, 2020-02-18:
> @andrew-krv @Ben139348 are you using Android 10 OS on your devices?

Yes, we are. Issue seems to be reproducible only using Galaxy S10 or Galaxy Note 10 (any of their variation) and Android 10.

rjcasson, 2020-02-25:
We have just flashed to the latest version on release/v3.3 and are no longer seeing this specific issue on our Galaxy S10+ devices running Android 10, but on one of the two (seemingly identical) devices we are finding that a similar but different issue has surfaced. More often than not, the mobile app that is connecting to the ESP32 will send a message but will never get a response (the BLE connection stays intact though). To complicate debugging this issue, when we have the ESP32 print all sent/received messages to the console, the issue disappears. We are thinking it might have something to do with message timing on the ESP32 side, but will need to do additional testing to confirm. Still no issue with other Android or iOS devices that we've tested with though. 

@andrew-krv and @Ben139348, we'd love to hear if the latest version fixes your original issues and if you come across anything similar to what I've mentioned above. Thanks!

GYC-Espressif, 2020-02-26:
@rjcasson  Thank you for your information. We have indeed backported the solution in release / v3.3.  We did not encounter the new issue you mentioned. If you have a way to reproduce the problem, please let me know. Thank you.

andrew-krv, 2020-02-26:
@rjcasson issue seems to be fixed with latest release/v3.3 branch, currently works on our device.

Alvin1Zhang, 2020-02-26:
@andrew-krv Glad to know that your issue is resolved @GYC-Espressif Thanks for the support. Feel free to reopen if the issue still happens. Thanks.

closed 2020-02-27 by Alvin1Zhang