espressif/esp-idf #4690: ULP wakeup compatibility with RTC peripherals forced on

boarchuz, 2020-01-29:
There are notes in the docs[1] and API[2] that:

> In revisions 0 and 1 of the ESP32, ULP wakeup source can not be used when RTC_PERIPH power domain is forced to be powered on (ESP_PD_OPTION_ON) or when ext0 wakeup source is used.

I believe this note about RTC_PERIPH is in error. Having tested several WROOM32s (chip rev 1), ULP wakeup seems to work perfectly well regardless of RTC_PERIPH being AUTO or forced ON.

Demo program:
- ULP runs every 1 second, triggers SoC interrupt every 5 seconds.
- Every wakeup, SoC alternates RTC_PERIPH_PD between AUTO and ON then returns to deep sleep.
- An LED (or scope) may be connected to GPIO 33 for visual confirmation of RTC power settings: when AUTO, it can be observed blinking at 1Hz as RTC powers up briefly for the ULP; when ON, it is always on.

```
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp32/ulp.h"
#include "driver/rtc_io.h"
#include "esp_log.h"

static const char* TAG = "ulpwake";

#define ULP_SOC_WAKE_COUNTER_THRESHOLD 5
#define ULP_WAKEUP_INTERVAL (1ULL * 1000 * 1000)

#define RTC_LED_INDICATOR_PIN GPIO_NUM_33
#define RTC_LED_INDICATOR_ON_LEVEL 0

RTC_DATA_ATTR bool peripherals_forced_on = false;

void start_ulp()
{   
    const ulp_insn_t program[] = {
        //Load counter variable (offset == 8) and increment
        I_MOVI(R1, 8),
        I_LD(R0, R1, 0),
        I_ADDI(R0, R0, 1),
        //Check if counter has reached threshold
        I_BL(3, ULP_SOC_WAKE_COUNTER_THRESHOLD),
            //Then wake SoC and reset counter
            I_WAKE(),
            I_MOVI(R0, 0),
        //Save counter and sleep
        I_ST(R0, R1, 0),
        I_HALT(),

        //Reserved word for ULP counter variable (offset == 8)
            I_HALT(),
    };

    ESP_ERROR_CHECK( ulp_set_wakeup_period(0, ULP_WAKEUP_INTERVAL) );
    size_t program_size = sizeof(program) / sizeof(ulp_insn_t);
    ESP_ERROR_CHECK( ulp_process_macros_and_load(0, program, &program_size) );
    ESP_ERROR_CHECK( ulp_run(0) );
}

void app_main()
{
    if(esp_reset_reason() == ESP_RST_DEEPSLEEP)
    {
        ESP_LOGI(TAG, "Woken from deep sleep!");

        esp_sleep_wakeup_cause_t reason = esp_sleep_get_wakeup_cause();
        if(reason == ESP_SLEEP_WAKEUP_ULP && peripherals_forced_on)
        {
            ESP_LOGE(TAG, "Woken by ULP despite ESP_PD_DOMAIN_RTC_PERIPH == ESP_PD_OPTION_ON");
        }
    }
    else
    {
        ESP_LOGI(TAG, "Starting ULP");
        start_ulp();

        ESP_LOGI(TAG, "Turning on RTC LED");
        rtc_gpio_init(RTC_LED_INDICATOR_PIN);
        rtc_gpio_set_direction(RTC_LED_INDICATOR_PIN, RTC_GPIO_MODE_OUTPUT_ONLY);
        rtc_gpio_set_level(RTC_LED_INDICATOR_PIN, RTC_LED_INDICATOR_ON_LEVEL);
    }

    //Alternate RTC peripherals between ON and AUTO each wakeup
    if(peripherals_forced_on)
    {
        ESP_LOGI(TAG, "Setting RTC peripherals to AUTO");
        ESP_ERROR_CHECK(esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_AUTO));
    }
    else
    {
        ESP_LOGI(TAG, "Forcing RTC peripherals ON");
        ESP_ERROR_CHECK(esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON));
    }
    peripherals_forced_on = !peripherals_forced_on;

    ESP_LOGI(TAG, "Sleeping");
    ESP_ERROR_CHECK( esp_sleep_enable_ulp_wakeup() );
    esp_deep_sleep_start();
}
```

[1] https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/sleep_modes.html#ulp-coprocessor-wakeup
[2] https://github.com/espressif/esp-idf/blob/master/components/esp32/include/esp_sleep.h#L92

igrr, 2020-01-29:
I think this part of code will force RTC_PERI to be powered down if ULP wakeup is enabled: https://github.com/espressif/esp-idf/blob/f5b82c5b1f33f5287f9b5f1993594bf53755a4f2/components/esp32/sleep_modes.c#L667, so not sure if your test actually successfully keeps RTC_PERI powered on?

I think the issue with coexistence of RTC_PERI force-power-on and ULP/Touch is not related to an inability to wake up; rather, the issue is that there is a chance that ULP may start running when the RTC fast clock is not enabled yet, and it will run from RTC slow clock (150kHz or 32kHz) instead.

boarchuz, 2020-01-30:
Looks like that will be skipped because of this:
https://github.com/espressif/esp-idf/blob/f5b82c5b1f33f5287f9b5f1993594bf53755a4f2/components/esp32/sleep_modes.c#L661

Also output alternates between 
> D (162) sleep: RTC_PERIPH: ON, RTC_SLOW_MEM: ON, RTC_FAST_MEM: ON
and
> D (172) sleep: RTC_PERIPH: OFF, RTC_SLOW_MEM: ON, RTC_FAST_MEM: ON

So that's all fine, and the LED I put on the RTC pin does what you might expect: PD_ON = always on even in deep sleep; PD_AUTO = off in deep sleep (pin is hi-z I suppose) except for a ~105uS blink every time RTC powers on for ULP.
_____________________________________________________________________________

I've had a good try at breaking it by trying to run before fast clock is ready, but I can't no matter what I do. I've tried setting the wakeup interval to 0 cycles (removing internal checks), fiddling with RTC_CNTL_TIMER1_REG, RTC_CNTL_TIMER2_REG, RTC_CNTL_TIMER5_REG, etc. The only way I've managed to trigger the ULP to run at a slower clock speed is to disable the fast clock with rtc_clk_8m_enable(false, false) or set it to RTC_FAST_FREQ_XTALD4 (neither of which are relevant here).
In normal conditions, it seems the 8MHz is always ready when needed.

So I have more questions:

1. Do you have any tips for reproducing the issue?
2. For the sake of ULP wakeup, does it matter if the 8MHz is ready or not? As far as I can tell, the ULP is able to wake the SoC even if running at 150khz. Is the notice necessary?
3. Is it possible that this exception no longer applies at all? Perhaps, as ESP-IDF has matured, the RTC timing has been refined to the point that 8MHz has become reliable for this purpose? I've written several ULP bitbanging protocols with PD_ON without ever encountering this slow clock speed phenomenon. Is it worth reviewing? I think this functionality would be valuable but I might be in a tiny minority!
4. Why is the conflict mentioned in the ULP wakeup section? I think it has little to do with ULP wakeup specifically; it's more of a general ULP programming gotcha.

Thanks as always for your time and insight, @igrr

boarchuz, 2021-11-29:
I can't believe I hadn't noticed this before, but there is a delay, equivalent to the ULP wakeup period, after the first instruction if RTC peripherals are forced on and the SoC is in deep sleep. It's as if, every 2nd run, the ULP executes the first instruction then returns to sleep.

For example, with a wakeup period of 1000ms, with this program the pin would, bizarrely, be *low* also for 1000ms:
```
entry:
    set_pin_low
    set_pin_high
    halt
```

The solution is to increase RTC_CNTL_ULPCP_TOUCH_START_WAIT from the default 0x10, eg:
`REG_SET_FIELD(RTC_CNTL_TIMER2_REG, RTC_CNTL_ULPCP_TOUCH_START_WAIT, 0x20);`

Maybe this is the clock issue you observed @igrr? Maybe it's separate? Regardless, I'll close this now as officially unsupported, but maybe the above solution will help someone.

IDF: v5.0-dev-422-ga18cb70905

closed 2021-11-29 by boarchuz