espressif/esp-idf #4691: Task watchdog timeout due to btController when using Wi-Fi functions.

zacwbond, 2020-01-29:
## Environment

- Development Kit:      none (custom HW)
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.1-dev-2047-g600d542f5
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): gcc8_2_0-esp-2019r2-linux-amd64
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         external 3.3V

## Problem Description

I recently pulled the master branch to test out some Nimble fixes that I've been waiting on.  We're using BLE via the Nimble stack along with Wi-Fi (using software conexistence).  Both BLE and Wifi are pinned to core 0.

Almost immediately after boot, we bring up the radio in STA mode and join a network.  We have one task which configures the SSID and passphrase, then calls `esp_wifi_start()`.  We then wait until the event handler receives `WIFI_EVENT_STA_START` before calling `esp_wifi_connect()` in the event handler for that event.

The issue is that the call to `esp_wifi_connect()` takes around 15 seconds to return, and during that time I see the following message repeatedly:
```E (6047) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (6047) task_wdt:  - IDLE0 (CPU 0)
E (6047) task_wdt: Tasks currently running:
E (6047) task_wdt: CPU 0: btController
E (6047) task_wdt: CPU 1: IDLE1
```

No other tasks are able to run at this point.  It seems like btController is stuck in some kind of busy wait?  Eventually, whatever is going on finishes because both Wifi and BLE seem to work correctly (and simultaneously) after about 15 seconds.  

I have also observed the same problem with a Wifi scanning task.  In that case, the same error message appears during a call to `esp_wifi_scan_start()`.  

As a test, I progressively trimmed down our BLE configuration code.  I was able to delete everything except for the `esp_nimble_hci_and_controller_init()`.  If I remove that call, there are no problems using Wifi.  If I add that call, I see the error, even if I don't call any of the other Nimble initialization functions (e.g., `nimble_port_init()`, `nimble_port_freertos_init()`, etc.).  

I'd appreciate any insight.  Happy to provide logs, etc, although besides the task watchdog warning, there's not much to see.

zacwbond, 2020-01-30:
I have captured some better debug this morning, after I realized I was unintentionally disabling a lot of it just before connecting to the wifi network.  See the log below.  It was lightly edited to remove repeated messages, but I added comments to indicate where they were removed.

The most suspicious points are the long gap at around 1074 where it looks like something is going on (I'm not clear what these are--scan results showing a matching wifi network to join?) and then a 10 second pause around 20972.  It looks like a handshake was begun, but then there is a deauth and a leave.  

After that, there's an extremely long gap with nothing happening besides watchdog warnings between the "receive disconnect" notice and 10 seconds later when the event handler finally gets WIFI_EVENT_STA_DISCONNECTED.

```
// Message from host with SSID and PW to join network
E (1026) Network: Provision onto Wifi.

// Call esp_wifi_start() here. 
I (1026) wifi: connect status 0 -> 0
I (1026) wifi: Start wifi disconnect
I (1027) wifi: connect status 0 -> 0
I (1027) wifi: filter: set rx policy=8
I (1027) wifi: filter: set rx policy=2
I (1027) wifi: clear blacklist
I (1028) wifi: clear rc list
D (1028) event: running post WIFI_EVENT:3 with handler 0x40122204 on loop 0x3ffcf028
I (1028) wifi: flush txq
I (1029) wifi: stop sw txq
D (1030) esp_netif_handlers: esp_netif action stopped with netif0x3ffcfda8 from event_id=3
I (1031) wifi: filter: set rx policy=0
D (1031) esp_netif_lwip: check: remote, if=0x3ffcfda8 fn=0x401167a8

I (1032) wifi: lmac stop hw txq
D (1032) esp_netif_lwip: call api in lwip: ret=0x5002, give sem
D (1033) event: running post WIFI_EVENT:3 with handler 0x400e329c on loop 0x3ffcf028

// Wifi event handler receives WIFI_EVENT_STA_STOP
E (1034) ESP Wifi: Event received: WIFI_EVENT_STA_STOP
I (1035) wifi: Set ps type: 0

I (1035) wifi: clear blacklist
D (1036) phy_init: loading PHY init data from application binary
D (1036) nvs: nvs_open_from_partition phy 0
D (1037) nvs: nvs_get cal_version 4
D (1037) nvs: nvs_get_str_or_blob cal_mac
D (1038) efuse: coding scheme 0
D (1038) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 8 bit
D (1039) efuse: coding scheme 0
D (1039) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 0 bit
D (1040) efuse: coding scheme 0
D (1041) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 24 bit
D (1041) efuse: coding scheme 0
D (1042) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 16 bit
D (1043) efuse: coding scheme 0
D (1043) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 8 bit
D (1044) efuse: coding scheme 0
D (1044) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 0 bit
D (1045) efuse: coding scheme 0
D (1046) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 16 bit
D (1047) nvs: nvs_get_str_or_blob cal_data
D (1052) nvs: nvs_close 19
D (1052) efuse: coding scheme 0
D (1052) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 8 bit
D (1052) efuse: coding scheme 0
D (1053) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 0 bit
D (1054) efuse: coding scheme 0
D (1054) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 24 bit
D (1055) efuse: coding scheme 0
D (1055) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 16 bit
D (1056) efuse: coding scheme 0
D (1057) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 8 bit
D (1058) efuse: coding scheme 0
D (1058) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 0 bit
D (1059) efuse: coding scheme 0
D (1059) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 16 bit
I (1062) wifi: filter: set rx policy=0
I (1062) wifi: mode : sta (b4:e6:2d:d9:69:85)
[... 194 lines cut ...]

// At this point, BLE receives a request from host to begin scanning.
E (30701) BLE Central: Start scanning!
GAP procedure initiated: discovery; own_addr_type=0 filter_policy=0 passive=0 limited=0 filter_duplicates=1 duration=forever
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000b len=7
0x0b 0x20 0x07 0x01 0x50 0x00 0x30 0x00 0x00 0x00
E (30705) Heap: 100368
E (30705) HeapMin: 84296
D (30705) event: running post WIFI_EVENT:5 with handler 0x40122324 on loop 0x3ffcf028
D (30705) esp_netif_handlers: esp_netif action disconnected with netif0x3ffcfda8 from event_id=5
D (30707) esp_netif_lwip: check: remote, if=0x3ffcfda8 fn=0x4011681c

D (30707) esp_netif_lwip: esp_netif_down_api esp_netif:0x3ffcfda8
D (30708) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffcfda8
Command complete: cmd_pkts=5 ogf=0x8 ocf=0xbD (30709) esp_netif_lwip: if0x3ffcfda8 start ip lost tmr: no need start because netif=0x3ffcfe4c interval=120 ip=0
 status=0
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000c len=2
0x0c 0x20 0x02 0x01 D (30712) esp_netif_lwip: call api in lwip: ret=0x0, give sem
0x01
D (30713) event: running post WIFI_EVENT:5 with handler 0x400e329c on loop 0x3ffcf028

// Wifi event handler receives WIFI_EVENT_STA_DISCONNECTED
E (30713) ESP Wifi: Event received: WIFI_EVENT_STA_DISCONNECTED
E (30714) ESP Wifi: SYSTEM_EVENT_STA_DISCONNECTED: reason: WIFI_REASON_4WAY_HANDSHAKE_TIMEOUT (15)
```

lagrecjo, 2020-02-04:
I have the same issue using coexist with WIFI and BR/EDR SPP protocol:
"task_wdt: CPU 0: btController"
I tried just starting the BT, waiting for an SPP connection, then starting the WIFI. The WIFI starts then the same result with the WDT on the bt controller task.
Did you ever arrive at a solution?

zacwbond, 2020-02-04:
> I have the same issue using coexist with WIFI and BR/EDR SPP protocol:
> "task_wdt: CPU 0: btController"
> I tried just starting the BT, waiting for an SPP connection, then starting the WIFI. The WIFI starts then the same result with the WDT on the bt controller task.
> Did you ever arrive at a solution?

Nope, but the change that broke things must have been since September or so.  Everything was fine when I first ported my code to Nimble; it's only with the latest master that I've seen this error.

I haven't yet tried to step back to find the first instance of breakage.

zacwbond, 2020-02-10:
I am closing this issue because the following is a clear duplicate:

closed 2020-02-10 by zacwbond