espressif/esp-idf #4693: bin file reading issue

sowmimalu, 2020-01-30:
Hai ,
 i working on gatt client example to send a data to other controller (nrf52840) . i am paring esp32 and nrf using ble . then i need to update nrf  on ota ...for that i am using esp32 to sending files to nrf ,to send my firmware file i kept my bin file in local server and downloading  bin file using http and the writing whole data in file(using fat file system) ...i can able to read data from file in hex format ..
but i need to seperate my file as several objects .. and to send through ble 
how do i parse my file as packets ????
can  you please tell me??
thanks ,

0xjakob, 2020-02-06:
Hi @sowmimalu, thanks for posting the issue. 
This sounds like a general programming question. If that's the case, please consider asking this in the esp forum or any other programming forum.
If it's not the case, please add some more specific information where you have problems with the esp-idf (code, idf version, hardware, etc... the issue template will help you).

Thank you.

mahavirj, 2020-02-07:
@sowmimalu As @0xjakob mentioned please file new issue with more detailed information, I will go ahead and close this one.

closed 2020-02-07 by mahavirj