espressif/esp-idf #4694: Bluetooth Controller: Classic Link fails after Role Change from Master to Slave

mringwal, 2020-01-30:
## Environment

- Development Kit:      ESP32-Thing
- IDF version v4.1-dev-1757-g826b9f638
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Power Supply: USB

## Problem Description

When connection from ESP32 to an iPhone as HFP HF with Role Switch Allowed, the link fails after encryption and role change by iPhone.

Without the Role Switch Allowed as Link Policy, the connection works as expected.

An older iPhone 5S with iOS 9 does not request a role change, but it happens on an iPhone SE with iOS 13.3. I expect the iOS version to be relevant for this.

### Expected Behavior

I don't expect the link to fail after the role change.

### Actual Behavior

Link fails

### Steps to reproduce

Follow BTstack instructions for ESP32: https://github.com/bluekitchen/btstack/blob/master/port/esp32/README.md

- Switch to btstack/port/esp32/example/hfp_hf_demo
- In main/hfp_hf_demo.c, set the `device_addr_string` to your iPhone
- In main/hfp_hf_demo: add this line to enable role switch in btstack_main():

    `gap_set_default_link_policy_settings( LM_LINK_POLICY_ENABLE_ROLE_SWITCH );`

- enable HCI log in ESP-IDF/components/btstack/main.c by uncommenting in app_main()

    `hci_dump_open(NULL, HCI_DUMP_STDOUT);` 

- start demo and connect to iPhone by pressing 'a'

## Debug Logs

HCI Dump for Wireshark

[link-loss-after-role-change.pklg.zip](https://github.com/espressif/esp-idf/files/4135137/link-loss-after-role-change.pklg.zip)

AbnerFederer, 2020-02-19:
Hi, @mringwal 

I tested with iPhone 6/6s/8/8p/xs max/11(with the iOS version above 12.0.0).
Form the air log I scratched, iPhone always asks for the `Master` role while an ACL link or SCO link.

Thanks
Abner

mringwal, 2020-02-19:
@AbnerFederer Hi! You write that iPhone with iOS >= 12.0.0 initiates a role change to become Master with your devices. That's the same as with my iPhone SE.

After the role change, is the link stable? If yes, what example did you try + can you share an air trace?
 
Thanks Matthias

AbnerFederer, 2020-02-19:
hi 

I use the hfp_hf demo as master to initiate a link with iPhone. But iPhone always acquires role switch. 

The mentioned iPhones are all right except iPhone 6/6s.

Thanks
Abner 

发自我的iPhone

mringwal, 2020-02-19:
I'm glad role change works with iPhone 8 and newer. Any idea what's the problem with iPhone 6/6S/SE? Could you post air traces of working/non-working setup to compare (if the tool to view the air trace is free)?

Thanks Matthias

AbnerFederer, 2020-02-20:
Hi, @mringwal 

I misunderstand you.
From my test, the ACL link and SCO link between ESP32(HF Unit) and iPhone 6/6s and newer are good with role switch. And we have done debug on role switch in lib file,  so maybe you can wait for the update then try again.

Thanks
Abner

mringwal, 2020-02-20:
Hi @AbnerFederer 

To clarify: Have you seen an issue with role switch triggered by iPhone with the Bluetooth lib file on current master, and, then made changes that fixed this issue, but the fix is not public yet?

A customer needs to connect to two iPhones (either role) and this didn't work. While testing, I did try the role switch, which lead to this issue.

Could you also check out the original issue please? 
https://github.com/espressif/esp-idf/issues/4695

mringwal, 2020-02-28:
I'm happy to confirm that https://github.com/espressif/esp-idf/commit/d11a86076f33282184080c4503549ba521b11df0 allows role switch with iPhone SE and iPhone 6S. 

In addition, it's also possible to connect to two iPhones in Slave role, while this is not possible in Master mode, see https://github.com/espressif/esp-idf/issues/4695.

closed 2020-02-28 by mringwal