espressif/esp-idf #4695: Bluetooth Controller: Second classic HCI link to iPhone not working in Master Role

mringwal, 2020-01-30:
## Environment

- Development Kit:      ESP32-Thing
- IDF version v4.1-dev-1757-g826b9f638
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Power Supply: USB

## Problem Description

When connection from ESP32 to two iPhone as HFP HF, the second link only works for a few ACL packets before it gets disconnected with a supervision timeout, or by a HCI disconnect from the remote as L2CAP protocol timeouts are hit because ACL packets are not correctly send/received.
 
The test app can connect to both iPhones without problems. Also, in some tests, two Android phones worked as well. It seems to fail if the first connected device is an iPhone with a newer iOS version.

Used devices:  
- first device: iPhone SE, with iOS 13.3
- second device: iPhone 5S, with iOS 9.0.2

### Expected Behavior

I expect that both connections work fine as the ESP32 is in Master role of the Piconet and should be able to handle multiple slave devices.

### Actual Behavior

Link doesn't work

### Steps to reproduce

Follow BTstack instructions for ESP32: https://github.com/bluekitchen/btstack/blob/master/port/esp32/README.md

- Switch to btstack/port/esp32/example/hfp_hf_demo
- Replace main/hfp_hf_demo.c with the attached one.
- In main/hfp_hf_demo.c, set `device_addr_string_0` and `device_addr_string_1` to your iPhones
- enable HCI log in ESP-IDF/components/btstack/main.c by uncommenting in app_main()

    `hci_dump_open(NULL, HCI_DUMP_STDOUT);` 

- start demo. It automatically connects to the first device and then to the second device. Make sure that your in the iPhone Settings->Bluetooth menu to be able to accept the pairing request.

[hfp_hf_demo.c.zip](https://github.com/espressif/esp-idf/files/4135284/hfp_hf_demo.c.zip)

## Debug Logs

HCI Dump for Wireshark

[hfp-to-2-iPhones.pklg.zip](https://github.com/espressif/esp-idf/files/4135286/hfp-to-2-iPhones.pklg.zip)

Wth-Esp, 2021-01-20:
Hi @mringwal 

I have no iPhone SE and iPhone 5s. Does any other phone can reproduce the issue?

Thanks

mringwal, 2021-01-20:
Hi @Wth-Esp, thanks for picking this up. We've got new iPhones (iPhone SE 2nd and iPhone 12 Mini), I will check if this shows up with them as well and or try other devices as well.

Wth-Esp, 2021-01-20:
@mringwal  OK, we have some android phones. So, let's share the test result here.

xiongweichao, 2021-01-21:
Hi @mringwal 

Could you please provide running logs?

Thanks

mringwal, 2021-01-25:
HI there!

I've update esp-idf to current master branch and tried to reproduce the issue with the devices I've used last year. 

Good news! The issue does not show up anymore.

The only changes apart from the esp-idf update was an update of the iPhone SE to 14.2. Therefore, I would conclude that the underlying issue has been fixed in the Bluetooth subsystem last year by Espressif. 

Excellent! Thanks!

closed 2021-01-25 by mringwal

Wth-Esp, 2021-01-26:
@mringwal 
What good news! Congrats! 
And we are testing with BTStack's demo to find some other issues, to make the ESP32 series chip better and better. Thanks for your support.

mringwal, 2021-01-26:
@Wth-Esp thanks, that's great. my main issue with the ESP32 ist that I don't know how to send SCO packets over VHIC correctly. Please have a look here https://github.com/espressif/esp-idf/issues/1118

The question is this: after configuring SCO over VHCI, when must the next SCO packet be sent? 

SCO flow control isn't supported as far as I know and I didn't fully read through the Bluedroid sources, and also didn't try the Bluedroid examples to see if outgoing audio works without glitches.

Wth-Esp, 2021-01-26:
@mringwal  

We know that the API of which datapath for HFP is not so effective for the development and usually make developer confused. That is a bad thing. But please give us a little more time, we have discussed how to fix this. And also our internal test cases of HFP become more and more strict. In 2021, one of our targets is to make audio applications easier to use, including A2DP and HFP.

mringwal, 2021-01-26:
@Wth-Esp I didn't realize that SCO Flow Control is supported now (I think it wasn't in 2018 when I tried first). With SCO Flow Control, it's clear when to send data. In fact, our hfp_hf_demo now works on the ESP32, although there are glitches in the HFP audio received on an iPhone. I'll create a new issue for that. Thanks!