espressif/esp-idf #4696: Correct way to set CXX flags in CMakeLists build system

thecashit, 2020-01-31:
Hello!

I don't think that it is a bug, but not a feature request at all.
After update to 4.0-rc we tried to use new build system to be up to date.

Before we use component.mk file to set CXX flags belongs to required component.
`CXXFLAGS += -D LOG_LOCAL_LEVEL=ESP_LOG_VERBOSE`

After updating build system we founded in documentation that we need to use  equivalent CMake commands instead to set CFLAGS, CPPFLAGS, CXXFLAGS.

So, CMakeLists.txt in my case seem like following:

```
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -D LOG_LOCAL_LEVEL=ESP_LOG_VERBOSE")

set(srcs 
    "1.cpp"
    "2.cpp")

idf_component_register(SRCS "${srcs}"
                    INCLUDE_DIRS "."
                    PRIV_INCLUDE_DIRS "."
                    REQUIRES " .... ")
```
 
And it didn't worked.
Please suggest what is correct way to set CXX flags in new build system.
Thnx.

renzbagaporo, 2020-02-03:
Hi, we also use `target_compile_options` to add flags. Can you verify the output `build/compile_commands.json` (has the list of compile commands for each file)?

thecashit, 2020-02-03:
Hi,
Seems like it was my local problem.
I tried to fresh install ESP-IDF 4.0rc after deleting everything and now 
`target_compile_options(${COMPONENT_LIB} PRIVATE "-D LOG_LOCAL_LEVEL=ESP_LOG_VERBOSE") ` works as exptected.
And yes... i can see this flags in build/compile_commands.json

Thnx.
Close this issue

closed 2020-02-03 by thecashit