espressif/esp-idf #4698: ESP32-DevkitC v4 - Wifi unstable, not working

myucesan, 2020-01-31:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for DevKitC): [v4]
- Module or chip used:  [ESP32-WROVER-B]
- IDF version:
    Arduino the latest board manager version 1.0.4. ESP-IDF release v4, MicroPython(esp32-idf4-20200131-v1.12-107-g31ba06ce8.bin, built on ESP-IDF release v4)
- Build System:         [CMake for ESP-IDF]
- Compiler version : 8.2.0 for ESP-IDF
- Operating System:     [Windows]
- (Windows only) environment type: [MSYS2 mingw32 (installed) but using through Visual GDB].
- Using an IDE?: [Yes, Arduino IDE for arduino-esp32, Visual Studio (with Visual GDB addon] for ESP-IDF and just REPL with Micropython.
- Power Supply:         [USB]

## Problem Description

I am having an issue with a wifi connection. I am using **ESP32_DevKitC_V4**, so the fourth revision of ESP32. It is the Wrover-B version. I bought it from [here](https://eu.mouser.com/ProductDetail/Espressif-Systems/ESP32-DevKitC-VB?qs=sGAEpiMZZMuJ3l9lTgMBp1ZUJQFPkBG29IbyAlyFBKGOK2c6lCs%2FVw==).

I tried to connect to multiple routers using Arduino-ESP32, ESP-IDF and even MicroPython. I did this just make sure it is not the software. The routers all offer a 2.4gHz signal and support the 802.11 b/g/n protocols. There is no firewall activated on the routers. I tried using different AP names different passwords and even without a password. I tried setting different channels on the router. The strange thing if after making some changes the ESP32 connected to the router but I could not reproduce it later on. It sometimes works but mostly does not and if it works I cannot reproduce the solution. It only works once. 

### Expected Behavior
It should give a message that it is connected to AP and that it received an IP.
### Actual Behavior
It continually tried to reconnect and often claims the AP name cannot be found.
### Steps to reproduce

1. Open Arduino, Visual Studio or REPL (for micropython)
2. Run example codes for ESP32 for wifi connection after setting SSID & password correctly.

### Code to reproduce this issue

1. [Arduino code](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiClient/WiFiClient.ino)
2. [ESP-IDF code](https://github.com/espressif/esp-idf/tree/release/v4.0/examples/wifi/scan)
3. [MicroPython code](https://docs.micropython.org/en/latest/esp8266/tutorial/network_basics.html)

## Debug Logs
```
For Arduino, verbose debug, it keeps repeating:

[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 2 - AUTH_EXPIRE
.....[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
....[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[... 320 lines cut ...]
I (1653198) wifi: new:<8,0>, old:<8,2>, ap:<255,255>, sta:<8,2>, prof:1
I (1653198) wifi: STA_DISCONNECTED, reason:2
I (1655248) wifi: STA_DISCONNECTED, reason:205
I (1655378) wifi: new:<8,2>, old:<8,0>, ap:<255,255>, sta:<8,2>, prof:1
I (1655378) wifi: state: init -> auth (b0)
I (1656378) wifi: state: auth -> init (200)
I (1656378) wifi: new:<8,0>, old:<8,2>, ap:<255,255>, sta:<8,2>, prof:1
I (1656378) wifi: STA_DISCONNECTED, reason:2
I (1658428) wifi: STA_DISCONNECTED, reason:201
no AP found
I (1660488) wifi: STA_DISCONNECTED, reason:201
no AP found
I (1660608) wifi: new:<8,2>, old:<8,0>, ap:<255,255>, sta:<8,2>, prof:1
I (1660608) wifi: state: init -> auth (b0)
I (1661608) wifi: state: auth -> init (200)
I (1661608) wifi: new:<8,0>, old:<8,2>, ap:<255,255>, sta:<8,2>, prof:1
I (1661608) wifi: STA_DISCONNECTED, reason:2
I (1663668) wifi: STA_DISCONNECTED, reason:205
I (1665718) wifi: STA_DISCONNECTED, reason:201
no AP found
```

## Other items if possible

- [sdkconfig.zip](https://github.com/espressif/esp-idf/files/4138840/sdkconfig.zip)

myucesan, 2020-02-10:
I just tried an older revision of the ESP32 DevKit and it works just fine. I am afraid the WiFi module of the newest revision is broken so do not buy [this](https://nl.mouser.com/ProductDetail/Espressif-Systems/ESP32-DevKitC-VB?qs=sGAEpiMZZMuJ3l9lTgMBp1ZUJQFPkBG29IbyAlyFBKGOK2c6lCs%2FVw==_ one.

esp-HarveyRong, 2020-03-27:
Hi @myucesan, 
According to the log analysis you provided, ESP32 did not receive the Auth frame, which caused the ESP32 connection to be unsuccessful and added the route to the blacklist.
```
I (1916) wifi: state: init -> auth (b0)
I (1916) wifi: start 1s AUTH timer
I (1916) wifi: clear scan ap list
I (1926) wifi: rsn valid: gcipher=1 ucipher=3 akm=5
I (2916) wifi: auth timeout
I (2916) wifi: state: auth -> init (200)
I (2916) wifi: connect status 1 -> 4
I (2916) wifi: stop beacon/connect timer
I (2926) wifi: reason: auth expire(2)
I (2926) wifi: add bssid 74:da:da:4a:54:71 to blacklist, cnt=0
```
There are two possibilities for ESP32 not receiving a routed Auth frame:
1. The route was sent but ESP32 was not received.
2. The AP did not sent.

The best way is that you can provide WiFi packets.
The recommended Macbook's built-in packet capture tool does not require a packet capture card. It can be used to capture 802.11a/b/g/n/ac packets: 
https://osxdaily.com/2015/04/23/sniff-packet-capture-packet-trace-mac-os-x-wireless-diagnostics/
Or you can use the tutorial on the official Espressif website to capture packets using Wireshark
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wireshark-user-guide.html

Alvin1Zhang, 2020-04-16:
@myucesan Would you please help share if any updates or details for this issue? Thanks.

myucesan, 2020-04-16:
> @myucesan Would you please help share if any updates or details for this issue? Thanks.

I am currently not in the possession of the ESP32 Devkit due to the current pandemic and I will not retrieve it anytime soon. It was a side project. I personally think the AP **did not sent** because I believe it could not find the SSID it had to connect to.

If we check the Arduino Logs you can see the following:

`[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 2 - AUTH_EXPIRE
.....[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND`

This repeats for a while and it keeps giving reason 201 in the Arduino logs. In ESP-IDF the following keeps repeating:
`I (796) wifi: scan specific ssid=Delta-Proto-2
I (796) wifi: start max timer
I (916) wifi: scan end: arg=0x0, status=0, ss_state=0x3
I (916) wifi: oper_channel: arg=0x0 time=205928
I (916) wifi: perform scan: ss_state=0x9, chan<2,0>, dur<0,120>
I (926) wifi: change: chan<2,0>, dur<0,120>
I (926) wifi: ht20 freq=2417, chan=2
I (926) wifi: enter start op, arg=0x3ffb6318
I (936) wifi: scan operation: state=0x3, chan<2,0>, arg=0x3ffb6318, status=0
I (936) wifi: scan specific ssid=Delta-Proto-2
I (946) wifi: start max timer
I (1066) wifi: scan end: arg=0x0, status=0, ss_state=0x3
I (1066) wifi: oper_channel: arg=0x0 time=354645
I (1066) wifi: perform scan: ss_state=0x9, chan<3,0>, dur<0,120>
I (1076) wifi: change: chan<3,0>, dur<0,120>
I (1076) wifi: ht20 freq=2422, chan=3
I (1076) wifi: enter start op, arg=0x3ffb6318
I (1086) wifi: scan operation: state=0x3, chan<3,0>, arg=0x3ffb6318, status=0`

It cannot seem to find it. In Micropython it is the following:

`
I (1588608) wifi: STA_DISCONNECTED, reason:2
I (1590658) wifi: STA_DISCONNECTED, reason:205
I (1590778) wifi: new:<8,2>, old:<8,0>, ap:<255,255>, sta:<8,2>, prof:1
I (1590788) wifi: state: init -> auth (b0)
I (1591788) wifi: state: auth -> init (200)
I (1591788) wifi: new:<8,0>, old:<8,2>, ap:<255,255>, sta:<8,2>, prof:1
I (1591788) wifi: STA_DISCONNECTED, reason:2
I (1593838) wifi: STA_DISCONNECTED, reason:205`

It continually gives reason 2, which is AUTH_EXPIRE and reason 205. Reason 205 seems to be a general connection fail. It really seems it cannot find the SSID which would mean it did not send the AUTH frame. While I cannot offer any closure by providing WiFi packets capture it seems clear that the AUTH did not sent then.

Thank you!

esp-HarveyRong, 2020-04-24:
> It continually gives reason 2, which is AUTH_EXPIRE and reason 205. Reason 205 seems to be a general connection fail.

Hi @myucesan,  
Yes, the essence is still because ESP32 did not receive AUTH frames, of course, if you can capture WiFi packets, you can further confirm.
Event 205 is because ESP32 scanned the matching AP, but because the AP is in the blacklist, this event is returned. The reason why the AP entered the blacklist is that ESP32 did not receive the AUTH frame.

Alvin1Zhang, 2020-05-26:
@myucesan Thanks for reporting, would you please help share if any updates for the issue? Thanks.

enricop, 2020-06-08:
Hi,
we also had reason 205 and 2. Could be that the AP blacklisted the esp32 module?

```
minimum internal_dma free mem: 72176
I (172599) hap_platform_network: Wi-Fi Event: 5
I (172849) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (172849) wifi:state: init -> auth (b0)
I (173849) wifi:state: auth -> init (200)
I (173849) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
W (173849) wifi_event_handler: WIFI DISCONNECTED from SSID RUT950_78D6 for REASON 2
I (173859) led_add_event: Adding event #12
I (173859) netstatus: setting net_status to 3
minimum internal_dma free mem: 72176
I (173869) hap_platform_network: Wi-Fi Event: 5
W (177019) wifi_event_handler: WIFI DISCONNECTED from SSID RUT950_78D6 for REASON 205
```
thanks for your support

esp-HarveyRong, 2020-06-10:
Hi @enricop,
Can you help provide Wi-Fi data package?
The Log shows that ESP32 did not receive the AP's Auth frame within the specified time. 
There are two possibilities: 

- The AP was not sent.
- The AP was sent, and the ESP was not received.

Alvin1Zhang, 2020-08-18:
Thanks for reporting and updates, would you please help share more details/updates for the issue? Thanks.

closed 2020-11-02 by Alvin1Zhang