espressif/esp-idf #4699: Master & slave modbus use the same timer index

ClementMarrast, 2020-01-31:
## Environment

- Module or chip used:  ESP32-WROOM-32D
- IDF version: v4.1-dev-1094-g1938a459e
- Build System:         CMake
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Windows
- (Windows only) environment type: PowerShell.
- Using an IDE?: Yes (visual studio code)
- Power Supply:         external 3.3V

## Problem Description
The slave and master use the same timer index and time index group.
porttimer_m.c: l.53:
// Timer group and timer number to measure time (configurable in KConfig)
#define MB_TIMER_INDEX          (CONFIG_FMB_TIMER_INDEX)
#define MB_TIMER_GROUP          (CONFIG_FMB_TIMER_GROUP)
static const USHORT usTimerIndex = MB_TIMER_INDEX
static const USHORT usTimerGroupIndex = MB_TIMER_GROUP

porttimer.c l.65:
static const USHORT usTimerIndex = CONFIG_FMB_TIMER_INDEX
static const USHORT usTimerGroupIndex = CONFIG_FMB_TIMER_GROUP

### Expected Behavior
Slave and master should use their own timer to detect timeout on communication.

### Actual Behavior
The first to be initialized will not manage the communication timeout. 

### Steps to reproduce
1. Initialize master: 
```c
mbc_master_init()
```
2. Initialize slave:
```c
mbc_slave_init()
```
3. Create a timeout of the master communication
The master register callback from porttimer_m.c will not be called
```c
static void IRAM_ATTR vTimerGroupIsr(void *param)
```

Alvin1Zhang, 2020-03-30:
Thanks for reporting and sorry for slow turnaround, we will look into.

alisitsyn, 2020-03-30:
Hi @ClementMarrast,
Thanks for this issue. Could you provide more information. Are you trying to initialize serial master and slave port together? There is a constraint is that only one instance of modbus port can be initialized and used at a time.

Alvin1Zhang, 2020-05-26:
@ClementMarrast Thanks for reporting. Would you please help share if any updates for the issue? Thanks.

ClementMarrast, 2020-05-26:
Hi @alisitsyn and @Alvin1Zhang,

Sorry for the late reply, I'm using serial master on one port and serial slave on another port.
To solve the issue, I've used the following settings for master:
`#define MB_TIMER_INDEX          (CONFIG_FMB_TIMER_INDEX +1)
#define MB_TIMER_GROUP          (CONFIG_FMB_TIMER_GROUP +1)`
But it's not a good fix, I'm not using parameters from menuconfig.

Alvin1Zhang, 2021-08-24:
Thanks for reporting, sorry for very slow turnaround, fix on master is available at https://github.com/espressif/esp-idf/commit/59444334249d4940b59795c8e0df779ec25333bd, we are now back porting the fix to other release branches, feel free to reopen.

closed 2021-08-24 by Alvin1Zhang