espressif/esp-idf #4701: Ignore some data in https_requst

DheeranYuvaraj, 2020-02-02:
Hi,

I am trying https_requst example code for receiving json from web server. 
I am able to recive the json, but the received json comes alosn with some data. 
Example output:
205 Bytes read
HTTP/1.1 200 OK
Date: Sun, 02 Feb 2020 13:22:39 GMT
Server: Apache
Connection: close
Content-Type: text/html; charset=UTF-8
xaxmxjx9x{"i":"am001", "i":"am002", "i":"am003", "i":"am009", "i":"am008"}
[1B][0;32mI (11616) https: connection closed[1B][0m

Some one advice me how to extract json({"i":"am001", "i":"am002", "i":"am003", "i":"am009", "i":"am008"} data from this above output.

0xjakob, 2020-02-06:
Hi @DheeranYuvaraj!
You can use cJSON. It's available in the IDF components already. 
I also used the http_request example to extract a json a few weeks ago. You just feed the buffer with the response (`buf`) into cjson.
Unfortunately, we don't have a cjson example (or I can't find it right now).
Hope that helps.

0xjakob, 2020-02-06:
OK, I think I missunderstood a bit. So to separate the data, something like `char* json = strstr(buf, "{\"")` could work. This is btw a basic programming question and it not specific for the `esp-idf`. Please consider posting in the forum the next time :)

DheeranYuvaraj, 2020-02-07:
Hi @0xjakob,

Thanks for the information. closing now.

closed 2020-02-07 by DheeranYuvaraj