espressif/esp-idf #4702: esp_bt and bt

rochamorelli, 2020-02-02:
INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit: [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):
    v4.1-dev-2055-gf5b82c5b1
- Build System: [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System: [Linux]
- Using an IDE?: [Yes VSCode Insiders just as text editor, running idf.py on a headless Ubuntu container]
- Power Supply: [USB]

## Problem Description
When including header file on my component to use BLE features, have to add `esp_bt.h` but on CMakeLists.txt of this same component I have to add `REQUIRES bt`.

Took me some time to study cmake and esp-id organization to figure out why I was having error `Failed to resolve component 'esp_bt'.`

tree for the file organization
```
esp-idf
└── components
    └── bt
        ├── include
        |   └── esp_bt.h
        └── controller
            └── bt.c
```

### Expected Behavior
Component and header have the same name
### Actual Behavior
Component is `bt` and header is `esp_bt.h`.
### Steps to reproduce
1. Include `esp_bt.h` in your module
2. Include `esp_bt` in your `REQUIRE` directive on `CMakeFiles.txt` of a component
### Code to reproduce this issue

```c
#include "esp_bt.h"
```

```cmake
idf_component_register(SRCS "ble_uart.c"
                    INCLUDE_DIRS "include"
                    REQUIRES esp_bt nvs_flash)
```

## Debug Logs

```
root@aeaabd925802:/home/esp/X# idf.py build
Executing action: all (aliases: build)
Running ninja in directory /home/esp/X/build
Executing "ninja all"...
[0/1] Re-running CMake...
-- Building ESP-IDF components for target esp32
CMake Error at /home/esp/esp-idf/tools/cmake/build.cmake:186 (message):
  Failed to resolve component 'esp_bt'.
Call Stack (most recent call first):
  /home/esp/esp-idf/tools/cmake/build.cmake:212 (__build_resolve_and_add_req)
  /home/esp/esp-idf/tools/cmake/build.cmake:213 (__build_expand_requirements)
  /home/esp/esp-idf/tools/cmake/build.cmake:426 (__build_expand_requirements)
  /home/esp/esp-idf/tools/cmake/project.cmake:395 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "/home/esp/X/build/CMakeFiles/CMakeOutput.log".
FAILED: build.ninja 
/usr/bin/cmake -H/home/esp/X -B/home/esp/X/build
ninja: error: rebuilding 'build.ninja': subcommand failed
ninja failed with exit code 1
```

## Other items if possible
- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

Wth-Esp, 2020-05-12:
Hi @rochamorelli 

Please add `set(srcs "controller/bt.c")` (referring to $IDF_PATH/components/bt/CMakeLists.txt) and have a try.

Thanks
Weitianhua

Alvin1Zhang, 2020-05-26:
@rochamorelli Thanks for reporting and sorry for the slow turnaround. Would you please help share if any updates for the issue? Thanks.

closed 2021-02-26 by Alvin1Zhang