espressif/esp-idf #4704: _WindowOverflow4 and Failed to configure netif

surfskidude, 2020-02-02:
## Environment
Development Kit: ESP32-WROVER-B
Module or chip used: WINGONEER T-Koala ESP32
IDF version: v4.1-dev-2055-gf5b82c5b1
Build System: CMake
Compiler version: crosstool-NG esp-2019r2) 8.2.0
Operating System: Linux
Power Supply: USB

## Problem Description

Can someone help me understand the following problems.
The following is the text printed in red from the full trace below:

0x40080000: _WindowOverflow4 at /home/surfskidude/esp/esp-idf/components/freertos/xtensa_vectors.S:1778
0x400d0020: _stext at ??:?
0x400870a8: lmacTxFrame at ??:?
0x400811c8: call_start_cpu1 at /home/surfskidude/esp/esp-idf/components/esp32/cpu_start.c:275
E (989) esp_netif_lwip: esp_netif_new: Failed to configure netif with config=0x3ffc0070 (config or if_key is NULL or duplicate key)

Full trace:

make -j8  monitor
Toolchain path: /home/surfskidude/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: esp-2019r2
Compiler version: 8.2.0
Python requirements from /home/surfskidude/esp/esp-idf/requirements.txt are satisfied.
Project is not inside a git repository, or git repository has no commits
will not use 'git describe' to determine PROJECT_VER.
App "file_server" version: 1
Python requirements from /home/surfskidude/esp/esp-idf/requirements.txt are satisfied.
MONITOR
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7144
load:0x40078000,len:14864
ho 0 tail 12 room 4
load:0x40080400,len:4568
entry 0x400806f0
I (31) boot: ESP-IDF v4.1-dev-2055-gf5b82c5b1-dirty 2nd stage bootloader
I (31) boot: compile time 13:36:40
I (32) boot: chip revision: 1
I (36) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (43) boot.esp32: SPI Speed      : 40MHz
I (47) boot.esp32: SPI Mode       : DIO
I (52) boot.esp32: SPI Flash Size : 4MB
I (56) boot: Enabling RNG early entropy source...
I (62) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (80) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (88) boot:  2 factory          factory app      00 00 00010000 00200000
I (95) boot: End of partition table
I (99) boot_comm: chip revision: 1, min. application chip revision: 0
I (106) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x45560 (284000) map
I (217) esp_image: segment 1: paddr=0x00055588 vaddr=0x3ffb0000 size=0x039d8 ( 14808) load
I (223) esp_image: segment 2: paddr=0x00058f68 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/surfskidude/esp/esp-idf/components/freertos/xtensa_vectors.S:1778

I (224) esp_image: segment 3: paddr=0x00059370 vaddr=0x40080400 size=0x06ca8 ( 27816) load
I (244) esp_image: segment 4: paddr=0x00060020 vaddr=0x400d0020 size=0xe637c (942972) map
0x400d0020: _stext at ??:?

I (582) esp_image: segment 5: paddr=0x001463a4 vaddr=0x400870a8 size=0x0ebe0 ( 60384) load
0x400870a8: lmacTxFrame at ??:?

I (621) boot: Loaded app from partition at offset 0x10000
I (621) boot: Disabling RNG early entropy source...
I (621) cpu_start: Pro cpu up.
I (625) cpu_start: Application information:
I (630) cpu_start: Project name:     file_server
I (635) cpu_start: App version:      1
I (639) cpu_start: Compile time:     Feb  2 2020 13:36:39
I (646) cpu_start: ELF file SHA256:  15c900c9c8236301...
I (652) cpu_start: ESP-IDF:          v4.1-dev-2055-gf5b82c5b1-dirty
I (658) cpu_start: Starting app cpu, entry point is 0x400811c8
0x400811c8: call_start_cpu1 at /home/surfskidude/esp/esp-idf/components/esp32/cpu_start.c:275

I (0) cpu_start: App cpu up.
I (669) heap_init: Initializing. RAM available for dynamic allocation:
I (676) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (682) heap_init: At 3FFBA100 len 00025F00 (151 KiB): DRAM
I (688) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (694) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (701) heap_init: At 40095C88 len 0000A378 (40 KiB): IRAM
I (707) cpu_start: Pro cpu start user code
I (725) spi_flash: detected chip: generic
I (726) spi_flash: flash io: dio
I (726) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (829) wifi: wifi driver task: 3ffc1a94, prio:23, stack:3584, core=0
I (829) system_api: Base MAC address is not set
I (829) system_api: read default base MAC address from EFUSE
I (849) wifi: wifi firmware version: 94efb94
I (849) wifi: config NVS flash: enabled
I (849) wifi: config nano formating: disabled
I (849) wifi: Init dynamic tx buffer num: 32
I (859) wifi: Init data frame dynamic rx buffer num: 32
I (859) wifi: Init management frame dynamic rx buffer num: 32
I (869) wifi: Init management short buffer num: 32
I (869) wifi: Init static rx buffer size: 1600
I (879) wifi: Init static rx buffer num: 10
I (879) wifi: Init dynamic rx buffer num: 32
I (889) example_connect: Connecting to DD-WRT...
I (979) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (989) wifi: mode : sta (b4:e6:2d:df:63:01)
E (989) esp_netif_lwip: esp_netif_new: Failed to configure netif with config=0x3ffc0070 (config or if_key is NULL or duplicate key)
I (999) example_connect: Waiting for IP
I (3049) example_connect: Wi-Fi disconnected, trying to reconnect...
I (5099) example_connect: Wi-Fi disconnected, trying to reconnect...
I (7149) example_connect: Wi-Fi disconnected, trying to reconnect...
I (9199) example_connect: Wi-Fi disconnected, trying to reconnect...
I (11259) example_connect: Wi-Fi disconnected, trying to reconnect...
I (13309) example_connect: Wi-Fi disconnected, trying to reconnect...
I (15359) example_connect: Wi-Fi disconnected, trying to reconnect...
I (17409) example_connect: Wi-Fi disconnected, trying to reconnect...
I (19459) example_connect: Wi-Fi disconnected, trying to reconnect...
I (21519) example_connect: Wi-Fi disconnected, trying to reconnect...
I (23569) example_connect: Wi-Fi disconnected, trying to reconnect...
I (25619) example_connect: Wi-Fi disconnected, trying to reconnect...
I (27669) example_connect: Wi-Fi disconnected, trying to reconnect...

david-cermak, 2020-02-03:
Hi @surfskidude 

Can you please share some portions of your initialisation code?
The `Failed to configure netif` error suggests that an incorrect configuration was supplied to `esp_netif_new()`.

It looks like (from the logs `... Wi-Fi disconnected, trying to reconnect...` ) a standard example based initialization was used calling `esp_netif_create_default_wifi_sta();` -- Is it correct?
In that case, one must make sure that the function is not called multiple times (since it creates a netif instance with default configuration) -- Is it possible that the function gets called more than once?
Even if the above was true, then the function should abort if called incorrectly -- is not it possible that the asserts have been disabled in menuconfig?

surfskidude, 2020-02-03:
Hi David,

I am not directly calling esp_netif_new()

As a (new) test, I copied the following example "as is":
esp-idf/examples/protocols/http_server/file_serving
My only modifications to the above is the wi-fi name and password; set with: make menuconfig.

david-cermak, 2020-02-03:
This sounds very strange, as that example uses common `example_connect()` logic and creates wifi station from default params (so both config and key are not null) and the only issue could be duplicate interface key. Moreover the example-connect is called just once after the example startup.

Is not it possible that you've might have combined more examples in one application (like http-file-server plus wifi-simple-station)? This is very simple in `make` just forgotten `*.c` file somewhere in a main component? (seen that you've identified CMake as your build system, but typed `make ...` to the logs...)

Even so, if the the above failed as seen from the log, the program should `abort()` in the following assert (reset in the default config, perhaps) and not continue to fail as you describe. Could you please share the output of `cat sdkconfig | grep ASSERT`

surfskidude, 2020-02-03:
Just to make sure I did nothing wrong, I completely removed my installation and followed:
https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html

Here is the installation log, including copying example, compiling, and running. As you can see, I still get the same error.

cd ~/esp
git clone --recursive https://github.com/espressif/esp-idf.git
cd ~/esp/esp-idf
./install.sh
source export.sh
cd ..
cp -r ~/esp/esp-idf/examples/protocols/http_server/file_serving .
cd file_serving
make menuconfig

make -j8 flash monitor
Toolchain path: /home/surfskidude/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: esp-2019r2
Compiler version: 8.2.0
Python requirements from /home/surfskidude/esp/esp-idf/requirements.txt are satisfied.

Project is not inside a git repository, or git repository has no commits
will not use 'git describe' to determine PROJECT_VER.
App "file_server" version: 1
Flashing binaries to serial port /dev/ttyUSB0 (app at offset 0x10000)...
esptool.py v3.0-dev
Serial port /dev/ttyUSB0
Connecting.......
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: b4:e6:2d:df:63:01
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0220
Compressed 26672 bytes to 15928...
Wrote 26672 bytes (15928 compressed) at 0x00001000 in 1.4 seconds (effective 151.4 kbit/s)...
Hash of data verified.
Compressed 766224 bytes to 480470...
Wrote 766224 bytes (480470 compressed) at 0x00010000 in 42.6 seconds (effective 143.8 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 119...
Wrote 3072 bytes (119 compressed) at 0x00008000 in 0.0 seconds (effective 1391.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
MONITOR
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ta        01 02 00009�ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7144
load:0x40078000,len:14864
ho 0 tail 12 room 4
load:0x40080400,len:4568
entry 0x400806f0
I (31) boot: ESP-IDF v4.1-dev-2055-gf5b82c5b1-dirty 2nd stage bootloader
I (31) boot: compile time 08:31:04
I (32) boot: chip revision: 1
I (36) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (43) boot.esp32: SPI Speed      : 40MHz
[... 43 lines cut ...]
I (538) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (641) wifi: wifi driver task: 3ffc0a60, prio:23, stack:3584, core=0
I (641) system_api: Base MAC address is not set
I (641) system_api: read default base MAC address from EFUSE
I (661) wifi: wifi firmware version: 94efb94
I (661) wifi: config NVS flash: enabled
I (661) wifi: config nano formating: disabled
I (661) wifi: Init dynamic tx buffer num: 32
I (671) wifi: Init data frame dynamic rx buffer num: 32
I (671) wifi: Init management frame dynamic rx buffer num: 32
I (681) wifi: Init management short buffer num: 32
I (681) wifi: Init static rx buffer size: 1600
I (691) wifi: Init static rx buffer num: 10
I (691) wifi: Init dynamic rx buffer num: 32
I (701) example_connect: Connecting to DD-WRT...
I (791) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (801) wifi: mode : sta (b4:e6:2d:df:63:01)
I (801) example_connect: Waiting for IP
I (2851) example_connect: Wi-Fi disconnected, trying to reconnect...
I (4901) example_connect: Wi-Fi disconnected, trying to reconnect...
I (6951) example_connect: Wi-Fi disconnected, trying to reconnect...
I (9001) example_connect: Wi-Fi disconnected, trying to reconnect...
I (11051) example_connect: Wi-Fi disconnected, trying to reconnect...
I (13111) example_connect: Wi-Fi disconnected, trying to reconnect...

david-cermak, 2020-02-04:
Thanks for starting over from the clean repository. Now it looks a bit different. No error related to netif configuration, example is just trying to connect  but for some reason cannot. Is the password correct? You can switch to debug level (components -> log) to see more details about connecting.

As for the `_WindowOverflow4`, it's not an exception nor any error. This is just the way IDF monitor interprets addresses, documented [here](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html#automatic-address-decoding) and very useful if program crashes, but might be confusing too if numbers starting with `0x4...` appear in the log.

surfskidude, 2020-02-04:
Yes, I had a typo in the password and it is now working.

As for the original project: I made a fork of the "file_serving" example and used it as a base for a different server. I then upgraded esp-idf and it failed to work as initially stated. I now see that several API calls are different in the new "file_serving" example. After updating my project based on the new "file_serving" example, everything appears to be working, however, I have a question:

Will you continue to change APIs and break backward compatibility in future releases?

david-cermak, 2020-02-07:
@surfskidude glad to hear that it works now.

> Will you continue to change APIs and break backward compatibility in future releases?

IDF releases follow the versioning scheme documented [here](https://docs.espressif.com/projects/esp-idf/en/latest/versions.html#versioning-scheme), so I am afraid the incompatibilities may appear when switching between major release versions. Sorry for the inconvenience.

david-cermak, 2020-03-05:
@surfskidude Any other trouble or could this issue be closed?

closed 2020-03-05 by surfskidude