espressif/esp-idf #4705: BLE scan weird rssi values

muhambykar, 2020-02-03:
Hi everyone! I am working for a location detect system, the tags which have nrf modules from nordic semiconductor make broadcast 20 messages in one second. ESP32 will work as gateway, scans broadcast values and detect rssi values.  Thanks to rssi values, I want to detect its distance from the gateway. I took reliable rssi values, however sometimes it could be wrong. I have also scans the broadcast with nrf module at the same time, it shows always reliable rssi values. Therefore, this situation is normal for ESP32 modules, or it can be correctible situation? 

My outputs like that:

(41965) Device mac: D0:4C:06:8C:C9:51                   rssi: -33    44 AF 41 03 DA FE 00 00 D0 4C 06  rssi: -33 counter: DA FE  56062 detected error: 13  total: 5562 min max: -36  -29
(41975) Device mac: D0:4C:06:8C:C9:51                   rssi: -32    44 AF 41 03 DB 00 00 00 D0 4C 06  rssi: -32 counter: DB 00  56064 detected error: 13  total: 5563 min max: -36  -29
(41981) Device mac: D0:4C:06:8C:C9:51                   rssi: -34    44 AF 41 03 DB 01 00 00 D0 4C 06  rssi: -34 counter: DB 01  56065 detected error: 13  total: 5564 min max: -36  -29
(41986) Device mac: D0:4C:06:8C:C9:51                   rssi: -33    44 AF 41 03 DB 02 00 00 D0 4C 06  rssi: -33 counter: DB 02  56066 detected error: 13  total: 5565 min max: -36  -29
(41991) Device mac: D0:4C:06:8C:C9:51                   rssi: -32    44 AF 41 03 DB 03 00 00 D0 4C 06  rssi: -32 counter: DB 03  56067 detected error: 13  total: 5566 min max: -36  -29
(41996) Device mac: D0:4C:06:8C:C9:51                   rssi: -34    44 AF 41 03 DB 04 00 00 D0 4C 06  rssi: -34 counter: DB 04  56068 detected error: 13  total: 5567 min max: -36  -29
(42001) Device mac: D0:4C:06:8C:C9:51                   rssi: -94    44 AF 41 03 DB 05 00 00 D0 4C 06  rssi: -94 counter: DB 05  56069 detected error: 13  total: 5568 min max: -36  -29
 detected !!!!!!!!!!   -94
(42010) Device mac: D0:4C:06:8C:C9:51                   rssi: -33    44 AF 41 03 DB 07 00 00 D0 4C 06  rssi: -33 counter: DB 07  56071 detected error: 14  total: 5569 min max: -36  -29
(42016) Device mac: D0:4C:06:8C:C9:51                   rssi: -33    44 AF 41 03 DB 08 00 00 D0 4C 06  rssi: -33 counter: DB 08  56072 detected error: 14  total: 5570 min max: -36  -29
(42026) Device mac: D0:4C:06:8C:C9:51                   rssi: -34    44 AF 41 03 DB 0A 00 00 D0 4C 06  rssi: -34 counter: DB 0A  56074 detected error: 14  total: 5571 min max: -36  -29
(42031) Device mac: D0:4C:06:8C:C9:51                   rssi: -33    44 AF 41 03 DB 0B 00 00 D0 4C 06  rssi: -33 counter: DB 0B  56075 detected error: 14  total: 5572 min max: -36  -29
(42046) Device mac: D0:4C:06:8C:C9:51                   rssi: -33    44 AF 41 03 DB 0E 00 00 D0 4C 06  rssi: -33 counter: DB 0E  56078 detected error: 14  total: 5573 min max: -36  -29
(42056) Device mac: D0:4C:06:8C:C9:51                   rssi: -34    44 AF 41 03 DB 10 00 00 D0 4C 06  rssi: -34 counter: DB 10  56080 detected error: 14  total: 5574 min max: -36  -29
(42060) Device mac: D0:4C:06:8C:C9:51                   rssi: -32    44 AF 41 03 DB 11 00 00 D0 4C 06  rssi: -32 counter: DB 11  56081 detected error: 14  total: 5575 min max: -36  -29
(42066) Device mac: D0:4C:06:8C:C9:51                   rssi: -32    44 AF 41 03 DB 12 00 00 D0 4C 06  rssi: -32 counter: DB 12  56082 detected error: 14  total: 5576 min max: -36  -29
(42071) Device mac: D0:4C:06:8C:C9:51                   rssi: -34    44 AF 41 03 DB 13 00 00 D0 4C 06  rssi: -34 counter: DB 13  56083 detected error: 14  total: 5577 min max: -36  -29
(42076) Device mac: D0:4C:06:8C:C9:51                   rssi: -33    44 AF 41 03 DB 14 00 00 D0 4C 06  rssi: -33 counter: DB 14  56084 detected error: 14  total: 5578 min max: -36  -29

chegewara, 2020-02-04:
[image]
Yes, difference between 2 consecutive RSSI readings 60dB looks bad, but it may happen (interference).
As you can see from my picture RSSI fluctuating 20dB is nothing unusual.

muhambykar, 2020-02-05:
hi @chegewara , thanks for your comment.
I have also scanned with android phone as you did. It did not show weird values as ESP32. As I mentioned, I have also scanned with another NRF Bluetooth module, it also showed reliable values. This is why I asked is there any problem with ESP32 for bluetooth scan.

WCCWCC, 2020-06-01:
Hi @muhambykar ，
 * At present, ESP32 may have a 2db error when scanning. I have two ways to provide accuracy:

 * A: If you scan a single-channel data packet, the rssi value error will be smaller and will not exceed 1db. For example: Configure the ESP32 device to scan only on channel 37.

 * B: I can provide an interface so that you can know the channel and rssi information of the packet.

serenchoe, 2020-09-17:
Hi @WCCWCC  Could you please provide any method mentioned above? Either A or B would be of great help.

WCCWCC, 2020-09-18:
Hi ， @serenchoe 

I need the commit information of the idf you are currently using.

I provide a bt_lib to increase the single-channel scan function.

You only need to replace the bt_lib library.

serenchoe, 2020-09-19:
Hi @WCCWCC, 

Following is git revision of idf that I'm using now. 
6c17e3a64c02eff3a4f726ce4b7248ce11810833

Thanks

serenchoe, 2020-10-05:
Hi @WCCWCC 
I'm still waiting for your help.

WCCWCC, 2020-10-09:
Hi @serenchoe ,
Set the method of scanning only on a single channel.
Need to add the following code:

```c
typedef enum
{
   SCAN_ALL_CHANNEL = 0,
   ONLY_SCAN_CHANNEL_37,
   ONLY_SCAN_CHANNEL_38,
   ONLY_SCAN_CHANNEL_39,
}
scan_mode_config_t;
extern scan_mode_config_t scan_channel_setting(scan_mode_config_t scan_mode);
```

Before calling the API(esp_ble_gap_start_scanning), Call function(scan_channel_setting) to set scan channel.

serenchoe, 2020-10-16:
Hi @WCCWCC 
It works fine. Thanks a lot!

Alvin1Zhang, 2020-10-19:
Thanks for reporting and glad it works, feel free to reopen.

closed 2020-10-19 by Alvin1Zhang