espressif/esp-idf #4706: `esp_wifi_set_channel` return ESP_FAIL (-1) in promiscuous mode

mikhail-khalizev, 2020-02-03:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v3.3.1-91-g3386cb540
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG e082013) 5.2.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

`esp_wifi_set_channel` return ESP_FAIL (-1) in promiscuous mode with message:

```
W (2429) wifi: STA is scanning or connecting, or AP has connected with external STAs, cannot set channel
```

### Expected Behavior

`esp_wifi_set_channel` change channel without error.

### Actual Behavior

### Steps to reproduce

Run code below to reproduce this issue.

### Code to reproduce this issue

```cpp
#include <iostream>
#include <iomanip>
#include <esp_wifi.h>
#include <esp_system.h>
#include <esp_event_loop.h>
#include <nvs_flash.h>
#include <esp_log.h>

void wifi_promiscuous_cb(void* buf, wifi_promiscuous_pkt_type_t type)
{
    auto wifi_promiscuous_pkt = (wifi_promiscuous_pkt_t*)buf;
    printf("channel: %u\n", wifi_promiscuous_pkt->rx_ctrl.channel);
}

extern "C" void app_main()
{
    sleep(1);

    auto status = nvs_flash_init();
    if (status == ESP_ERR_NVS_NO_FREE_PAGES || status == ESP_ERR_NVS_NEW_VERSION_FOUND)
    {
        ESP_ERROR_CHECK(nvs_flash_erase());
        status = nvs_flash_init();
    }
    ESP_ERROR_CHECK(status);

    tcpip_adapter_init();
    ESP_ERROR_CHECK(esp_event_loop_init(NULL, NULL));
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg))
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_NULL));
    ESP_ERROR_CHECK(esp_wifi_start());

    wifi_promiscuous_filter_t flt;
    flt.filter_mask = WIFI_PROMIS_FILTER_MASK_DATA;
    ESP_ERROR_CHECK(esp_wifi_set_promiscuous_filter(&flt))
    ESP_ERROR_CHECK(esp_wifi_set_promiscuous_rx_cb(wifi_promiscuous_cb))
    ESP_ERROR_CHECK(esp_wifi_set_promiscuous(true))
    ESP_ERROR_CHECK(esp_wifi_set_channel(9, WIFI_SECOND_CHAN_NONE)); // <--- Fail

    std::cout << "Restart in 1 sec." << std::endl;
    sleep(1);
    esp_restart();
}
```
## Debug Logs

```
--- idf_monitor on /dev/ttyS5 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:7032
load:0x40078000,len:12096
ho 0 tail 12 room 4
load:0x40080400,len:6096
entry 0x40080714
I (70) boot: Chip Revision: 1
I (34) boot: ESP-IDF v3.3.1-91-g3386cb540-dirty 2nd stage bootloader
I (34) boot: compile time 12:35:05
I (34) boot: Enabling RNG early entropy source...
I (40) boot: SPI Speed      : 40MHz
I (44) boot: SPI Mode       : DIO
I (48) boot: SPI Flash Size : 4MB
I (52) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 otadata          OTA data         01 00 0000a000 00002000
I (70) boot:  1 nvs_key          NVS keys         01 04 0000c000 00001000
I (78) boot:  2 nvs              WiFi data        01 02 0000d000 00005000
I (85) boot:  3 fat              Unknown data     01 81 00012000 000ee000
I (93) boot:  4 factory          factory app      00 00 00100000 00180000
I (100) boot:  5 ota_0            OTA app          00 10 00280000 00180000
I (108) boot: End of partition table
I (112) boot: Defaulting to factory image
I (117) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x2a4b4 (173236) map
I (184) esp_image: segment 1: paddr=0x0012a4dc vaddr=0x3ffb0000 size=0x0302c ( 12332) load
I (189) esp_image: segment 2: paddr=0x0012d510 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /mnt/c/Users/micky/Documents/Source/espressif/esp-idf/components/freertos/xtensa_vectors.S:1779

I (192) esp_image: segment 3: paddr=0x0012d918 vaddr=0x40080400 size=0x026f8 (  9976) load
I (204) esp_image: segment 4: paddr=0x00130018 vaddr=0x400d0018 size=0x869b4 (551348) map
0x400d0018: _flash_cache_start at ??:?

I (395) esp_image: segment 5: paddr=0x001b69d4 vaddr=0x40082af8 size=0x10cf0 ( 68848) load
0x40082af8: pthread_mutex_lock_internal at /mnt/c/Users/micky/Documents/Source/espressif/esp-idf/components/pthread/pthread.c:494

I (435) boot: Loaded app from partition at offset 0x100000
I (435) boot: Disabling RNG early entropy source...
I (435) cpu_start: Pro cpu up.
I (439) cpu_start: Application information:
I (444) cpu_start: Project name:     awatem-master
I (449) cpu_start: App version:      v8-dirty
I (454) cpu_start: Compile time:     Feb  3 2020 12:34:57
I (460) cpu_start: ELF file SHA256:  64323d1a41e82e81...
I (466) cpu_start: ESP-IDF:          v3.3.1-91-g3386cb540-dirty
I (473) cpu_start: Starting app cpu, entry point is 0x400810fc
0x400810fc: call_start_cpu1 at /mnt/c/Users/micky/Documents/Source/espressif/esp-idf/components/esp32/cpu_start.c:269

I (0) cpu_start: App cpu up.
I (483) heap_init: Initializing. RAM available for dynamic allocation:
I (490) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (496) heap_init: At 3FFB9E58 len 000261A8 (152 KiB): DRAM
I (502) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (509) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (515) heap_init: At 400937E8 len 0000C818 (50 KiB): IRAM
I (521) cpu_start: Pro cpu start user code
I (95) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (2149) wifi: wifi driver task: 3ffc18a4, prio:23, stack:3584, core=0
I (2150) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (2154) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (2181) wifi: wifi firmware version: 3135e1d
I (2182) wifi: config NVS flash: enabled
I (2182) wifi: config nano formating: disabled
I (2183) wifi: Init dynamic tx buffer num: 32
I (2186) wifi: Init data frame dynamic rx buffer num: 32
I (2191) wifi: Init management frame dynamic rx buffer num: 32
I (2197) wifi: Init management short buffer num: 32
I (2202) wifi: Init static rx buffer size: 1600
I (2206) wifi: Init static rx buffer num: 10
I (2210) wifi: Init dynamic rx buffer num: 32
I (2298) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (2299) wifi: mode : null
I (2300) wifi: ic_enable_sniffer
W (2301) wifi: STA is scanning or connecting, or AP has connected with external STAs, cannot set channel

ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x40087208
0x40087208: _esp_error_check_failed at /mnt/c/Users/micky/Documents/Source/espressif/esp-idf/components/esp32/panic.c:720

file: "/mnt/c/Users/micky/Documents/Source/mikhail-khalizev/iot-esp32/source/main/main.cpp" line 39
func: void app_main()
expression: esp_wifi_set_channel(9, WIFI_SECOND_CHAN_NONE)

ELF file SHA256: 64323d1a41e82e81466e6164b5f877e5749e8ab9e4a5edabb5066f9a02718085

Backtrace: 0x40086c03:0x3ffbbd20 0x4008720b:0x3ffbbd40 0x400d3244:0x3ffbbd60 0x400d1a6f:0x3ffbbe20
0x40086c03: invoke_abort at /mnt/c/Users/micky/Documents/Source/espressif/esp-idf/components/esp32/panic.c:715

0x4008720b: _esp_error_check_failed at /mnt/c/Users/micky/Documents/Source/espressif/esp-idf/components/esp32/panic.c:721

0x400d3244: app_main at /mnt/c/Users/micky/Documents/Source/mikhail-khalizev/iot-esp32/source/main/main.cpp:25 (discriminator 1)

0x400d1a6f: main_task at /mnt/c/Users/micky/Documents/Source/espressif/esp-idf/components/esp32/cpu_start.c:552

Rebooting...
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

mikhail-khalizev, 2020-02-03:
In IDF version (run `git describe --tags` to find it): `v3.3` works without error

sagb2015, 2020-02-04:
@mikhail-khalizev  Thanks for reporting the issue! We will get back to you soon.

zhangyanjiaoesp, 2020-02-17:
@mikhail-khalizev It's our bug, and I have found the root cause, will merge it as soon as possible. Thanks for reporting.

mikhail-khalizev, 2020-03-11:
Any activity?

zhangyanjiaoesp, 2020-03-12:
@mikhail-khalizev This issue has been fix in master commit 4822e468c48bf8790a84997ade169e745656cebb. And we will backport it to v3.3.2

mikhail-khalizev, 2020-03-12:
Oh, great!
Could you also backport it to the `release/v4.0` branch?

mikhail-khalizev, 2020-03-12:
Since in v3.3 this issue fixed, I close this ticket.

closed 2020-03-12 by mikhail-khalizev

zhangyanjiaoesp, 2020-03-12:
> Oh, great!
> Could you also backport it to the `release/v4.0` branch?

Yes, we will backport to v4.0 also.

kunjanrshah, 2021-05-26:
is it resolved in 4.2 because with "WIFI_MODE_APSTA" mode in WiFi/SmartConfig Example I am getting "wifi:STA is scanning or connecting, or AP has connected with external STAs, cannot set channel" Error.Can someone please help?