espressif/esp-idf #4707: UART pattern detection generates unexpected event

iCarletto, 2020-02-03:
## Environment

- Development Kit: Pycom FiPy (ESP32 + Sequans Monarch SQN3330 LTE-M/NB-IoT)
- Module or chip used:  ESP32D0WDQ6 (revision 1)
- IDF version: v4.1-dev-2055-gf5b82c5b1
- Build System: idf.py
- Compiler version: (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Windows 10
- Environment type: Plain Command Prompt
- IDE: Visual Studio Code + Espressif IDF 0.1.3 
- Power Supply: USB

## Problem Description

The uart mechanism for pattern detection does not work as expected approximately in the 15% of the cases, generating a sequence of two events rather than one. An unexpected `UART_DATA` event is randomly emitted prior to the expected `UART_PATTERN_DET` event. The spurious `UART_DATA` event seems to appear especially after the escape sequence `\r\n`.

The problem can be observed for instance when exchanging AT commands with the modem SQN3330.

### Expected Behavior

Commands used and response:
- AT ->`\r\n`+`OK\r\n`
- AT+CGMI ->`\r\n`+`PYCOM\r\n`+`\r\n`+`OK\r\n`
- AT+CGMM ->`\r\n`+`FiPy\r\n`+`\r\n`+`OK\r\n`

Log:
```
I (782) uart: AT
EVENT_PATTERN   (2 bytes):      0d0a
EVENT_PATTERN   (4 bytes):      4f4b0d0a
I (782) uart: AT+CGMI
EVENT_PATTERN   (2 bytes):      0d0a
EVENT_PATTERN   (7 bytes):      5059434f4d0d0a
EVENT_PATTERN   (2 bytes):      0d0a
EVENT_PATTERN   (4 bytes):      4f4b0d0a
I (802) uart: AT+CGMM
EVENT_PATTERN   (2 bytes):      0d0a
EVENT_PATTERN   (6 bytes):      466950790d0a
EVENT_PATTERN   (2 bytes):      0d0a
EVENT_PATTERN   (4 bytes):      4f4b0d0a
```

### Actual Behavior

The `UART_DATA` event is randomly triggered before the `UART_PATTERN_DET`, when the uart buffer already contains the pattern to be detected.

```
I (43602) uart: AT
EVENT_PATTERN   (2 bytes):      0d0a
EVENT_DATA      (4 bytes):      4f4b0d0a
EVENT_PATTERN   (0 bytes):
W (43602) uart: Pattern Queue Size too small
I (43612) uart: AT+CGMI
EVENT_PATTERN   (2 bytes):      0d0a
EVENT_PATTERN   (7 bytes):      5059434f4d0d0a
EVENT_PATTERN   (2 bytes):      0d0a
EVENT_DATA      (4 bytes):      4f4b0d0a
EVENT_PATTERN   (0 bytes):
W (43632) uart: Pattern Queue Size too small
I (43632) uart: AT+CGMM
EVENT_PATTERN   (2 bytes):      0d0a
EVENT_PATTERN   (6 bytes):      466950790d0a
EVENT_PATTERN   (2 bytes):      0d0a
EVENT_DATA      (4 bytes):      4f4b0d0a
EVENT_PATTERN   (0 bytes):
W (43652) uart: Pattern Queue Size too small
Detection error rate: 14.98% of 8351
```

The warning "Pattern Queue Size too small" is generated because the `UART_DATA` event handler leaves an empty buffer to the following `UART_PATTERN_DET` event handler, which is then mistakenly interpreted as a queue overflow.
[... 388 lines cut ...]
EVENT_DATA      (4 bytes):      4f4b0d0a
EVENT_PATTERN   (0 bytes):
W (5712) uart: Pattern Queue Size too small
I (5712) uart: AT+CGMI
EVENT_PATTERN   (2 bytes):      0d0a
EVENT_PATTERN   (7 bytes):      5059434f4d0d0a
EVENT_PATTERN   (2 bytes):      0d0a
EVENT_DATA      (4 bytes):      4f4b0d0a
EVENT_PATTERN   (0 bytes):
W (5732) uart: Pattern Queue Size too small
I (5732) uart: AT+CGMM
EVENT_PATTERN   (2 bytes):      0d0a
EVENT_PATTERN   (6 bytes):      466950790d0a
EVENT_PATTERN   (2 bytes):      0d0a
EVENT_DATA      (4 bytes):      4f4b0d0a
EVENT_PATTERN   (0 bytes):
W (5752) uart: Pattern Queue Size too small
Detection error rate: 18.37% of 49
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

negativekelvin, 2020-02-04:
Is there any delay between bytes?  Have you checked with a logic analyzer?

iCarletto, 2020-02-04:
@negativekelvin I just checked with the logic analyzer and effectively there is a delay between groups of bytes. For instance, the response to `AT` come in three groups `\r\n` + `d1` + `OK` + `d2` + `\r\n`.

The delay `d2` between `OK` and  `\r\n` is approximately 82 us @ 921600 baud, but this does not result in a deterministic behavior.
`UART_DATA` emitted:
[image]
`UART_DATA` not emitted:
[image]

Scaling the baud rate to 115200 actually reduces the delays of a proportional factor resulting in zero emitted `UART_DATA` events.
`UART_DATA` never emitted:
[image]

However, the function
`uart_enable_pattern_det_baud_intr(uart_port, '\n',1,MIN_PATTERN_INTERVAL,MIN_POST_IDLE,MIN_PRE_IDLE)`
does not provide means for handling correctly these delays. Maybe a `MAX_INTRA_IDLE` parameter in the configuration of the pattern detection might actually help to solve this issue at higher baud rates.

negativekelvin, 2020-02-04:
Yes that's because at 921600 the RX timeout is only 10us but at 115200 it is 86us.

iCarletto, 2020-02-04:
Is it possible to manually change the isr timeout?

negativekelvin, 2020-02-04:
uart_set_rx_timeout

iCarletto, 2020-02-05:
@negativekelvin thanks a lot for the suggestion! it works great

closed 2020-02-05 by iCarletto