espressif/esp-idf #4708: VS code internal terminal(cmd.exe) arrow key didn't work!

imliubo, 2020-02-03:
## Environment

- Development Kit:     ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- IDF version (run ``git describe --tags`` to find it):
    v4.0-rc
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    1.22.0-80-g6c4433a
- Operating System:     Windows
- (Windows only) environment type:ESP Command Prompt
- Using an IDE?: VS code
- Power Supply:   USB

## Problem Description

I try compile the projects with VS code(just use for code edit),I use the terminal with VS code internal(cmd.exe), build&flash works well,but the  **menuconfig**  has problem,when I config the projects use **idf.py menuconfig** the arrow key didn't work.

Without VS code internal terminal works well.

### Expected Behavior
When I press ↓ key,the menu can move down. 

### Actual Behavior
[image]
**Tab** key can move the bottom row menu,and **Enter** key can enter menu.

igrr, 2020-02-08:
Hi @imliubo, I recommend having a look at https://github.com/espressif/vscode-esp-idf-extension, which is a dedicated VS Code extension for ESP-IDF projects. It comes with a configuration editor GUI, which may be easier to use than the command-line mconf window.

The problem related to arrow keys in VS Code terminal has been reported before (https://github.com/espressif/vscode-esp-idf-extension/issues/6), but I'm afraid it is out of scope of IDF to debug these terminal related issues.

imliubo, 2020-02-10:
@igrr 
Thanks,I didn't notice VS Code extension for ESP-IDF projects.I will try it,tnanks for help me.
Have a nice day!

closed 2020-02-10 by imliubo

Kusanagi242, 2024-04-06:
Use the J & K keys for UP/Down instead