espressif/esp-idf #4710: lastest IDF malfunction

binhla, 2020-02-04:
When change IDF from **version 3.3.1** to the lastest version, i cannot compile **hello_world** or any project anymore. The compile process is stucked on 

> CXX build/asio/asio/asio/src/asio.o

Anyone know what happenning, any thing i missing or how to solve it?

And this is the error log 

> Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
> WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-80-g6c4433a5
> Expected to see version: esp-2019r2
> Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
> WARNING: Compiler version is not supported: 5.2.0
> Expected to see version(s): 8.2.0
> Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
> Python requirements from C:/msys32/home/laian/esp/esp-idf/requirements.txt are satisfied.
> WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-80-g6c4433a5
> Expected to see version: esp-2019r2
> Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
> WARNING: Compiler version is not supported: 5.2.0
> Expected to see version(s): 8.2.0
> Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
> CC build/bootloader/bootloader_support/src/bootloader_random.o
> CC build/bootloader/bootloader_support/src/flash_encrypt.o
> CC build/bootloader/bootloader_support/src/esp_image_format.o
> CC build/bootloader/bootloader_support/src/flash_partitions.o
> CC build/bootloader/bootloader_support/src/bootloader_flash_config_esp32.o
> CC build/bootloader/bootloader_support/src/bootloader_clock.o
> CC build/bootloader/bootloader_support/src/bootloader_common.o
> CC build/bootloader/bootloader_support/src/bootloader_utility.o
> CC build/bootloader/bootloader_support/src/bootloader_init.o
> CC build/bootloader/bootloader_support/src/flash_qio_mode.o
> CC build/bootloader/bootloader_support/src/bootloader_flash.o
> CC build/bootloader/bootloader_support/src/bootloader_efuse_esp32.o
> CC build/bootloader/bootloader_support/src/esp32/flash_encrypt.o
> CC build/bootloader/bootloader_support/src/esp32/bootloader_esp32.o
> CC build/bootloader/bootloader_support/src/esp32/bootloader_sha.o
> CC build/bootloader/bootloader_support/src/esp32/secure_boot.o
> CC build/bootloader/bootloader_support/src/esp32/secure_boot_signatures.o
> AR build/bootloader/bootloader_support/libbootloader_support.a
> CC build/bootloader/log/log_buffers.o
> CC build/bootloader/log/log_noos.o
> CC build/bootloader/log/log.o
> AR build/bootloader/log/liblog.a
> CC build/bootloader/spi_flash/esp32/spi_flash_rom_patch.o
> AR build/bootloader/spi_flash/libspi_flash.a
> CC build/bootloader/micro-ecc/micro-ecc/uECC.o
> AR build/bootloader/micro-ecc/libmicro-ecc.a
> CC build/bootloader/soc/esp32/emac_hal.o
> CC build/bootloader/soc/esp32/rtc_clk.o
> CC build/bootloader/soc/esp32/rtc_time.o
> CC build/bootloader/soc/esp32/rtc_sleep.o
> CC build/bootloader/soc/esp32/gpio_periph.o
> CC build/bootloader/soc/esp32/i2s_periph.o
> CC build/bootloader/soc/esp32/rtc_clk_init.o
> CC build/bootloader/soc/esp32/ledc_periph.o
> CC build/bootloader/soc/esp32/spi_periph.o
> CC build/bootloader/soc/esp32/rtc_init.o
> CC build/bootloader/soc/esp32/sdio_slave_periph.o
> CC build/bootloader/soc/esp32/rtc_periph.o
> CC build/bootloader/soc/esp32/uart_periph.o
> CC build/bootloader/soc/esp32/rtc_wdt.o
> CC build/bootloader/soc/esp32/sdmmc_periph.o
> CC build/bootloader/soc/esp32/adc_periph.o
> CC build/bootloader/soc/esp32/i2c_periph.o
> CC build/bootloader/soc/esp32/interrupts.o
> CC build/bootloader/soc/esp32/cpu_util.o
> CC build/bootloader/soc/esp32/touch_sensor_periph.o
> CC build/bootloader/soc/esp32/rtc_io_periph.o
> CC build/bootloader/soc/esp32/touch_sensor_hal.o
[... 103 lines cut ...]
> In file included from C:/msys32/home/laian/esp/esp-idf/components/asio/asio/asio/include/asio/impl/src.hpp:51:0,
>                  from C:/msys32/home/laian/esp/esp-idf/components/asio/asio/asio/src/asio.cpp:11:
> C:/msys32/home/laian/esp/esp-idf/components/asio/asio/asio/include/asio/detail/impl/signal_set_service.ipp: In member function 'asio::error_code asio::detail::signal_set_service::clear(asio::detail::signal_set_service::implementation_type&, asio::error_code&)':
> C:/msys32/home/laian/esp/esp-idf/components/asio/asio/asio/include/asio/detail/impl/signal_set_service.ipp:404:50: error: no matching function for call to 'sigaction::sigaction(int&, sigaction*, int)'
>        if (::sigaction(reg->signal_number_, &sa, 0) == -1)
>                                                   ^
> In file included from c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\signal.h:5:0,
>                  from c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\csignal:42,
>                  from C:/msys32/home/laian/esp/esp-idf/components/asio/asio/asio/include/asio/detail/posix_signal_blocker.hpp:22,
>                  from C:/msys32/home/laian/esp/esp-idf/components/asio/asio/asio/include/asio/detail/signal_blocker.hpp:25,
>                  from C:/msys32/home/laian/esp/esp-idf/components/asio/asio/asio/include/asio/detail/impl/scheduler.ipp:26,
>                  from C:/msys32/home/laian/esp/esp-idf/components/asio/asio/asio/include/asio/impl/src.hpp:48,
>                  from C:/msys32/home/laian/esp/esp-idf/components/asio/asio/asio/src/asio.cpp:11:
> c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\sys\signal.h:145:8: note: candidate: sigaction::sigaction()
>  struct sigaction
>         ^
> c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\sys\signal.h:145:8: note:   candidate expects 0 arguments, 3 provided
> c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\sys\signal.h:145:8: note: candidate: constexpr sigaction::sigaction(const sigaction&)
> c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\sys\signal.h:145:8: note:   candidate expects 1 argument, 3 provided
> c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\sys\signal.h:145:8: note: candidate: constexpr sigaction::sigaction(sigaction&&)
> c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\sys\signal.h:145:8: note:   candidate expects 1 argument, 3 provided
> At global scope:
> cc1plus.exe: warning: unrecognized command line option '-Wno-frame-address'
> make[1]: *** [/home/laian/esp/esp-idf/make/component_wrapper.mk:292: asio/asio/src/asio.o] Error 1
> make: *** [C:\msys32\home\laian\esp\esp-idf/make/project.mk:604: component-asio-build] Error 2

AxelLin, 2020-02-04:
You need to update the toolchain to esp-2019r2.
See https://github.com/espressif/esp-idf/issues/4490

binhla, 2020-02-05:
For those guys that followed the stable version and using windows setup like me and want to upgrade your idf to lastest version. I had following these step
*Note: all command i used in **mingw32**
1. First step:
go to your esp directory
`cd ~esp`
these already old version of esp-idf, change ist name into esp-idf-old
`mv esp-idf esp-idf-3.3.1`
2. Second step:
download lastest version
`git clone --recursive https://github.com/espressif/esp-idf.git`
after  that, you might want to run
'python -m pip install --user -r $IDF_PATH/requirements.txt'
3. Third step:
Now if you try to compile a project, you might have error like i did.
So what you need to do is following instruction from this [https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#get-started-get-esp-idf](url) ultil finish step 3
4. Fouth step:
in C:\msys32\opt rename folder xtensa-esp32-elf
from C:\Users\your_name\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0
copy folder xtensa-esp32-elf to C:\msys32\opt

Then reopen **mingw32** and run make menuconfig, make with your project

binhla, 2020-02-05:
This is quite confuse to me because the ESP-IDF programming guide for stable and lastest vesion seemed using different setup

closed 2020-02-05 by binhla

David-E-Thomas, 2020-04-03:
After a number of incremental upgrades as version 4 of the idf came along, I'd reached a point where compiling asio.o failed as above - followed all the remedies above, but it didn't work.

Finally got it going by cleaning up my PATH (removing all references to .espressif folders).

Also removed the existing esp-idf and past versions (renamed folders), the entire .espressif folder in ~ and reinstalled the entire toolchain again.

I suspect the issue could have been solved by cleaning up the PATH variable though...