espressif/esp-idf #4712: ESP SD card error

rebeliousconformist, 2020-02-04:
## Environment

- Development Kit:      [ESP32-Wrover-Kit
- Kit version (for WroverKit: [v4]
- Module or chip used:  ESP32-WROVER-B
- IDF version (run ``git describe --tags`` to find it):v4.1-dev-2055-gf5b82c5b1
- Build System:         [Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
   xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0

- Operating System:     |Linux

- Power Supply:         [USB|

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

<pre>
Welcome to minicom 2.7.1

OPTIONS: I18n 
Compiled on Aug 13 2017, 15:25:34.
Port /dev/ttyUSB1, 17:29:28

Press CTRL-A Z for help on special keys

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7104
ho 0 tail 12 room 4
load:0x40078000,len:14864
load:0x40080400,len:4568
entry 0x400806f0
I (31) boot: ESP-IDF v4.1-dev-2055-gf5b82c5b1 2nd stage bootloader
I (31) boot: compile time 15:43:58
I (31) boot: chip revision: 1
I (35) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (42) boot.esp32: SPI Speed      : 40MHz
I (47) boot.esp32: SPI Mode       : DIO
I (51) boot.esp32: SPI Flash Size : 4MB
I (56) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (80) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (87) boot:  2 factory          factory app      00 00 00010000 00100000
I (95) boot: End of partition table
I (99) boot_comm: chip revision: 1, min. application chip revision: 0
I (106) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x099ac ( 39340) map
I (129) esp_image: segment 1: paddr=0x000199d4 vaddr=0x3ffb0000 size=0x021c4 (  8644) load
I (133) esp_image: segment 2: paddr=0x0001bba0 vaddr=0x40080000 size=0x00400 (  1024) load
I (136) esp_image: segment 3: paddr=0x0001bfa8 vaddr=0x40080400 size=0x04070 ( 16496) load
I (152) esp_image: segment 4: paddr=0x00020020 vaddr=0x400d0020 size=0x1fec4 (130756) map
I (200) esp_image: segment 5: paddr=0x0003feec vaddr=0x40084470 size=0x064f4 ( 25844) load
I (218) boot: Loaded app from partition at offset 0x10000
I (218) boot: Disabling RNG early entropy source...
I (219) cpu_start: Pro cpu up.
I (222) cpu_start: Application information:
I (227) cpu_start: Project name:     sd_card
I (232) cpu_start: App version:      1
[... 162 lines cut ...]
        ESP_LOGE(TAG, "Rename failed");
        return;
    }

    // Open renamed file for reading
    ESP_LOGI(TAG, "Reading file");
    f = fopen("/sdcard/foo.txt", "r");
    if (f == NULL) {
        ESP_LOGE(TAG, "Failed to open file for reading");
        return;
    }
    char line[64];
    fgets(line, sizeof(line), f);
    fclose(f);
    // strip newline
    char* pos = strchr(line, '\n');
    if (pos) {
        *pos = '\0';
    }
    ESP_LOGI(TAG, "Read from file: '%s'", line);

    // All done, unmount partition and disable SDMMC or SPI peripheral
    esp_vfs_fat_sdmmc_unmount();
    ESP_LOGI(TAG, "Card unmounted");
}

igrr, 2020-02-04:
Hi @rebeliousconformist, the `ESP_ERR_INVALID_CRC` error usually indicates a signal integrity issue, or signal level issue, between the ESP32 and the SD card.

Please check the troubleshooting section of the sd_card example, in particular the part about the pull-up resistors, and the part about GPIO12 strapping.

If these tips do not help, please attach the schematics of your PCB.

rebeliousconformist, 2020-02-05:
[image]

rebeliousconformist, 2020-02-05:
I'm using the stock, out of the box hardware

rebeliousconformist, 2020-02-05:
Rx Tx and the Cts Dts jumpers are in place

rebeliousconformist, 2020-02-05:
I tried using the SPI mode in the SD card example
Here is my prompt

<pre>
rst:0x1 (POWERON_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7208
load:0x40078000,len:14864
ho 0 tail 12 room 4
load:0x40080400,len:4568
entry 0x400806f0
I (31) boot: ESP-IDF v4.1-dev-2055-gf5b82c5b1 2nd stage bootloader
I (31) boot: compile time 16:39:57
I (31) boot: chip revision: 1
I (35) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (42) boot.esp32: SPI Speed      : 40MHz
I (47) boot.esp32: SPI Mode       : DIO
I (51) boot.esp32: SPI Flash Size : 4MB
I (56) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (80) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (87) boot:  2 factory          factory app      00 00 00010000 00100000
I (95) boot: End of partition table
I (99) boot_comm: chip revision: 1, min. application chip revision: 0
I (106) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x09e48 ( 40520) map
I (130) esp_image: segment 1: paddr=0x00019e70 vaddr=0x3ffb0000 size=0x021e4 (  8676) load
I (133) esp_image: segment 2: paddr=0x0001c05c vaddr=0x40080000 size=0x00400 (  1024) load
I (137) esp_image: segment 3: paddr=0x0001c464 vaddr=0x40080400 size=0x03bb4 ( 15284) load
I (152) esp_image: segment 4: paddr=0x00020020 vaddr=0x400d0020 size=0x2040c (132108) map
I (201) esp_image: segment 5: paddr=0x00040434 vaddr=0x40083fb4 size=0x06cf8 ( 27896) load
I (220) boot: Loaded app from partition at offset 0x10000
I (220) boot: Disabling RNG early entropy source...
I (221) cpu_start: Pro cpu up.
I (224) cpu_start: Application information:
I (229) cpu_start: Project name:     sd_card
I (234) cpu_start: App version:      1
I (239) cpu_start: Compile time:     Feb  5 2020 16:39:56
I (245) cpu_start: ELF file SHA256:  c4c27a575b879605...
I (251) cpu_start: ESP-IDF:          v4.1-dev-2055-gf5b82c5b1
I (257) cpu_start: Starting app cpu, entry point is 0x400810b0
I (0) cpu_start: App cpu up.
I (267) heap_init: Initializing. RAM available for dynamic allocation:
I (274) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (280) heap_init: At 3FFB2AB0 len 0002D550 (181 KiB): DRAM
I (287) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (293) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (299) heap_init: At 4008ACAC len 00015354 (84 KiB): IRAM
I (306) cpu_start: Pro cpu start user code
I (324) spi_flash: detected chip: generic
I (324) spi_flash: flash io: dio
I (325) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (333) example: Initializing SD card
I (333) example: Using SPI peripheral
I (343) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (393) sdspi_transaction: cmd=5, R1 response: command not supported
Name: SL16G
Type: SDHC/SDXC
Speed: 20 MHz
Size: 15193MB
I (613) example: Opening file hello95463
E (613) example: Failed to open file for writing
</pre>

Attached is the project

[sd_card.zip](https://github.com/espressif/esp-idf/files/4160469/sd_card.zip)

rebeliousconformist, 2020-02-05:
[D.zip](https://github.com/espressif/esp-idf/files/4160792/D.zip)
The SD Card contents

igrr, 2020-02-05:
@rebeliousconformist Thanks a lot for the extra details! Could you please try removing the RTS/CTS jumpers? They connect RTS/CTS signals of the FT2232H to GPIOs 13/15, which are also used for the SD interface. Since your application is not using the UART hardware flow control, you should be able to disconnect these jumpers.

[image]

rebeliousconformist, 2020-02-06:
hi, it tried it out its still has the same error.

igrr, 2020-02-07:
Thanks for trying this out. It seems there is some difference between our setups, either in the dev board (possibly a defect) or in the SD card.

In order to rule out some possible issues related to the board, could you please try running the following program on your dev board, removing the SD card first, and report the results?

https://gist.github.com/igrr/11e900beb5797e8448d674cce0b4c86d

rebeliousconformist, 2020-02-15:
hi @igrr im on vacation at the monent with unfortunately no access to hardware. I shall get back to you as soon as feasible.

rebeliousconformist, 2020-03-14:
@igrr  here is the output of the gist above.

<pre>
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:6884
load:0x40078000,len:13720
load:0x40080400,len:3920
entry 0x40080680
I (29) boot: ESP-IDF v4.2-dev-662-g06fa9476a 2nd stage bootloader
I (29) boot: compile time 23:38:22
I (29) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x06844 ( 26692) map
I (123) esp_image: segment 1: paddr=0x0001686c vaddr=0x3ffb0000 size=0x02194 (  8596) load
I (127) esp_image: segment 2: paddr=0x00018a08 vaddr=0x40080000 size=0x00400 (  1024) load
I (131) esp_image: segment 3: paddr=0x00018e10 vaddr=0x40080400 size=0x07208 ( 29192) load
I (152) esp_image: segment 4: paddr=0x00020020 vaddr=0x400d0020 size=0x13f70 ( 81776) map
I (184) esp_image: segment 5: paddr=0x00033f98 vaddr=0x40087608 size=0x0238c (  9100) load
I (193) boot: Loaded app from partition at offset 0x10000
I (193) boot: Disabling RNG early entropy source...
I (195) cpu_start: Pro cpu up.
I (199) cpu_start: Application information:
I (204) cpu_start: Project name:     sd_card
I (209) cpu_start: App version:      1
I (213) cpu_start: Compile time:     Mar 14 2020 23:38:27
I (219) cpu_start: ELF file SHA256:  d326b0e6fe380587...
I (225) cpu_start: ESP-IDF:          v4.2-dev-662-g06fa9476a
I (232) cpu_start: Starting app cpu, entry point is 0x4008101c
I (0) cpu_start: App cpu up.
I (242) heap_init: Initializing. RAM available for dynamic allocation:
I (249) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (255) heap_init: At 3FFB2980 len 0002D680 (181 KiB): DRAM
I (261) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (267) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (274) heap_init: At 40089994 len 0001666C (89 KiB): IRAM
I (280) cpu_start: Pro cpu start user code
I (298) spi_flash: detected chip: generic
I (299) spi_flash: flash io: dio
I (299) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
start code from ivan
SD Card I/O check. Make sure the SD card is NOT plugged in!

        **** Checking idle levels ****

PIN 12  D2  0.0V
PIN 13  D3  1.6V
PIN 14 CLK  3.3V
PIN 15 CMD  3.3V
PIN  2  D0  3.3V
PIN  4  D1  3.3V

        **** Checking levels with weak pullup ****

PIN 12  D2  2.3V
PIN 13  D3  1.6V
PIN 14 CLK  3.3V
PIN 15 CMD  3.3V
PIN  2  D0  3.3V
PIN  4  D1  3.3V

        **** Checking pin recovery time ****

PIN 12  D2  10011 cycles
PIN 13  D3  10112 cycles
PIN 14 CLK  154 cycles
PIN 15 CMD  156 cycles
PIN  2  D0  326 cycles
PIN  4  D1  326 cycles

        **** Checking pin recovery time with weak pullup ****

PIN 12  D2  342 cycles
PIN 13  D3  170 cycles
PIN 14 CLK  152 cycles
PIN 15 CMD  170 cycles
PIN  2  D0  324 cycles
PIN  4  D1  152 cycles

        **** Checking pin cross-talk ****

               D2    D3   CLK   CMD    D0    D1   
PIN 12  D2     --   1.6V  3.3V  3.3V  3.3V  3.3V  
PIN 13  D3    0.1V   --   3.3V  3.3V  3.3V  3.3V  
PIN 14 CLK    0.3V  1.6V   --   3.3V  3.3V  3.3V  
PIN 15 CMD    0.4V  1.6V  3.3V   --   3.3V  3.3V  
PIN  2  D0    0.4V  1.6V  3.3V  3.3V   --   3.3V  
PIN  4  D1    0.1V  1.6V  3.3V  3.3V  3.3V   --   

        **** Checking pin cross-talk with weak pull-up ****

               D2    D3   CLK   CMD    D0    D1   
PIN 12  D2     --   1.6V  3.3V  3.3V  3.3V  3.3V  
PIN 13  D3    2.3V   --   3.3V  3.3V  3.3V  3.3V  
PIN 14 CLK    2.3V  1.6V   --   3.3V  3.3V  3.3V  
PIN 15 CMD    2.3V  1.6V  3.3V   --   3.3V  3.3V  
PIN  2  D0    2.3V  1.6V  3.3V  3.3V   --   3.3V  
PIN  4  D1    2.1V  1.6V  3.3V  3.3V  3.3V   -- </pre>

igrr, 2020-03-16:
Thanks. Unfortunately It doesn't bring us closer to finding the issue, as the output is the same as what I get locally. Could you please try adding jumpers on "MTDI" and "MTCK" pin pairs? This should connect extra pull-up resistors to these pins, possibly resolving the issue.

rebeliousconformist, 2020-03-22:
Hi Igor, 
the card is working .. I had to reformat it to have bs of 512. They of the partition is msdos. 
Im sorry i just dont know what the issue was.

closed 2020-03-22 by rebeliousconformist