espressif/esp-idf #4713: ulp_s_sources is not working as expected in CMakeList.txt

sch0bert, 2020-02-05:
## Environment

- Development Kit: **Any** [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): **Any** [v1|v2|v3|v4]
- Module or chip used: **Any** [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it):
   **v4.0-rc**
- Build System:       **CMake idf.py**
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    **xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0**
- Operating System:     **Windows**
- (Windows only) environment type: **ESP Command Prompt.**
- Using an IDE?: **No**
- Power Supply: **External 3.3V**

## Problem Description

I'm migrating to CMake a project that I used to run in 3.x from IDF. After solving all compilation issues so far, I'm now facing one that seems to me is a ESP-IDF issue. I'm trying to add multiple .S files into the CMakeList.txt, in the "set(ulp_s_sources" area and seems to only add the first file, which in this case is "ulp/BatteryCharger.S"

```cpp
#
# ULP support additions to component CMakeLists.txt.
#
# 1. The ULP app name must be unique (if multiple components use ULP).
set(ulp_app_name ulp_${COMPONENT_NAME})
#
# 2. Specify all assembly source files.
#    Files should be placed into a separate directory (in this case, ulp/),
#    which should not be added to COMPONENT_SRCS.
set(ulp_s_sources ulp/BatteryCharger.S ulp/BatteryGauge.S ulp/HallSensor.S ulp/i2c.S ulp/i2c_BMI160.S ulp/i2c_LC709203F.S ulp/main.S ulp/stack.S)
#
# 3. List all the component source files which include automatically
#    generated ULP export file, ${ulp_app_name}.h:
set(ulp_exp_dep_srcs "main.c")
#
# 4. Call function to build ULP binary and embed in project using the argument
#    values above.
ulp_embed_binary(${ulp_app_name} ${ulp_s_sources} ${ulp_exp_dep_srcs})`
```

I noticed it because after compiling it (either clean or dirty), the ulp_main.h file generated have only variables declared extern from "ulp/BatteryCharger.S" and not also from the others files.

### Expected Behavior

To generate "ulp_main.h" with all the variables inside the files listed in "set(ulp_s_sources"

### Actual Behavior

Just generating "ulp_main.h" with variables from the first file listed which makes my compilation to fail because is not able to find variables I have in main from the other .S files.

### Steps to reproduce

1. Create a project that makes use of ULP
2. Make sure your project uses at least 2 assembly files with global variables each.
3. List them inside CMakeList.txt in main folder.
4. Compile and look for "ulp_main.h" inside "yourprojectname\build\esp-idf\main\ulp_main\ulp_main.h"

AeyrtonVillalobos, 2020-02-07:
I got an issue similar to yours, I hope they solve that asap

Alvin1Zhang, 2020-02-11:
@sch0bert Thanks for reporting and thanks @AeyrtonVillalobos for updates. We will look into this and provide feedback when available. Thanks.

renzbagaporo, 2020-02-13:
Hi @sch0bert,

This is an embarassing oversight for me, but adding quotes `""` around `${ulp_s_sources}` should make it work.

The examples work because `ulp_s_sources` and `ulp_exp_dep_srcs` contain only one file. We should probably update them and the docs.

sch0bert, 2020-02-13:
Hello @renzbagaporo,

Well I don't know if what you mean is from what I currently have:

`set(ulp_s_sources ulp/BatteryCharger.S ulp/BatteryGauge.S ulp/HallSensor.S ulp/i2c.S ulp/i2c_BMI160.S ulp/i2c_LC709203F.S ulp/main.S ulp/stack.S)
`

change it to this way:

`set(ulp_s_sources "ulp/BatteryCharger.S ulp/BatteryGauge.S ulp/HallSensor.S ulp/i2c.S ulp/i2c_BMI160.S ulp/i2c_LC709203F.S ulp/main.S ulp/stack.S")
`

Because I did it and it behaves the same way, it doesn't add any of the global variables from the others files (apart ulp/BatteryCharger.S) to ulp_main.h

How is the right way to make it to work?

iMazstick, 2020-02-17:
Hi @sch0bert ,

Any updates on your issue? I am also experiencing the same thing. 

Thanks.

sch0bert, 2020-02-17:
Not yet @iMazstick I'm still waiting for their reply and resolution :(

sch0bert, 2020-02-25:
Hello guys, 8 days have passed since my last message. Any one looking at this?

renzbagaporo, 2020-02-26:
What I meant was to change to:

```
ulp_embed_binary(${ulp_app_name} "${ulp_s_sources}" "${ulp_exp_dep_srcs}")
```

You should keep 

```
set(ulp_s_sources ulp/BatteryCharger.S ulp/BatteryGauge.S ulp/HallSensor.S ulp/i2c.S ulp/i2c_BMI160.S ulp/i2c_LC709203F.S ulp/main.S ulp/stack.S)
```

as is.

sch0bert, 2020-02-26:
This is not fixing the issue, it behaves the same even with your modification.

iMazstick, 2020-02-27:
I can confirm what @sch0bert is saying. This does not resolve my issue either.

hectorMtz017, 2020-02-27:
Hello guys, I'm also stuck on that problem, I've also tried what @renzbagaporo post, but yet same issue without any fix... Please help...

AeyrtonVillalobos, 2020-02-28:
I noticed I'm not the only one still getting this issue...

Alvin1Zhang, 2020-02-28:
@sch0bert @AeyrtonVillalobos @hectorMtz017 @iMazstick Thanks for feedback, we will look into, thanks.

petterarvidsson, 2020-03-03:
@Alvin1Zhang Any update? I also have the same issue and the fix from @renzbagaporo does not solve it.

This issue seems to have been around for a while and there is some additional information available here: https://esp32.com/viewtopic.php?t=12273

sch0bert, 2020-03-03:
I haven't tried that workaround yet but this could help the ESP guys to find a way to fix it quicker, I hope they make an official solution for this soon. It has been for so long...

igrr, 2020-03-03:
@sch0bert Very sorry for the delay, the fix for this issue has been submitted for review and will be merged shortly.

iMazstick, 2020-03-09:
I've update today and I am still unable to compile. Only the first assembly file is compiled.

renzbagaporo, 2020-03-09:
Hi @iMazstick, can you post the log of the build? @sch0bert has created an issue https://github.com/espressif/esp-idf/issues/4906, do you have the same experience?

iMazstick, 2020-03-09:
Hi @renzbagaporo, I have attached the log from the terminal as requested. I have also attached my CMake list file.

[CMakeLists.txt](https://github.com/espressif/esp-idf/files/4306506/CMakeLists.txt)

[esp32_idf_4.2_comp_8.2.0_multiple_assembler_build_log.txt](https://github.com/espressif/esp-idf/files/4306498/esp32_idf_4.2_comp_8.2.0_multiple_assembler_build_log.txt)

iMazstick, 2020-03-09:
Replacing
`set(ulp_s_sources "ulp/i2c.S" "ulp/i2c-adc.S" "ulp/i2c-util.S" "ulp/main.S" "ulp/stack.S" "ulp/adc.S")`
with
`set(ulp_s_sources "ulp/i2c.S ulp/i2c-adc.S ulp/i2c-util.S ulp/main.S ulp/stack.S ulp/adc.S")`
seems to help.

Essentially, all ulp files need to be declared within the two quotes.

renzbagaporo, 2020-03-09:
@iMazstick, this is somewhat equivalent to the initial practice of placing `${ulp_s_sources}` arg to `ulp_embed_binary` in quotes:

https://github.com/espressif/esp-idf/issues/4713#issuecomment-591195749

The docs and examples have also been updated to do it like this: bb639bb

closed 2020-03-09 by renzbagaporo

jgcjoec, 2020-03-11:
I can confirm that I've tried everything in this thread and the only thing that worked was back-porting the changes mentioned in 07a7152 into my local copy of 4.0.

Will this fix be back-ported into 4.0 or do we need to wait for 4.1?

renzbagaporo, 2020-03-12:
Yes this will be backported @jgcjoec.

sch0bert, 2020-03-16:
> Yes this will be backported @jgcjoec.

Any idea when is it going to be ported? I'm asking because my code used to compile with 4.0 version and now does not work with 4.2 due this issue: https://github.com/espressif/esp-idf/issues/4906