espressif/esp-idf #4715: heap Corruption after using BLE with Blufi

filzek, 2020-02-05:
Latest master as of 5/2/2020.

If use Blufi and after it completes invoke a task to write files to fat/wear leaveling, the system will corrupt after a malloc and a free. 

assertion "heap != NULL && "free() target pointer is outside heap areas"" failed
Backtrace: 0x40093019:0x3fffe7d0 0x400933c1:0x3fffe7f0 0x401beaff:0x3fffe810 0x40084c7b:0x3fffe840 0x4009d541:0x3fffe860 0x400e1341:0x3fffe880 0x40096c75:0x3ffff3d0

assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/esp-idf/components/heap/heap_caps.c", line 272, function: heap_caps_free

I did try everything possible, but this bug still happen all the time. Besides the code running perfect in any other part, but when run the BLUFI it is crashing all the system.

GYC-Espressif, 2020-03-03:
Hello. Sorry for the late reply. Could you help to check if the memory is malloc successfully? If you still have this issue, could you help give some code to reproduce this issue? Thank you.

Alvin1Zhang, 2020-05-26:
@filzek Thanks for reporting. Would you please help share if any updates for the issue? Thanks.

filzek, 2020-11-08:
Problem fixex in 4.1 release, but new problem surge if psiram is used and right after a ble connection you try to write to flash it will corrupt flash.

Takes around 9 seconds to clear the ble and allow you to write to flash.

WCCWCC, 2020-11-09:
Hi @filzek ,

There's a bug here that you can fix like this;
Otherwise, it will print incorrectly

**l2ble_update_att_acl_pkt_num not found p_tcb.**
The method of repair is shown below:
Modify the file (components/bt/host/bluedroid/btc/profile/esp/blufi/blufi_prf.c)
```c
`if (esp_ble_get_cur_sendable_packets_num(blufi_env.conn_id) > 0)
//Change the contents of this row to:
 if (esp_ble_get_cur_sendable_packets_num(BTC_GATT_GET_CONN_ID(blufi_env.conn_id)) > 0) 
```

If you still have a problem, please provide log and COMMIT information.

closed 2021-01-05 by filzek