espressif/esp-idf #4716: To update mbedtls library to latest stable relase

AxelLin, 2020-02-05:
Hi,
Current ESP-IDF master tree uses mbedtls v2.16.2.
The latest stable release for mbed TLS is v2.16.4 now.
There are a couple bug & security fixes, so suggest to upgrade the mbedTLS library.

BTW, the upgrade seems pretty trivial:
Generate a diff between v2.16.2 ~ v2.16.4 can be cleanly applied to ESP-IDF without conflict.
e.g. $ git diff mbedtls-2.16.2..mbedtls-2.16.4 > the_patch_to_apply.diff

AxelLin, 2020-02-24:
Hi,
The latest stable release is now mbedTLS-2.6.5.

Also FYI:
CVE-2019-16910 Side channel attack on deterministic ECDSA (fixed in 2.16.3)
CVE-2019-18222 Side channel attack on ECDSA (fixed in 2.16.4)

So this is not just feature request but also bug fix.

closed 2020-03-13 by espressif-bot

AxelLin, 2020-04-30:
Hi @mahavirj 
FYI:
The mbedtls-2.16.6 has been released which includes fix for CVE-2020-10932.
https://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2020-10932

The upgrade looks pretty trivial since the diff of 2.16.5~2.16.6 can be cleanly applied.

AxelLin, 2020-12-12:
Hi @mahavirj 
Just FYI, the Mbed TLS 2.16.9 has been released: https://github.com/ARMmbed/mbedtls/releases/tag/v2.16.9
And the upgrade looks pretty trivial since the diff of 2.16.7~2.16.9 can be cleanly applied.

mahavirj, 2020-12-13:
@AxelLin Thanks for notifying. I will start an internal review process for this change, please expect update in next few weeks.

AxelLin, 2021-03-15:
@mahavirj 

FYI, the mbedtls-v2.16.10 is released.
For detail, see https://github.com/ARMmbed/mbedtls/releases/tag/v2.16.10
The upgrade is trivial since the diff of 2.16.9-2.16.10 can be cleanly applied.

However, needs to check "default behavior changes":
In mbedtls_rsa_context objects, the ver field was formerly documented
as always 0. It is now reserved for internal purposes and may take
different values.

AxelLin, 2021-12-18:
@mahavirj 

New Mbed TLS versions available now with fixes for security issues.
Mbed TLS 2.16.12: https://github.com/ARMmbed/mbedtls/releases/tag/v2.16.12
Mbed TLS 2.28.0: https://github.com/ARMmbed/mbedtls/releases/tag/v2.28.0
Mbed TLS 3.1.0: https://github.com/ARMmbed/mbedtls/releases/tag/v3.1.0

Assume the esp-idf v5.0 will use MbedTLS-v3.1.x branch.
I think current esp-idf stable branches also need to update mbedtls library.
Since MbedTLS-v2.16.12 is the latest 2.16 LTS, I would suggest upgrade to MbedTLS-v2.28.0 instead
during the stable branches support period.

See https://lists.trustedfirmware.org/pipermail/mbed-tls/2021-December/000508.html

fellaz4sho, 2022-02-03:
Hello All,
I am new to git hub. I am using Arduino on a windows machine. In what folder do I update the Mbedtls in? I have downloaded the zip for the latest mbedtls but i dont what to do with it in other to make my arduino work. Thank you in advance.

AxelLin, 2022-07-12:
@mahavirj 

FYI, mbedTLS 3.2.0 and 2.28.1 released.
These releases of Mbed TLS address several security issues, provide bug fixes, and for 3.2.0, add various features.
https://lists.trustedfirmware.org/archives/list/mbed-tls@lists.trustedfirmware.org/thread/WJURSX7U55M35FIUXSDMLZCNH53U2ZTW/

AxelLin, 2022-12-15:
Hi @mahavirj 

FYI, Mbed TLS 3.3.0 and Mbed TLS 2.28.2 released.
https://lists.trustedfirmware.org/archives/list/mbed-tls@lists.trustedfirmware.org/thread/5B5RM7L4XOE7GIQUKXGQ4VQXZH4MYXQV/

AxelLin, 2023-03-28:
Hi @mahavirj 

FYI, New Mbed TLS releases : 3.4.0 and 2.28.3
https://lists.trustedfirmware.org/archives/list/mbed-tls@lists.trustedfirmware.org/thread/UID2W62AVOQNVIAJVVF664G3447UQ5RW/

AxelLin, 2023-08-04:
Hi @mahavirj 
FYI, New releases: Mbed TLS 3.4.1 and 2.28.4 LTS
https://lists.trustedfirmware.org/archives/list/mbed-tls@lists.trustedfirmware.org/thread/VACMQN4COUKJVNL2HLQSYUBU2YJ3OXQ3/

AxelLin, 2023-10-10:
Hi @mahavirj 

FYI, New Mbed TLS releases : 3.5.0 and 2.28.5

Both 3.5.0 and 2.28.5 include security fix, I think esp-idf needs to update mbedTLS.
https://lists.trustedfirmware.org/archives/list/mbed-tls@lists.trustedfirmware.org/thread/KELNGTOMKWFYVUJB4S6JBS4YSG4EKZCB/

AxelLin, 2023-11-09:
Hi @mahavirj 
FYI, Mbed TLS 3.5.1 and 2.28.6
These releases of Mbed TLS are made under a dual Apache-2.0 OR GPL-2.0-or-later license.
(Mbed TLS license change: https://lists.trustedfirmware.org/archives/list/mbed-tls@lists.trustedfirmware.org/thread/JSL65QQ3LSRKPGCBGWOQCV4BJS3E2LDR/)

AxelLin, 2024-01-28:
Hi @mahavirj 
FYI, Mbed TLS 3.5.2 and 2.28.7 released.
https://lists.trustedfirmware.org/archives/list/mbed-tls@lists.trustedfirmware.org/thread/HN7N7JHOJPIVYJKBV6W72GET5FH3MBN7/

These releases contain security fixes for: a timing side channel in private key RSA operations; and a buffer overflow in mbedtls_x509_set_extension.

AxelLin, 2024-03-29:
Hi @mahavirj 

FYI, Mbed TLS versions 3.6.0 LTS and 2.28.8 released
These releases of Mbed TLS address several security issues, provide bug fixes, and bring other minor changes.
https://lists.trustedfirmware.org/archives/list/mbed-tls@lists.trustedfirmware.org/thread/NUUTKSVEXXBFYMJA2BGMJMA267E7EWMJ/

AxelLin, 2024-08-31:
Hi @mahavirj 

FYI, Mbed TLS versions 3.6.1 and 2.28.9 released.
https://lists.trustedfirmware.org/archives/list/mbed-tls@lists.trustedfirmware.org/thread/S3IMUQNKZFXDQW6I5RYMFBQTFNV2HKAT/

AxelLin, 2024-10-15:
Hi @mahavirj 

FYI, Mbed TLS 3.6.2 release

This release of Mbed TLS provides the fix for a security vulnerability.
https://github.com/Mbed-TLS/mbedtls/releases/tag/mbedtls-3.6.2

AxelLin, 2025-03-24:
Hi @mahavirj 

FYI, Mbed TLS 3.6.3 and 2.28.10 released.
This release of Mbed TLS provides the fix for the TLS compatibility issue of handling fragmented handshake messages. This release includes fixes for security issues. So please update mbedtls for impacted esp-idf branches.
https://lists.trustedfirmware.org/archives/list/mbed-tls@lists.trustedfirmware.org/thread/AEPAQHMS7VK2TRF26VYJ2NC6CLV3O6Q4/

AxelLin, 2026-04-28:
Hi @mahavirj 
I'm wondering if espressif has timeline to address the mbedtls CVEs.
The dashboard shows CVEs in all esp-idf branches:
https://espressif.github.io/esp-idf-security-dashboard/