espressif/esp-idf #4717: Enable Secure Boot csum err:0xfa!=0x72(windows)

CJ0104, 2020-02-05:
## Environment

- Development Kit:none
- Module or chip used:  [ESP32-WROOM-32D|ESP32-WROOM-32U
- IDF version : v3.3.1
- Build System:         idf.py
- Compiler version :
xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System:     Windows
- (Windows only) environment type: Plain Command Prompt
- Using an IDE: No
- Power Supply:         USB

## Problem Description
I follow the https://docs.espressif.com/projects/esp-idf/en/latest/security/secure-boot.html#secure-boot-reflashable.
When I open the monitor it keep showing the
[image]

### Expected Behavior
Secure enable and led will blink.
### Actual Behavior
[image]

### Steps to reproduce
1.idf.py menuconfig 
[image]

2.idf.py build
[image]

3.
Python espefuse.py -p com18 burn_key secure_boot D:/ESP32/esp-idf-v3.3.1/blink/build/bootloader/secure-bootloader-key-256.bin
[image]

4.
Python esptool.py --chip esp32 -p com18 --baud 460800 --before default_reset --after no_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB  -u 0x1000 D:/ESP32/esp-idf-v3.3.1/blink/build/bootloader/bootloader.bin
[image]

5.
Python esptool.py --chip esp32 -p com18 --baud 460800  --before default_reset --after no_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB  -u 0x0 D:/ESP32/esp-idf-v3.3.1/blink/build/bootloader/bootloader-reflash-digest.bin
[image]

6.idf.py flash
[image]

7.idf.py monitor
[image]

PS:python espefuse.py -p com18 summary
[image]
[image]

### Code to reproduce this issue
[image]

Please tell me which step I miss or wrong.
I also want to enable the flash encryption ,so what should I do next.
Sorry for bother

negativekelvin, 2020-02-05:
https://docs.espressif.com/projects/esp-idf/en/v3.3.1/security/secure-boot.html#bootloader-size

CJ0104, 2020-02-06:
hi @negativekelvin  thank you ,I follow the docs and change like this
[image]
And
[image]
Then ,It's work!! 
Thank you so much.
But now I want to enable the flash encryption too,Below it's my action.
I follow the https://docs.espressif.com/projects/esp-idf/en/v3.3.1/security/flash-encryption.html
---------------------------------------------------------------------------------------
1.python espefuse.py -p com18 summary
[image]
Secure boot is enable.

And I change code like this
[image]

2.
docs said:
[image]
My action:
[image]
3.
idf.py build
[image]

4.re-flash bootloader
`Python esptool.py --chip esp32 -p com18 --baud 460800 --before default_reset --after no_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB -u 0x0 D:/ESP32/esp-idf-v3.3.1/blink/build/bootloader/bootloader-reflash-digest.bin
`
[image]

5.idf.py flash monitor
[image]
And the first time restart the esp32 will take about 30 second to 1 minutes.

Thank you @negativekelvin .

closed 2020-02-06 by CJ0104

CJ0104, 2020-02-06:
hi @negativekelvin  sorry for bother again.
I sudden run idf.py flash again.
Then the monitor print
[image]
And my summary 
[image]
[image]

I saw the 
[image]
It's mean Flash encryption mode counter Read Only Right?
So this esp32 was broken Am I right?

negativekelvin, 2020-02-06:
Yes

arunmagesh, 2020-09-06:
Hi @CJ0104   
Did you found any fix for the flash read err, 1000. 

I have the same issue. It worked once and I reflashed it and same error and ever since it won't accept even the signed&flash encrypted binary.

CJ0104, 2020-09-07:
Hi @arunmagesh .
If your esp32 have been encrypted,you can't flash any firmware into that esp32.
If your did the esp32 will broken and keep showing "flash read err, 1000".

And if you didn't flash any encrypted firmware into esp32,please check other issue .
There are a lot of issue about "flash read err, 1000" in this github.
Thank you and sorry about my poor english.
Best regard